embox/embox #2082: MAiX-BiT boot code

anton-bondarev, 2020-07-04:


sk2sat, 2020-07-29:
I'm making template for MAiX BiT but when I tried serial output with arch_init(), it didn't seem to boot at all.
I was wondering if it was because of a linker script or permissions.
Howeber, this is actually because the stack initialization dowsn't working well on a real board.

anton-bondarev, 2020-07-29:
Could you please create branch for this aim.
I agree it can be because there are any troubles in the linker script

sk2sat, 2020-07-29:
I think I don't have write permission yet. So, I'm working on [my repo](https://github.com/sk2sat/embox/tree/maixbit) and it works on real board now.
Shoulld I send PR from my repo?

anton-bondarev, 2020-07-29:
> I think I don't have write permission yet. So, I'm working on [my repo](https://github.com/sk2sat/embox/tree/maixbit) and it works on real board now.
> Shoulld I send PR from my repo?

@sk2sat You are in Embox team so you have to have write permissions and you can create any branches :)
You may work at your repo if you want. And yes, please create a PR

PR #2117 (2020-07-31): add MAiX BiT
fix #2082.
add MAiX BiT template and fix riscv64 boot code.
closed 2020-07-31 by anton-bondarev