embox/embox #2161: Porting Embox to Arduino Due

easimonenko, 2020-08-26:
I am currently porting Embox kernel to _Arduino Due_. This controller is based on MCU _Atmel SAM3X8E_ with _ARM Cortex-M3_ core. To do this, I created three files:

- `templates/arm/arduino_due/build.conf`:
``` conf
TARGET = embox

ARCH = arm
PLATFORM = arduino_due

CROSS_COMPILE = arm-none-eabi-

CFLAGS += -O0 -g
CFLAGS += -mcpu=cortex-m3 -mtune=cortex-m3

CFLAGS += -mfloat-abi=soft
```
- `templates/arm/arduino_due/lds.conf`:
``` config
/*
 * Linkage configuration.
 */

/* region (origin, length) */
ROM (0x00080000, 512K)
RAM (0x20070000, 96K)

/* section (region[, lma_region]) */
text   (ROM)
rodata (ROM)
data   (RAM, ROM)
bss    (RAM)
```
- `templates/arm/arduino_due/`:
``` config
package genconfig

configuration conf {
  include embox.arch.arm.cortexm3.bundle
  include embox.arch.arm.libarch
}
```

Then:

``` sh
make confload-arm/arduino_due
make
```

And I get two errors:

```
BUILDMODEL
(Unknown):: error: No abstract realization: embox.arch.arch.
```
Which module should I include to avoid this error?

```
embox/src/drivers/diag/diag.c:24:2: error: #error No impl option provided
```
What does this driver do? Transmits diagnostics by USART? Is there any way to disable it? Or do I need to port it to SAM3X8E?

anton-bondarev, 2020-08-26:
> ```
> embox/src/drivers/diag/diag.c:24:2: error: #error No impl option provided
> ```
> 
> 
> What does this driver do? Transmits diagnostics by USART? Is there any way to disable it? Or do I need to port it to SAM3X8E?

You could just add to your config following lines
	```
/* Use writing/reading to/from memory instead of using real serial port */
	include embox.driver.diag.mem_diag
	include embox.driver.diag(impl="embox__driver__diag__mem_diag")
``
it allows don't implement UART driver for SAM3X8E if it does not need you.

As for

> (Unknown):: error: No abstract realization: embox.arch.arch.

I think it's enough to add
```
include embox.arch.generic.arch
```
This module does nothing but add 
```
void arch_init(void) {
}

void arch_idle(void) {
}

void arch_shutdown(arch_shutdown_mode_t mode) {
	while (1);
}
```

easimonenko, 2020-08-26:
Thank you! It works.

But I got the following error now:
```
embox/src/arch/arm/armmlib/irq_entry.S:11:41: fatal error: module/embox/kernel/irq_api.h: No such file or directory
```
To fix this, I added the following lines to the config:
``` config
include embox.driver.interrupt.cortexm_nvic(irq_table_size=32)
include embox.kernel.irq
```
Only I'm not sure about `irq_table_size=32`. But now the sources are compiled.

Now I need to deal with linking errors:
```
section `.bss' will not fit in region `RAM'
```
and
```
region `RAM' overflowed by 43296 bytes
```

anton-bondarev, 2020-08-26:
> To fix this, I added the following lines to the config:
> 
> ```
> include embox.driver.interrupt.cortexm_nvic(irq_table_size=32)
> include embox.kernel.irq
> ```
> 
> Only I'm not sure about `irq_table_size=32`. But now the sources are compiled.

I'm not sure about `irq_table_size=32` I appear that it's not required now
And you should try  `include embox.kernel.irq_static` it places IRQ table directly to a fixed address. I think it's better for your aims

> region `RAM' overflowed by 43296 bytes

hm, it's weird for your small mods.conf but I think it will fix if you base your mods.conf on a small existing config for example platform/stellaris or arm/minimal. I think the trouble is that some of the default settings are quite large

easimonenko, 2020-08-27:
Thank you! I looked at `mods.conf` for platform/stellaris, and found an option for the stack, adding which helped.

Below is the minimum config with which the kernel is built for the Cortex-M3.
``` config
package genconfig

configuration conf {
	include embox.arch.generic.arch
	include embox.arch.arm.cortexm3.bundle
	include embox.arch.arm.libarch

	include embox.kernel.stack(stack_size=2048,alignment=4)

	include embox.driver.interrupt.cortexm_nvic
	include embox.kernel.irq_static

	include embox.driver.diag.mem_diag
	include embox.driver.diag(impl="embox__driver__diag__mem_diag")
}
```
Now I need to figure out how to blink LED.

anton-bondarev, 2020-08-28:
> Now I need to figure out how to blink LED.

For this purpose, you have to implement GPIO controller driver how described on [wiki](https://github.com/embox/embox/wiki/GPIO)