embox/embox #2289: mips_clk has only clock_event device

anton-bondarev, 2020-12-12:


kssuraaj28, 2020-12-13:
I looked at the source, and I have an overview of the issue. Can you describe how one may proceed?

anton-bondarev, 2020-12-13:
Look at mips_clk (src/drivers/clock/mips_clk/mips_clk.c) there are following lines 
```
static struct time_event_device mips_event_device  = {
	.set_periodic = mips_clock_setup,
	.name = "mips_clk",
	.irq_nr = MIPS_IRQN_TIMER
};
...
CLOCK_SOURCE_DEF(mips_clk, mips_clock_init, NULL,
	&mips_event_device, NULL);
```
Here is defining clock source with `time_event_device` here. You can also define `time_event_device` with CLOCK_SOURCE_DEF macro. Look at cortexm_systick for example

kssuraaj28, 2020-12-14:
I'll try this one

kssuraaj28, 2020-12-15:
If I understand correctly, you want me to add a `time_counter_device` in `mips_clk`, which should return 'internal' clock cycles, thereby increasing precision?