embox/embox #2298: Add support for TMP117 Temperature sensor

puranjaymohan, 2020-12-17:
I have this temperature sensor from TI called TMP117 which works over the i2c bus. I want to add the driver for this sensor in embox, what is the best place to put it, and are there any similar sensors available in Embox?
I have st Nucleo l476rg, to which I have already ported embox #2043, so I can test it with it.

anton-bondarev, 2020-12-17:
> I have this temperature sensor from TI called TMP117 which works over the i2c bus. I want to add the driver for this sensor in embox, what is the best place to put it, 

I think `src/drivers/sersors` is a fit place for it

> and are there any similar sensors available in Embox?

There are some supported sensors in Embox, which are also placed in `src/drivers/sersors`.

puranjaymohan, 2020-12-17:
@anton-bondarev I will add the driver for tmp117. But how are the drivers interfaced, I mean I can include the drivers/xyz/xzy.h in my application and call it's functions right?
Is there a command for testing sensors?

anton-bondarev, 2020-12-17:
> @anton-bondarev I will add the driver for tmp117. But how are the drivers interfaced, I mean I can include the drivers/xyz/xzy.h in my application and call it's functions right?

Yes. you can include your interface header directly from a command

> Is there a command for testing sensors?

No, it is not. You can look at test applications for each sensor in `src/cmds/hardware/sensors/' folder. I think you should go a similar  way

puranjaymohan, 2020-12-18:
I was trying to build the `platform/stm32/l4/nucleo_l476rg`, I included `include embox.cmd.hardware.sensors.adxl345` in the conf and got following errors:

```
./build/base/obj/embox.o: in function `i2c_dw_enable_and_wait':
/src/drivers/i2c/adapters/i2c_designware/i2c_designware.c:70: undefined reference to `usleep'
```

where is this usleep defined, what should I include?

anton-bondarev, 2020-12-18:
> where is this usleep defined, what should I include?

Try `include embox.compat.posix.util.sleep`

puranjaymohan, 2020-12-18:
Yes, it worked now, I built and tried to see if it reads the device id of my sensor. It turns out there is not i2c support for nucleo_l476rg or I should say stm32 L4.
I will have to add that first, please help me do that. also, what is i2c_designware?

anton-bondarev, 2020-12-18:
> Yes, it worked now, I built and tried to see if it reads the device id of my sensor. It turns out there is not i2c support for nucleo_l476rg or I should say stm32 L4.

No, unfortunately, Embox does not support i2c for L4.

> I will have to add that first, please help me do that. also, what is i2c_designware?

Yes, you should add support i2c for STM32L4. I look at current STM32 i2c drivers, and I hope it is not so difficult task. May be it's enough to add module description similar to 
```
@BuildDepends(third_party.bsp.st_bsp_api)
module stm32_i2c_f4 extends stm32_i2c {
	option number log_level=1
	option boolean use_i2c_irq=false
	option number i2c1_ev_irq=31
	option number i2c1_er_irq=32
	option number i2c2_ev_irq=33
	option number i2c2_er_irq=34

	@IncludeExport(path="drivers/i2c", target_name="stm32_i2c_conf.h")
	source "i2c_conf_f4.h"

	source "stm32_i2c.c"

	depends embox.driver.i2c
	depends embox.driver.gpio.api
	depends third_party.bsp.st_bsp_api
}
```
and header. Currently, only f4 header is presented

> also, what is i2c_designware?

I guess It is I2C driver for DE0 Nano SoC [https://github.com/embox/embox/wiki/DE0-Nano-SoC](https://github.com/embox/embox/wiki/DE0-Nano-SoC)

puranjaymohan, 2020-12-19:
I am curios about this include, how does this work?
`#include <module/embox/driver/i2c/stm32_i2c_f4.h>`

anton-bondarev, 2020-12-19:
> I am curios about this include, how does this work?
> `#include <module/embox/driver/i2c/stm32_i2c_f4.h>`

There are some headers for each module generated by Embox build system. If you include this one it means that you include generated header from 
```
package embox.driver.i2c
...
module stm32_i2c_f4 .. {
...

	source "i2c_conf_f4.h"
...
}
```
so you include a file which has *.h and is pointed with 'source' keyword

But it is better using `#include <drivers/i2c/stm32_i2c_conf.h>` in this place because `@IncludeExport(path="drivers/i2c", target_name="stm32_i2c_conf.h")`. Or so module stm32_i2c_f4 extends from stm32_i2c you can use `#include <module/embox/driver/i2c/stm32_i2c.h>`

puranjaymohan, 2020-12-22:
@anton-bondarev I was curious because `#include <module/embox/driver/i2c/stm32_i2c_f4.h>` this line is in the file `embox/src/drivers/i2c/adapters/stm32/i2c_conf_f4.h`, how is this working? if embox build system generates the file using source "i2c_conf_f4.h", then how it is included in itself?

anton-bondarev, 2020-12-22:
> i2c_conf_f4.h

@puranjaymohan But what is a problem? Firstly this file is included from another file and 
of cause, there is a macro guard in it (the file is included only once)
```
#ifndef SRC_DRIVERS_ADAPTERS_STM32_I2C_CONF_F4_H_
#define SRC_DRIVERS_ADAPTERS_STM32_I2C_CONF_F4_H_
...
#endif /* SRC_DRIVERS_ADAPTERS_STM32_I2C_CONF_F4_H_ */
```

puranjaymohan, 2020-12-23:
@anton-bondarev There is a big big problem, this makes it very difficult. If you see the `stm32f4xx_hal_i2c.h` here [ST's repo](https://github.com/STMicroelectronics/STM32CubeF4/blob/e084518f363e04344dc37822210a75e87377b200/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_i2c.h)

The I2C_InitTypeDef structure has this definition:-
```
typedef struct
{
  uint32_t ClockSpeed;       /*!< Specifies the clock frequency.
                                  This parameter must be set to a value lower than 400kHz */

  uint32_t DutyCycle;        /*!< Specifies the I2C fast mode duty cycle.
                                  This parameter can be a value of @ref I2C_duty_cycle_in_fast_mode */

  uint32_t OwnAddress1;      /*!< Specifies the first device own address.
                                  This parameter can be a 7-bit or 10-bit address. */

  uint32_t AddressingMode;   /*!< Specifies if 7-bit or 10-bit addressing mode is selected.
                                  This parameter can be a value of @ref I2C_addressing_mode */

  uint32_t DualAddressMode;  /*!< Specifies if dual addressing mode is selected.
                                  This parameter can be a value of @ref I2C_dual_addressing_mode */

  uint32_t OwnAddress2;      /*!< Specifies the second device own address if dual addressing mode is selected
                                  This parameter can be a 7-bit address. */

  uint32_t GeneralCallMode;  /*!< Specifies if general call mode is selected.
                                  This parameter can be a value of @ref I2C_general_call_addressing_mode */

  uint32_t NoStretchMode;    /*!< Specifies if nostretch mode is selected.
                                  This parameter can be a value of @ref I2C_nostretch_mode */

} I2C_InitTypeDef
``` 

Now if you see the same structure in `stm32l4xx_hal_i2c.h`here [ST's repo](https://github.com/STMicroelectronics/STM32CubeL4/blob/d023c0d560ace11509f9b761c8913a9e48fcf194/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_i2c.h)

The same structure has the following definition:-
```
typedef struct
{
  uint32_t Timing;              /*!< Specifies the I2C_TIMINGR_register value.
                                  This parameter calculated by referring to I2C initialization
                                         section in Reference manual */

  uint32_t OwnAddress1;         /*!< Specifies the first device own address.
                                  This parameter can be a 7-bit or 10-bit address. */

  uint32_t AddressingMode;      /*!< Specifies if 7-bit or 10-bit addressing mode is selected.
                                  This parameter can be a value of @ref I2C_ADDRESSING_MODE */

  uint32_t DualAddressMode;     /*!< Specifies if dual addressing mode is selected.
                                  This parameter can be a value of @ref I2C_DUAL_ADDRESSING_MODE */

  uint32_t OwnAddress2;         /*!< Specifies the second device own address if dual addressing mode is selected
                                  This parameter can be a 7-bit address. */

  uint32_t OwnAddress2Masks;    /*!< Specifies the acknowledge mask address second device own address if dual addressing mode is selected
                                  This parameter can be a value of @ref I2C_OWN_ADDRESS2_MASKS */

  uint32_t GeneralCallMode;     /*!< Specifies if general call mode is selected.
                                  This parameter can be a value of @ref I2C_GENERAL_CALL_ADDRESSING_MODE */

  uint32_t NoStretchMode;       /*!< Specifies if nostretch mode is selected.
                                  This parameter can be a value of @ref I2C_NOSTRETCH_MODE */

} I2C_InitTypeDef;
```

See `ClockSpeed` and `DutyCycle`

anton-bondarev, 2020-12-23:
@puranjaymohan Don't worry. It's only a trouble not 'big big problem':)
You can do follow things
Add STM32L4_CUBE flags for your cube in third-party/bsp/stml4cube/cube/Mybuild
```
...
@BuildArtifactPath(cppflags="-DSTM32L4_CUBE")
module cube_cppflags {

}
...
```
After that you can use somthing like this
```
	i2c_handle->Instance             = i2c;
	i2c_handle->Init.AddressingMode  = I2C_ADDRESSINGMODE_7BIT;
	i2c_handle->Init.DualAddressMode = I2C_DUALADDRESS_DISABLE;
	i2c_handle->Init.GeneralCallMode = I2C_GENERALCALL_DISABLE;
	i2c_handle->Init.NoStretchMode   = I2C_NOSTRETCH_DISABLE;
#ifdef STM32L4_CUBE
	...
#else
	
	i2c_handle->Init.ClockSpeed      = 400000;
	i2c_handle->Init.DutyCycle       = I2C_DUTYCYCLE_16_9;
#endif

```