embox/embox #2322: getaddrinfo() does not support SOCK_RAW

anton-bondarev, 2021-01-07:


VedantParanjape, 2021-02-20:
Hi, can you provide more details on this issue, I would like contribute to this.

anton-bondarev, 2021-02-21:
@VedantParanjape The simplest way to understand the details reproduce it on Linux
if you call getaddrinfo() with  `struct addrinfo::ai_socktype = SOCK_RAW` on Linux it returns all connections (UDP TCP) if you call the same on Embox it returns failed. Because Embox has support only SOCK_STREAM & SOCK_DGRAM

VedantParanjape, 2021-02-21:
Ohk, tried this. I assume this is a feature that needs to be added in lwip ?

anton-bondarev, 2021-02-21:
> assume

Not in lwIP. There is an own TCP/IP stack in Embox

VedantParanjape, 2021-02-21:
> > assume
> 
> Not in lwIP. There is an own TCP/IP stack in Embox

Ohk, found it. I need to update `getaddrinfo()` defined in here: https://github.com/embox/embox/blob/master/src/compat/posix/net/getaddrinfo.c ?

anton-bondarev, 2021-02-21:
Yes, I think so. But you may suggest another decision.
It is useful if you start with an appropriate test

VedantParanjape, 2021-02-21:
> Yes, I think so. But you may suggest another decision.
> It is useful if you start with an appropriate test

But the tests are already implemented: https://github.com/embox/embox/blob/master/src/tests/posix/netdb/getaddrinfo_test.c

anton-bondarev, 2021-02-21:
Yes, but I mean test for SOCK_RAW case

VedantParanjape, 2021-02-21:
Sure, will do that :)