embox/embox #2329: hello world : Command not found

shubhamr2511, 2021-01-16:
Hi, I am trying to follow quick start. Its showing some errors in output...not able to make "hello_world" application. Need help.
OS : Ubuntu 20.04
Embox cloned from github repository on Jan 16 2021. (11:02)

shubham@ThinkPad-W520:~/github$ git clone https://github.com/embox/embox
shubham@ThinkPad-W520:~/github$ cd embox/
shubham@ThinkPad-W520:~/github/embox$ sudo apt-get install make gcc build-essential gcc-multilib curl libmpc-dev python
shubham@ThinkPad-W520:~/github/embox$ sudo apt-get install gcc
shubham@ThinkPad-W520:~/github/embox$ sudo apt-get install qemu-system
shubham@ThinkPad-W520:~/github/embox$ make confload-x86/qemu
shubham@ThinkPad-W520:~/github/embox$ make

Output(copied error part only):
...
 MYFILE third-party/bsp/stmf7cube/components/Mybuild
 MYLINK: 890 files 
 CONFIGFILE conf/mods.conf
 CONFIGLINK
 BUILDMODEL
 BUILDGEN /home/shubham/github/embox/build
Error: modules dependency cycle:
	embox.driver.common
	embox.fs.driver.devfs_old
	embox.fs.rootfs
	embox.fs.syslib.file_system_full
Error: modules dependency cycle:
	embox.fs.node
	embox.driver.block_dev
	embox.driver.common
	embox.fs.driver.devfs_old
	embox.fs.rootfs
Error: modules dependency cycle:
	embox.compat.posix.fs.open_old
	embox.compat.posix.fs.dirent_old
	embox.fs.core
	embox.fs.node
	embox.driver.block_dev
	embox.driver.common
	embox.fs.driver.devfs_old
	embox.fs.rootfs
	embox.fs.syslib.file_system_full
Error: modules dependency cycle:
	embox.fs.core
	embox.fs.node
	embox.driver.block_dev
	embox.driver.common
	embox.fs.driver.devfs_old
Error: modules dependency cycle:
	embox.fs.core
	embox.fs.node
	embox.driver.block_dev
	embox.driver.common
	embox.fs.driver.devfs_old
	embox.fs.rootfs
	embox.fs.syslib.file_system_full
Error: modules dependency cycle:
	embox.fs.syslib.fs_full
	embox.fs.core
	embox.fs.node
	embox.driver.block_dev
	embox.driver.common
	embox.fs.driver.devfs_old
	embox.fs.rootfs
	embox.fs.syslib.file_system_full
Error: modules dependency cycle:
	embox.driver.common
	embox.fs.driver.devfs_old
	embox.fs.rootfs
	embox.fs.syslib.file_system_full
Error: modules dependency cycle:
[... 384 lines cut ...]
	@Runlevel(2) include embox.net.rarp
	@Runlevel(2) include embox.net.icmpv4
	@Runlevel(2) include embox.net.udp
	@Runlevel(2) include embox.net.tcp
	@Runlevel(2) include embox.net.udp_sock
	@Runlevel(2) include embox.net.tcp_sock
	@Runlevel(2) include embox.net.raw_sock
	@Runlevel(2) include embox.net.net_entry
	/* AF_INET, SOCK_STREAM, default */
	include embox.net.lib.dns_file

	include embox.compat.libc.all
	include embox.compat.libc.stdio.asprintf
	include embox.compat.libc.math_builtins
	include embox.compat.posix.pthread_key
	include embox.compat.posix.proc.atexit_stub
	include embox.compat.posix.fs.rewinddir_stub

	include embox.compat.atomic.pseudo_atomic

	include embox.util.LibUtil
	include embox.framework.LibFramework
}

```

anton-bondarev, 2021-01-16:
@shubhamr2511 It seems that the problem is in your Mybuild
```
src/cmds/hello_world/Mybuild:3:36: error: Unrecognized characters '', '', ''.
src/cmds/hello_world/Mybuild:3:62: error: Unrecognized characters '', '', ''.
```
I seem it can be strange quotes around 'First Embox application'

```
@Cmd(name = "hello_world", help=”First Embox application”)
```

try
```
     @Cmd(name = "hello_world", help="First Embox application")
```

shubhamr2511, 2021-01-17:
There were many errors when I run `make` command before "hello_world" application. Should I ignore them? . (read #2329 before "Making hello_world application"

shubhamr2511, 2021-01-17:
@anton-bondarev Thanks its working now!

shubhamr2511, 2021-01-17:
There are many module dependency errors when I run `make` (without hello_world application). Should I ignore these errors? or how they can be resolved?

anton-bondarev, 2021-01-17:
> There are many module dependency cycle errors when I run `make` (without hello_world application). Should I ignore these errors? or how they can be resolved?

Yes, unfortunately, You are right. There are several 'dependencies loop' errors. But it is the internal errors. You can ignore it for now. You should only try to not add new cyclic dependencies

shubhamr2511, 2021-01-17:
@anton-bondarev Thanks!

anton-bondarev, 2023-07-26:
@shubhamr2511 I've closed the issue.
Fell free open any  issue

closed 2023-07-26 by anton-bondarev