embox/embox #2342: Issue with tinypy

surya-mistral, 2021-02-08:
Hi,

I am building EMBOX image for arm-qemu platform. My objective is to get python interpreter to run any python files/applications.
In the process, i found a third-party package known as "tinypy"  & followed below procedure to integrate it to image.
> Added to conf/mods.conf by writing "include third_party.tinypy".
> Build the image.
> "help" command shows  "tinypy - Python interpreter"

But when i am trying to run "tinypy" it shows error.

root@embox:/#tinypy
Exception:
tp_get: KeyError: 0

tish error: #255

Any suggestion, how we can run any python files in embox.

Thanks & regards
Surya

anton-bondarev, 2021-02-08:
@surya-mistral Hi

You have done everything right. But you should specify a python script which you want to interpret
I have checked it with the following file
```
def fib(n):
    if n == 0:
        return 0
    elif n == 1:
        return 1
    else:
        return fib(n-1) + fib(n-2)

print("fib(7) =" fib(7))
```
You just need to add fib.py (with this content) to conf/rootfs/ folder. Build Embox again `make`
And calling tinypy with it. 
```
tinypy fib.py
```
Of course, you can check whether the file exists or not
`ls fib.py`
and also the content
`cat fib.py`
It's only base checking of calling 'tinypy' on Embox

surya-mistral, 2021-02-10:
Thanks a lot anton-bondarev !!!

Its working now. :)

closed 2021-02-10 by surya-mistral