embox/embox #2344: python

surya-mistral, 2021-02-09:


anton-bondarev, 2021-02-09:
@surya-mistral Hi
What exactly do you mean?
You added #2342 with the trouble about 'tinypy'. Does 'tinypy' not fit for your purposes? Or you want something else?

surya-mistral, 2021-02-10:
Hi anton-bondarev,

Sorry, It was wrongly posted. Yes, python file does work with tinypy now. Thanks a lot !!
I have few more queries.
1. Does EMBOX supports STM32H753 series ?
2.  Does OpenCV in EMBOX supports STM32H753 ? I can see only two STM32F series chipsets are supported now.

anton-bondarev, 2021-02-11:
@surya-mistral 
>     1. Does EMBOX supports STM32H753 series ?

Now does not. But there are h743 and h745 chips support. So you could the wished series. It is rather simple I think. You should create an issue for this purpose where we can discuss this process

>     2. Does OpenCV in EMBOX supports STM32H753 ? I can see only two STM32F series chipsets are supported now.
Yes there are only two ready examples, but You could add OpenCV as a module to your config and it would work

surya-mistral, 2021-02-12:
> @surya-mistral
>
>
> Now does not. But there are h743 and h745 chips support. So you could the wished series. It is rather simple I think. You should create an issue for this purpose where we can [...]

Thanks a lot Anton-bondarev.