embox/embox #2348: Add tests for wcslen()

anton-bondarev, 2021-02-11:


VedantParanjape, 2021-02-20:
Hi, I would like to work on this issue. I successfully compiled embox and wrote a hello_world example. I am not sure where to add unit test for `wcslen()`, Do I need to add a new source file for testing this function here ?

https://github.com/embox/embox/tree/master/src/tests/c%2B%2B

anton-bondarev, 2021-02-20:
Closed by #2366

85ayush, 2021-02-20:
Hello I would love to work on this issue, I am new to open source contribution( have basic knowledge of git and git hub), and I would give my 100%.

wawuwo, 2023-08-11:
Hi @anton-bondarev! It appears that this issue could be closed. A couple of comments [higher](#issuecomment-782663943) you mention a PR which resolves this issue.

anton-bondarev, 2023-08-12:
> Hi @anton-bondarev! It appears that this issue could be closed. A couple of comments [higher](#issuecomment-782663943) you mention a PR which resolves this issue.

Oh, thanks a lot!

closed 2023-08-12 by anton-bondarev

anton-bondarev, 2023-08-12:
> Hello I would love to work on this issue, I am new to open source contribution( have basic knowledge of git and git hub), and I would give my 100%.

@85ayush Sorry for the long reply. You can extend this test for a few more cases or it's better if you work on another issue