embox/embox #2349: Add tests for wcsncasecmp()

anton-bondarev, 2021-02-11:


VirenderNayal, 2021-02-19:
Hello,
I want to work on this issue.
I am new to this repository, can you help me to figure out.

anton-bondarev, 2021-02-19:
@VirenderNayal I've assigned it to you.

You should start with the build and run Embox on QEMU See [Quick-Start](https://github.com/embox/embox/wiki/Quick-Start)
After this, you have to add tests for wcsncasecmp() debug it and create PR with your cahnges

VirenderNayal, 2021-02-19:
Thanks for assigning it to me..
I have followed the quick-start guide, after the `make` process

I ran this command :
`sudo ./scripts/qemu/auto_qemu`

It is giving an error:

[image]

I am using Endeavour OS (Arch Based)

alexkalmuk, 2021-02-20:
@VirenderNayal Can you please tell your QEMU version?

VirenderNayal, 2021-02-20:
@alexkalmuk I have tried to check, see this:

[image]

alexkalmuk, 2021-03-02:
@VirenderNayal Sorry for the delay, could you please tell me which template you build? `x86/qemu`? Also, what is your gcc version?

```
$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
```

Also, show, please

```
$ file ./build/base/bin/embox
./build/base/bin/embox: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), statically linked, with debug_info, not stripped
```

VirenderNayal, 2021-03-03:
@alexkalmuk Here are the details:
I used x86/qemu

[image]

alexkalmuk, 2021-03-04:
@VirenderNayal I built Embox with gcc version 10.2-0 and run with qemu-5.2.0, so I still have no idea what's going on. Can you please attach your `build/base/bin/embox` here?

VirenderNayal, 2021-03-06:
@alexkalmuk 
`build/base/bin/embox` : 

[image]

I have done all the processes again and ran the command and :
`./scripts/qemu/auto_qemu`

[image]

I think the problem is with these:

`VT is not supported by CPU
dmesg: read kernal buffer failed: Operation not permitted`

alexkalmuk, 2021-03-06:
@VirenderNayal I meant can you please attach your file build/base/bin/embox so I could launch it too?

alexkalmuk, 2021-03-06:
Can you please also try to launch it with sudo?

```
$ sudo ./scripts/qemu/auto_qemu
```

VirenderNayal, 2021-03-06:
I ran with sudo no change.

alexkalmuk, 2021-03-06:
Ok, then can you please attach your file here? (build/base/bin/embox)

VirenderNayal, 2021-03-06:
I am unable to attach my build/base/bin/embox
It says format not supported

alexkalmuk, 2021-03-06:
Ah. ok

VirenderNayal, 2021-03-06:
I got this after i ran with sudo

[image]

alexkalmuk, 2021-03-06:
Can you add your file to for example https://filebin.net/ and provide me a link?

VirenderNayal, 2021-03-06:
Here (build/base/bin/embox):

https://filebin.net/5wvs93u9nnukop6o

alexkalmuk, 2021-03-06:
Thanks, it reproduced.  I will go back as soon as we find a solution.

VirenderNayal, 2021-03-06:
okay thanks @alexkalmuk

alexkalmuk, 2021-03-06:
btw could you show

```
$ ld --version
```

VirenderNayal, 2021-03-06:
Yeah Sure.

[image]

niklvdanya, 2023-09-29:
Hi, I would like to get assigned.

anton-bondarev, 2023-09-30:
Hi @niklvdanya , I've assigend it with you

Khokholeva, 2023-09-30:
Hello, I would like to work on this issue

niklvdanya, 2023-10-01:
Cloded #2959

anton-bondarev, 2023-10-01:
Hello @Khokholeva
Today some tests have already been added. It's better if you choose another "googd first issue" (I've added some more now), but you can also expand existed test cases or add your implementation.

anton-bondarev, 2023-10-05:
Closed by #2959 #2966 #2967

closed 2023-10-05 by anton-bondarev