embox/embox #2364: Test for strlen uses strnlen to test

VedantParanjape, 2021-02-20:
Since, the test is meant for tesing strlen, it should be used. But both test cases use strnlen, isn't this unwanted ?
Even the test suite description is this
https://github.com/embox/embox/blob/929f2dde3629ceef2047c9b0e4b765bd1495f9cc/src/tests/stdlib/strlen.c#L11
and the code uses `strnlen()` instead
https://github.com/embox/embox/blob/929f2dde3629ceef2047c9b0e4b765bd1495f9cc/src/tests/stdlib/strlen.c#L19

alexkalmuk, 2021-02-20:
For me, it would be ok to test both `strlen` and `strnlen` in a single test suite. Probably, these test cases can be retained and add new test cases for `strlen` (in a simpler case these test cases can be copied and modified correspondingly).

VedantParanjape, 2021-02-20:
@alexkalmuk https://github.com/embox/embox/pull/2366/commits/7010283b2cfb3c3071b9349541d3af078a626b35 Will doing something like this suffice ?

alexkalmuk, 2021-02-20:
Your link related to `wcslen` nor `strlen`, but I told only about extending `strlen` test suite.

alexkalmuk, 2021-02-20:
But yes, I think we could have the tests for `strlen` test suite as well.

VedantParanjape, 2021-02-20:
> Your link related to `wcslen` not to `strlen`, but I told only about extending `strlen` test suite.

Yes, I meant, will a similar change to strlen suffice :)