embox/embox #2365: Repeat of same code in compat implementation of wcslen()

VedantParanjape, 2021-02-20:
I looked up implementation of `wcslen()` in here
https://github.com/embox/embox/blob/929f2dde3629ceef2047c9b0e4b765bd1495f9cc/src/compat/libc/wchar/wcslen.c#L16

Why is this line repeated thrice in the while loop ?
```c
if (s[++len] == L'\0') {
	return len;
}
```

alexkalmuk, 2021-02-20:
I do not think it's a bug, most likely it was taken from GLIBC's [implementation](https://code.woboq.org/userspace/glibc/wcsmbs/wcslen.c.html#27). It looks like "loop unrolling" optimization.

VedantParanjape, 2021-02-20:
> I do not think it's a bug, most likely it was taken from GLIBC's [implementation](https://code.woboq.org/userspace/glibc/wcsmbs/wcslen.c.html#27). It looks like "loop unrolling" optimization.

Figured, since size of wchar_t is 4 bytes, that's why 4 times.

closed 2021-02-20 by VedantParanjape