embox/embox #2381: OpenCV on STM32 : Sample build fails

Ripitchip00, 2021-03-08:
Hello, 

I'm trying to build OpenCV first for Qemu (https://github.com/embox/embox/wiki/OpenCV). The first _make_ gaves the same error as described. The second time it gaves me [another error](https://zupimages.net/up/21/10/3dcf.jpg).

Then, I tried to build for the STM32F746G-DISCOVERY and I had [an error](https://zupimages.net/up/21/10/rf3r.png) as well.

Do you know how I could fix it ?

Environment : Linux Ubuntu 20.04.2.0 LTS with gcc-arm-none-eabi.

Best regards

anton-bondarev, 2021-03-09:
@Ripitchip00 Hello,

I can not get error descriptions. Could you attach the logs file in text format?
`make > build_arm_qemu.log`

alexkalmuk, 2021-03-09:
@Ripitchip00 I see in your screenshot:

```
cmake: not found
```

Can you install `cmake`?

Ripitchip00, 2021-03-09:
Oooops, my bad ! So for Qemu, it's all right, I had to install cmake and Qemu.

For the STM32F746G-DISCOVERY, it doesn't find the net/net_namespace.h file. Should I add it myself ?

FYI : Two commands error :

[First](https://github.com/embox/embox/wiki/OpenCV-on-STM32) : 
```
make confload-project/opencv/stm32f7**46**g-discovery
```
[Second](https://github.com/embox/embox/wiki/OpenCV) :
```
make confload-**project**/opencv/arm-qemu
```

anton-bondarev, 2021-03-09:
@Ripitchip00 Thanks. We fixed `make confload-project/opencv/stm32f746g-discovery` build in #2382

anton-bondarev, 2021-03-09:
> FYI : Two commands error :
> 
> [First](https://github.com/embox/embox/wiki/OpenCV-on-STM32) :
> 
> ```
> make confload-project/opencv/stm32f7**46**g-discovery
> ```
> 
> [Second](https://github.com/embox/embox/wiki/OpenCV) :
> 
> ```
> make confload-**project**/opencv/arm-qemu
> ```

Do you mean you use '*'?
Unfortunately, there is no support for regular expression now

Ripitchip00, 2021-03-09:
No I don't mean to use ** **, I meant to show what should be corrected :) 
- 46 instead of 64.
- project instead of platform.

Thanks for PR, I'll try it.

alexkalmuk, 2021-03-10:
> However, when doing : sudo openocd -f /usr/share/openocd/scripts/board/stm32f7discovery.cfg, I get :

Did you insert your stm32 board into USB before launching openocd?

Ripitchip00, 2021-03-16:
Thanks for your help, it's okey now :)

closed 2021-03-16 by Ripitchip00