embox/embox #2405: Wierd code in compat/posix/curses/curses.c

anton-bondarev, 2021-04-05:
There is wierd code
```
	if (ch == '#') {
		ch = '#';
	}
```

https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L74

wawuwo, 2023-08-11:
Hi @anton-bondarev! It's been two years since this weird code fragment was discovered and it moved to another line https://github.com/embox/embox/blob/master/src/compat/posix/curses/curses.c#L86 since then. As I understand it is a do-nothing construct in any language and could be safely removed. Would it be OK if I prepare a patch to close the issue?

anton-bondarev, 2023-08-12:
> As I understand it is a do-nothing construct in any language and could be safely removed. Would it be OK if I prepare a patch to close the issue?

@wawuwo Yes, sure

PR #2891 (2023-08-12): posix/curses: Remove do-nothing block
This PR  closes #2405 by removing a do-nothing block from `curses.c`
commit c2a60142 (2023-08-12): posix/curses: Remove do-nothing block

Closes #2405
closed 2023-08-12 by anton-bondarev