embox/embox #2436: Linking fail with riscv64/qemu tamplate

AmrNashat, 2021-05-07:
Template: riscv64/qemu

Error message when using make:
EMBOX_GCC_LINK=full /home/user/Desktop/embox-workdir/embox/mk/extbld/arch-embox-gcc /home/user/Desktop/embox-workdir/embox/mk/extbld/toolchain_test.c -o /home/user/Desktop/embox-workdir/embox/build/base/obj/toolchain_test
./build/base/obj/embox.o: in function `__atomic_fetch_add_2':
/src/compat/atomic/pseudo_atomic/pseudo_atomic.c:63:(.text+0x1f7e4): relocation truncated to fit: R_RISCV_HI20 against `.L10'
collect2: error: ld returned 1 exit status
mk/extbld/toolchain.mk:73: recipe for target '/home/user/Desktop/embox-workdir/embox/build/base/obj/toolchain_test' failed
make[4]: *** [/home/user/Desktop/embox-workdir/embox/build/base/obj/toolchain_test] Error 1
mk/build.mk:19: recipe for target 'build' failed
make[3]: *** [build] Error 2
mk/load.mk:41: recipe for target 'build' failed
make[2]: *** [build] Error 2
mk/main.mk:30: recipe for target 'build' failed
make[1]: *** [build] Error 2
Makefile:40: recipe for target 'all' failed
make: *** [all] Error 2

AmrNashat, 2021-05-07:
This issue appeared as I was compiling Embox master branch, so I reverted to Tag: v0.5.2. However, I had the same issue.

anton-bondarev, 2021-05-07:
Thanks.
Please give me your gcc version. I've checked 
`make confload-riscv64/qemu`
and it built successful and correctly run on qemu
My gcc version
```
riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (SiFive GCC 8.3.0-2019.08.0) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

AmrNashat, 2021-05-07:
Thanks for checking.
My gcc version:

`riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

AmrNashat, 2021-05-12:
@anton-bondarev 
May you give me your `qemu-system-riscv64 --version`
Mine is 
`QEMU emulator version 6.0.0 (v6.0.0-dirty)`
`Copyright` (c) 2003-2021 Fabrice Bellard and the QEMU Project developers`

anton-bondarev, 2021-05-12:
@AmrNashat 
Oh, yes, it's a good idea
```
qemu-system-riscv64 --version
QEMU emulator version 4.2.1 (Debian 1:4.2-3ubuntu6.15)
Copyright (c) 2003-2019 Fabrice Bellard and the QEMU Project developers

```

anton-bondarev, 2021-05-17:
@AmrNashat Thank you for the report.

I reproduced a new problem with qemu-6.0.0
sudo AUTOQEMU_PREFIX=~/Downloads/qemu-6.0.0/build/riscv64-softmmu/ ./scripts/qemu/auto_qemu
AUTOQEMU_ARCH: riscv64 (guessed)
AUTOQEMU_MEM: 512 (guessed)
AUTOQEMU_NICS: (guessed)
AUTOQEMU_NICS_CONFIG: (got)
AUTOQEMU_AUDIO: (guessed)
AUTOQEMU_AUDIO_CONFIG: (got)
AUTOQEMU_USB: (guessed)
AUTOQEMU_USB_CONFIG: (got)
AUTOQEMU_GRAPHIC: (guessed)
AUTOQEMU_GRAPHIC_CONFIG: (got)
AUTOQEMU_DIAG: embox__driver__serial__ns16550 (guessed)
AUTOQEMU_DIAG_CONFIG: (got)
AUTOQEMU_KVM_ARG: (guessed)
AUTOQEMU_LOAD_ARG: -kernel ./build/base/bin/embox (guessed)
AUTOQEMU_MACHINE: -M virt (guessed)
```
/home/anton/Downloads/qemu-6.0.0/build/riscv64-softmmu/qemu-system-riscv64 -M virt -kernel ./build/base/bin/embox -m 512 -nographic
qemu-system-riscv64: Some ROM regions are overlapping
These ROM regions might have been loaded by direct user request or by default.
They could be BIOS/firmware images, a guest kernel, initrd or some other file loaded into guest memory.
Check whether you intended to load all this guest code, and whether it has been built to load to the correct addresses.

The following two regions overlap (in the memory address space):
  /home/anton/Downloads/qemu-6.0.0/build/pc-bios/opensbi-riscv64-generic-fw_dynamic.bin (addresses 0x0000000080000000 - 0x0000000080012558)
  ./build/base/bin/embox ELF program header segment 0 (addresses 0x0000000080000000 - 0x0000000080051b78)

But I have not still reproduced the compilation problem

```

AmrNashat, 2021-05-17:
Hello Anton, 
I forgot to update with mu input.

The compilation error can be resolved by adding option --with-cmodel=medany, while compiling the riscv-gnu-toolchain from source. It turns out that to enable this option both the code and the toolchain should be build with `code mode = medany`
More on this topic and explanation:  
https://forums.sifive.com/t/relocation-truncated-to-fit-r-riscv-hi20/2430

For this error is what I got after compiling qemu-6.0.0 from source and trying to run it. The opensbi binary is not aligned with binary from embox.
When I rolled back to qemu version 4.2.1, I got a different issue; qemu didn't find a bios binary at all.
I will post the exact output when I get my machine, it's under some maintenance, sorry for that! 

I was going to try to debug the embox linker script to shift the addresses, and avoid the overlap.