embox/embox #2453: fb_direct_access fails for platform/raspberry_pi/rpi0

kpishere, 2021-06-05:
A really incredible project embox!

In just getting introduced and doing some basic tests, it got far but the first test is failing at frame buffer step.

I'm attempting to run the platform/raspberry_pi/rpi0 built target on qemu and I get the following (below).   

```
Mac-mini:embox kpishere$ qemu-system-arm -kernel ./build/base/bin/embox -m 512 -serial stdio  -machine raspi0

Interrupt controller: raspi_intc

Embox kernel start
	unit: initializing embox.kernel.task.task_resource: done
	unit: initializing embox.kernel.task.task_table: done
	unit: initializing embox.kernel.task.kernel_task: done
	unit: initializing embox.driver.periph_memory_mmap: done
	unit: initializing embox.arch.arm.mmu_small_page: set ctx 0xae8000
done
	unit: initializing embox.mem.phymem: 
			start=0x0a5cd000 end=0x20008000 size=363048960
		done
runlevel: init level is 0
	unit: initializing embox.mem.static_heap: done
	unit: initializing embox.profiler.tracing: done
	unit: initializing embox.kernel.sched.sched: done
	unit: initializing embox.driver.clock.raspi_systick: done
	test: running embox.test.math.math_test .. done
runlevel: init level is 1
	unit: initializing embox.fs.buffer_cache: done
	unit: initializing embox.driver.flash.flash_nofs: done
	unit: initializing embox.driver.tty.serial: done
	unit: initializing embox.driver.video.raspi_video: done
	unit: initializing embox.driver.serial.pl011: done
	test: running embox.test.math.fpu_context_consistency_test . done
	test: running embox.test.posix.getopt_test .............. done
	test: running embox.test.posix.environ_test ......... done
	test: running embox.test.stdlib.qsort_test ....... done
	test: running embox.test.stdlib.bsearch_test ...... done
	test: running embox.test.posix.ppty_test ... done
	test: running embox.test.posix.pipe_test ... done
	test: running embox.test.posix.select_test ...... done
	test: running embox.test.posix.poll_test ......... done
	test: running embox.test.stdio.printf_test ......... done
	test: running embox.test.recursion . done
	test: running embox.test.critical .. done
	unit: initializing embox.fs.rootfs_dvfs: done
runlevel: init level is 2
runlevel: init level is 3
Default IO device[ttyS0]
>export PWD=/
>export HOME=/
>fb_direct_access
Data abort exception!
regs:
00050078 00000000 00000000 0000001f
5c100000 00033120 00033154 00032d40
00033120 e59f0060 00000000 01c58dfc
01c58de0 00077d14
```

kpishere, 2021-06-05:
Failing at `if (fb_draw_put_pix(r, g, b, 32, &screen_info, scr_pos + x) != 0) {` 

in 

`main () at /src/cmds/testing/fb_direct_access/fb_direct_access.c:24`

at 

`/src/lib/fb_draw/fb_draw.c:84`

the structure `screen_info` has bad pointer to frame buffer?  

```
(gdb) print *screen_info
$9 = {fbp = 0x5c100000 <error: Cannot access memory at address 0x5c100000>, width = 1024, height = 768,
  bpp = 16}
```

This could be embox or qemu issue.  Investigating.

kpishere, 2021-06-05:
By the following output on qemu (this is why I love qemu and not real hardware for development!) it looks like fbp should be pointing to the following 

`    vcram-base = 469762048 (0x1c000000)`

```
qemu) info qtree
bus: main-system-bus
  type System
  dev: unimplemented-device, id ""
    size = 256 (0x100)
    name = "bcm2835-sdramc"
    mmio ffffffffffffffff/0000000000000100
  dev: unimplemented-device, id ""
    size = 4096 (0x1000)
    name = "bcm2835-v3d"
    mmio ffffffffffffffff/0000000000001000
  dev: unimplemented-device, id ""
    size = 32768 (0x8000)
    name = "bcm2835-ave0"
    mmio ffffffffffffffff/0000000000008000
  dev: unimplemented-device, id ""
    size = 32768 (0x8000)
    name = "bcm2835-dbus"
    mmio ffffffffffffffff/0000000000008000
  dev: unimplemented-device, id ""
    size = 128 (0x80)
    name = "bcm2835-otp"
    mmio ffffffffffffffff/0000000000000080
  dev: unimplemented-device, id ""
    size = 32 (0x20)
    name = "bcm2835-i2c2"
    mmio ffffffffffffffff/0000000000000020
  dev: unimplemented-device, id ""
    size = 32 (0x20)
    name = "bcm2835-i2c1"
    mmio ffffffffffffffff/0000000000000020
  dev: unimplemented-device, id ""
    size = 32 (0x20)
    name = "bcm2835-i2c0"
    mmio ffffffffffffffff/0000000000000020
  dev: unimplemented-device, id ""
    size = 256 (0x100)
    name = "bcm2835-spis"
    mmio ffffffffffffffff/0000000000000100
  dev: unimplemented-device, id ""
    size = 32 (0x20)
    name = "bcm2835-spi0"
    mmio ffffffffffffffff/0000000000000020
  dev: unimplemented-device, id ""
    size = 256 (0x100)
    name = "bcm2835-smi"
    mmio ffffffffffffffff/0000000000000100
  dev: unimplemented-device, id ""
    size = 256 (0x100)
    name = "bcm2835-i2s"
    mmio ffffffffffffffff/0000000000000100
  dev: unimplemented-device, id ""
    size = 276 (0x114)
    name = "bcm2835-powermgt"
    mmio ffffffffffffffff/0000000000000114
  dev: unimplemented-device, id ""
    size = 64 (0x40)
    name = "bcm2835-sp804"
    mmio ffffffffffffffff/0000000000000040
  dev: unimplemented-device, id ""
    size = 4096 (0x1000)
    name = "bcm2835-txp"
    mmio ffffffffffffffff/0000000000001000
  dev: dwc2-usb, id ""
    gpio-out "sysbus-irq" 1
[... 93 lines cut ...]
    clock-out (alias) "gp0-out" freq_hz=0 Hz
    clock-out (alias) "dpi-out" freq_hz=0 Hz
    clock-out (alias) "dsi0p-out" freq_hz=0 Hz
    clock-out (alias) "dsi0e-out" freq_hz=0 Hz
    clock-out (alias) "ccp2-out" freq_hz=0 Hz
    clock-out (alias) "cam1-out" freq_hz=0 Hz
    clock-out (alias) "cam0-out" freq_hz=0 Hz
    clock-out (alias) "v3d-out" freq_hz=0 Hz
    clock-out (alias) "isp-out" freq_hz=0 Hz
    clock-out (alias) "h264-out" freq_hz=0 Hz
    clock-out (alias) "perii-out" freq_hz=0 Hz
    clock-out (alias) "peria-out" freq_hz=0 Hz
    clock-out (alias) "sys-out" freq_hz=0 Hz
    clock-out (alias) "vpu-out" freq_hz=0 Hz
    clock-out (alias) "gnric-out" freq_hz=0 Hz
    xosc-freq-hz = 19200000 (0x124f800)
    mmio ffffffffffffffff/0000000000002000
  dev: bcm2835-ic, id ""
    gpio-in "arm-irq" 8
    gpio-in "gpu-irq" 64
    mmio ffffffffffffffff/0000000000000200
  dev: bcm2835-peripherals, id ""
    gpio-out "sysbus-irq" 2
    mmio 0000000020000000/0000000001000000
```

kpishere, 2021-06-05:
So, I've found the fix for this.  Only, I don't know (yet) how to prepare this for a patch that would get accepted.  Here is the essence of the fix.  The scope of change is for raspi0 target emulated on qemu.

```
diff --git a/src/drivers/video/raspi_video.c b/src/drivers/video/raspi_video.c
index d78f307b3..d3f135623 100644
--- a/src/drivers/video/raspi_video.c
+++ b/src/drivers/video/raspi_video.c
@@ -63,7 +63,18 @@ static int raspi_init(void) {
         * By adding 0x40000000, we tell the GPU not to use its cache for these
         * writes, which ensures we will be able to see the change
         */
-       bcm2835_mailbox_write(((uint32_t) &raspi_fb) + 0x40000000, BCM2835_FRAMEBUFFER_CHANNEL);
+       #define VC_BUS_CACHED_L1_L2                     0x00000000      /* Only mode that works on qemu w. raspi0 emulation */
+       #define VC_BUS_CACHED_L2                                0x80000000
+       #define VC_BUS_DIRECT                                   0xC0000000
+       #define VC_BUS_CACHE_COHRNT_L2_NOALLOC  0x40000000
+
+       #if 1 /* For Raspberry Pi Zero and qemu */
+       #define VC_BUS VC_BUS_CACHED_L1_L2
+       #else
+       #define VC_BUS VC_BUS_CACHE_COHRNT_L2_NOALLOC
+       #endif
+
+       bcm2835_mailbox_write(((uint32_t) &raspi_fb) + VC_BUS, BCM2835_FRAMEBUFFER_CHANNEL);
        if (bcm2835_mailbox_read(BCM2835_FRAMEBUFFER_CHANNEL) != 0 ||
                        !raspi_fb.gpu_pointer) {
                return -EAGAIN;
```

anton-bondarev, 2021-06-05:
> So, I've found the fix for this. Only, I don't know (yet) how to prepare this for a patch that would get accepted. Here is the essence of the fix. The scope of change is for raspi0 target emulated on qemu.

@kpishre Thanks a lot!
Could you 
* Clone this repository. 
* Create a new branch 
* Commit your patch
* And create a new PR

[This page](https://github.com/embox/embox/wiki/Contributing%3A-Git-workflow) describes git workflow

kpishere, 2021-06-05:
@anton-bondarev   Yes, that part, I'm totally familiar with thanks.  The thing that holds me up as I'm not seeing examples of how, at compile time, I can tell what the target is.  I'd like it to sense that and use your default in all cases except for the raspi0 target.

anton-bondarev, 2021-06-06:
> @anton-bondarev Yes, that part, I'm totally familiar with thanks. The thing that holds me up is I'm not seeing examples of how, at compile time, I can tell what the target is. I'd like it to sense that and use your default in all cases except for the raspi0 target. Heck, this could even be switchable at boot time if that info were handy.

Do you mean that you want to specify a target in runtime? Or  it is enough what you have done in #2455
It is possible to modify the default options (Mybuild) in your 'mods.conf' file

kpishere, 2021-06-06:
Love the simplicity of Mybuild! It is really what makes this project stand out. I was initally looking for how to know target at compile time and choose then but with this mybuild methodology all these decisions go to mods.conf which in big picture makes for one place to check and define. So, with PR acceped, this can be closed.

closed 2021-06-06 by kpishere