embox/embox #2459: Unsupported items for target raspi0

kpishere, 2021-06-12:
Just to be sure I'm understanding correctly, I've gone over the code for embox and it looks like the following items are missing support for the raspi0 target: 

1) At boot, chip is not switched to Exception Level 3 which is typical exception level for the kernel to run in.
2) GPIO is unsupported
3) DMA is unsupported

I'm investigating adding these and would like to port a driver for a few SPI based LCDs, gonna have to get the above stuff working first though.  Also, my SWD ST-LINK, just does not appear to work with BCM2835!  So, will be some weeks before my new j-LINK V9 device arrives, QEMU simply will not do for this level of work.

anton-bondarev, 2021-06-13:
@kpishere Yes, you are right. Now all three points do not support Embox for RasPi. So it would be great if you added them.

> So, will be some weeks before my new j-LINK V9 device arrives,
No problem.

mfp20, 2021-06-25:
> my SWD ST-LINK, just does not appear to work with BCM2835! So, will be some weeks before my new j-LINK V9 device arrives, QEMU simply will not do for this level of work.

buy a Raspberry Pi Pico! It's a 5$ SWD programmer (using picoprobe firmware found in rpi fundation repo)! I stomped on this thread searching for someone working on the Pico (rp2040, it's an arm m0) port but I need to run in on a Zero as well.

kpishere, 2021-06-25:
@mfp20 Hah, I can't say that isn't good advise for those that are new.  Already got a V9 J-Link device, all is wired, preparing a boot firmware option now to set GPIO pins early and as needed and weekend is coming so stay tuned! PR is on it's way for GPIO support and JTAG enabled option on Pi Zero (and similar like 1 and 1+ ...).  Gathering the details has been an archeology exercise so I've been busy with this whilst waiting for hardware delivery to do some testing and validation.  Will probably make a video too to go along with showing test rig etc.

kpishere, 2021-07-10:
@anton-bondarev So, I'm working on implementing DMA.  The device is working fine, with bus memory addresses.  I'm having issues understanding accessing bus addressed memory, physical memory, and system memory.

Early on, in boot process, DMA memory to memory copying is working fine, but when I try to use same VC based memory allocations, it works until I try un-allocating the memory I'd allocated.  Then, it hangs.  It appears that my allocation then de-allocation is interfering with the other memory functions in embox after those drivers start up.

Hmm... any pointers on this?  Am I going to have to change the memory device implementation to use the VC memory allocation functions so that I can get the bus address of every memory handle?

Below is the code in my test case to give an idea of what I'm talking about.

'''
#include <drivers/bcm283x_dma.h>
#include <unistd.h>
#include <string.h>
#include <stdio.h>
#include <stdlib.h>
#include <mem/sysmalloc.h>

#include <drivers/mailbox/bcm2835_mailbox_property.h>

#define MEM_SIZE 1024
#define CPY_BLK 256
#define DMA_CHAN 5

void fill_test(uint32_t *memLoc, uint32_t bytes) {
    uint32_t pattern = 0xDEADBEEF;
    for(uint32_t i = 0 ; i < bytes / ( 2 * sizeof(uint32_t) ) ; i+= 2 * sizeof(uint32_t) ) {
        memLoc[i] = pattern;
        memLoc[i+1] = i;
    }
}

// return zero if equil
// memcmp(const void *_dst, const void *_src, size_t n)

// Allocate and lock bus addressed memory
// populate data DE AD BE EF 00 00 00 00 .. incrementing counter to fill
// Init DMA for copy to new memory location
// compare memories

// DMA memory-to-memory test
int dma_test_mem_transfer(void *mem)
{
    Dma_conbk *cbp = mem;
    char *srce = (char *)((uint32_t)cbp+sizeof(Dma_conbk));
    char *dest = srce + CPY_BLK;
 
    memset((void *)cbp, 0, MEM_SIZE);   // clear all working memory
    fill_test((uint32_t *)srce,CPY_BLK);    // set values to copy
    cbp->ti = DMA_TI_SRC_INC | DMA_TI_DEST_INC;
    cbp->source_ad = (uint32_t)srce;
    cbp->dest_ad = (uint32_t)dest;
    cbp->txfr_len = CPY_BLK;
    bcm283x_dma_transfer_conbk(DMA_CHAN, cbp);
    usleep(10);

    while(bcm283x_dma_in_progress(DMA_CHAN)) {
#if 1
        bcm283x_dma_disp(DMA_CHAN);
#endif
    }
    return memcmp(srce,dest,CPY_BLK);
}

// Returns pointer to handle, address of bus address memory is in resp->data.value_32
//
void *malloc_uncached(size_t bytes) {
    bcm2835_mailbox_property_t *resp;
    
    bcm2835_property_init();
    bcm2835_property_add_tag( TAG_ALLOCATE_MEMORY,  bytes, 0, BCM2835_MEM_FLAG_NORMAL );
    bcm2835_property_process();
    resp = bcm2835_property_get( TAG_ALLOCATE_MEMORY );
    void *handle = (void *)resp->data.value_32;
    printf("Alloc handle: %x\n",resp->data.value_32);

    return handle;
}

void *mem_busaddr(void *handle) {
    bcm2835_mailbox_property_t *resp;

    bcm2835_property_init();
    bcm2835_property_add_tag( TAG_LOCK_MEMORY,  handle );
    bcm2835_property_process();
    resp = bcm2835_property_get( TAG_LOCK_MEMORY );
    printf("Locked handle %x (bus address): %x\n",(uint32_t)handle, resp->data.value_32);
    return (void *)resp->data.value_32;
}

int dealloc_uncached(void *handle) {
    bcm2835_mailbox_property_t *resp;
/*
    bcm2835_property_init();
    bcm2835_property_add_tag( TAG_UNLOCK_MEMORY,  handle );
    bcm2835_property_process();
    resp = bcm2835_property_get( TAG_UNLOCK_MEMORY );
    printf("Unlocked (handle: %x) status: %x\n",(uint32_t)handle,resp->data.value_32);
*/
    bcm2835_property_init();
    bcm2835_property_add_tag( TAG_RELEASE_MEMORY,  handle );
    bcm2835_property_process();
    resp = bcm2835_property_get( TAG_RELEASE_MEMORY );
    printf("Release (handle: %x) status: %x\n",(uint32_t)handle,resp->data.value_32);
    return 0;
}

int main(int argc, char **argv) {
    int res = 0;
//    void *mem_bus_handle = 0;

    bcm283x_dma_config(DMA_CHAN);
//    void *mem_handle = malloc(MEM_SIZE);
//    void *mem_handle = malloc_uncached(MEM_SIZE);
    void *mem_handle = sysmalloc(MEM_SIZE);
//    if(mem_handle != 0x00) mem_bus_handle = mem_busaddr(mem_handle);
    if(mem_handle != 0x00) res = dma_test_mem_transfer(mem_handle);
    if(res == 0) {
        printf(" * Sucsess * ");
    } else {
        printf(" * Fail * ");
    }
    sysfree(mem_handle);
//    dealloc_uncached(mem_handle);
	return 0;
}

'''

anton-bondarev, 2021-07-11:
@kpishere I'm not quite sure what do you mean.

I think you can use sysmalloc/sysfree if you want to allocate memory for all processes and malloc/free for using a single process. Of course, you should remember about alignment and properties of the allocated memory region
As for special allocations as I see in your example, you  can use them if it needs

kpishere, 2021-07-11:
Thanks.  Part of what was throwing me off was what VC/MMU was.  I is called System MMU by other documents.  With this SOC, we think of the ARM cpu as the primary one but really, the VideoCore is the primary one (the one that boots), it allocates memory for the CPU and then from there you get the normal Virtual MMU behaviours.  From the following reddit, got reference to this so, all this reading just to validate if I can hard-code mapping from 0x7E to 0x20. :)  

Arm® System Memory Management Unit Architecture Specification

[`Reddit thread`](https://www.reddit.com/r/raspberry_pi/comments/4aonbh/why_are_there_two_mmus_on_the_bcm2835/)

anton-bondarev, 2021-07-12:
> Thanks. Part of what was throwing me off was what VC/MMU was. I is called System MMU by other documents. With this SOC, we think of the ARM cpu as the primary one but really, the VideoCore is the [...]

Hm, Maybe it is better to use PERIPH_MEMORY_DEFINE in this case (static memory allocation) if the address is unknown in static you may use dma_alloc_coherent() it allocates system memory and marks it as NOCACHED. You can allocate memory one time and not free it at all, I think

kpishere, 2021-07-12:
Making progress, got the most basic test working.  Was close but got to be very careful with "BUS Addresses" vs. Physical addresses that the ARM CPU can actually access.

Here is my most basic test that is working. This tutorial repo helped out hugely to sort it out (although, they are misusing term virtual in this case, what the cpu sees is 'physical' and what the user program sees is virtual according to BCM2835 peripherals document) https://github.com/fandahao17/Raspberry-Pi-DMA-Tutorial

```
cmd\dma.c:

#include <drivers/bcm283x_dma.h>
#include <unistd.h>
#include <string.h>
#include <stdio.h>
#include <stdlib.h>
#include <mem/sysmalloc.h>
#include <assert.h>

#define MEM_SIZE 1024
#define CPY_BLK 256
#define DMA_CHAN 5

void fill_test(uint32_t *memLoc, uint32_t bytes) {
    uint32_t pattern = 0xDEADBEEF;
    for(uint32_t i = 0 ; i < bytes / sizeof(uint32_t) ; i+= 2 ) {
        memLoc[i] = pattern;
        memLoc[i+1] = i;
    }
}

// DMA memory-to-memory test
int dma_test_mem_transfer(Dma_mem_handle *mem)
{
    Dma_conbk *cbp = mem->physical_addr;
    // Pointers for DMA 
    uint32_t srce = ((uint32_t)(mem->bus_addr)+sizeof(Dma_conbk));
    uint32_t dest = srce + CPY_BLK;
 
    // ARM uses physical addr pointer to same memory
    memset((void *)cbp, 0, MEM_SIZE);   // clear all working memory
    fill_test(DMA_BUS_TO_PHYS(srce),CPY_BLK);    // set values to copy

    // DMA will use bus address values, were writing to memory with the physical address pointer
    cbp->ti = DMA_TI_SRC_INC | DMA_TI_DEST_INC;
    cbp->source_ad = srce;
    cbp->dest_ad = dest;
    cbp->txfr_len = CPY_BLK;
    bcm283x_dma_transfer_conbk(DMA_CHAN, cbp);

    while(bcm283x_dma_in_progress(DMA_CHAN)) {
#if 1
        bcm283x_dma_disp(DMA_CHAN);
#endif
    }

    return memcmp(DMA_BUS_TO_PHYS(srce),DMA_BUS_TO_PHYS(dest),CPY_BLK);
}

int main(int argc, char **argv) {
    int res = -1;

    bcm283x_dma_config(DMA_CHAN);
    Dma_mem_handle *mem_handle = dma_malloc(MEM_SIZE);
    if(mem_handle != 0x00) res = dma_test_mem_transfer(mem_handle);
    if(res == 0) {
        printf("\n * Sucsess *\n");
    } else {
        printf("\n * Fail *\n");
    }
    dma_free(mem_handle);
	return 0;
}

```