embox/embox #2492: date -s 202108311902.40 set wrong time (2021-10-01 19:02:40)

anton-bondarev, 2021-08-31:
Parsing time string is ok ('struct tm date' contains a correct time)
but after
```

	cur_time = mktime(&date);
	tv.tv_sec = cur_time;

	settimeofday(&tv, NULL);
```
the month is 10 instead of 8 and the day 1 instead of 31

KlyukinSA, 2023-07-24:
@anton-bondarev  i see two rtc devices (find `RTC_DEVICE_DEF`). which of two was used here?

anton-bondarev, 2023-07-25:
@KlyukinSA I don't think that this issue depends on RTC_DEVICE. You can reproduce it on any qemu template.
As for RTC_DEV. I see RTC only in aarch64/qemu pl031. Dialog RTC is used on only dialog boards.

KlyukinSA, 2023-07-25:
> Parsing time string is ok ('struct tm date' contains a correct time)

@anton-bondarev  are you sure?

[manpage](https://linux.die.net/man/3/mktime):
> [struct tm::]tm_mon
> The number of months since January, in the range 0 to 11. 

but src/cmds/sys/date.c:set_date
```
end -= 2;
sscanf(end, "%d", &date.tm_mon);
*end = '\0';
```
and not
```
end -= 2;
sscanf(end, "%d", &date.tm_mon);
date.tm_mon -= 1;
*end = '\0';
```

anton-bondarev, 2023-07-26:
@KlyukinSA Thank you!

> @anton-bondarev are you sure?

No, I'm not sure.
But I checked 'date' cmd on x86/qemu
```
root@embox:/#date -s 202108311902.40

2021-10-01 19:02:39
```

So I think you are right about src/cmds/sys/date.c:set_date

```
end -= 2;
sscanf(end, "%d", &date.tm_mon);
date.tm_mon -= 1;
*end = '\0';
```

could you correct this place and check the result?

KlyukinSA, 2023-07-26:
@anton-bondarev 
```
root@embox:/#date -s 202108311902.40

2021-08-31 19:02:40
```

anton-bondarev, 2023-07-26:
@KlyukinSA Thank you very much!

closed 2023-07-26 by anton-bondarev