embox/embox #2502: Compile error with method __atomic_compare_exchange_4

kpishere, 2021-09-19:
I get compile errors with build '1314d2fbaad2b76a1e18607ddcb7453c5b38b2a1' when building target 

```
make confload-platform/raspberry_pi/rpi0-hw; make
```

I can revert to build 'd6cc929956b8d7f12860bde360f5967d0f16807e' and it builds fine.

The problem appears to be in change '56aae685699fc9954f710958690101ade8d85a05'

anton-bondarev, 2021-09-19:
@kpishere Thanks for the report
Could you please attach the build log (the error description) and your compiler version?
I've checked raspberry_pi/rpi0-hw build it is OK in my environment (I used current embox master)

kpishere, 2021-09-19:
Sure.  Ok, here is the update I'm making (it is compiling before this): 

```
From https://github.com/embox/embox
 * branch                master     -> FETCH_HEAD
Updating d6cc92995..1314d2fba
Fast-forward
 src/compat/atomic/pseudo_atomic/pseudo_atomic.c |  20 +++
 src/compat/cxx/include/cstring                  |   1 +
 src/include/kernel/thread/sync/cond.h           |   2 +-
 third-party/lib/boost/Makefile                  |   4 +-
 third-party/qpid/Makefile                       |  31 +++--
 third-party/qpid/Mybuild                        |   5 +-
 third-party/qpid/patch.txt                      | 507 ++++-------------------------------------------------------------------
 third-party/qpid/qpid_embox_compat.h            |  41 ------
 8 files changed, 70 insertions(+), 541 deletions(-)
```

With a fresh build, the following are output error messages.

```
make confload-platform/raspberry_pi/rpi0-hw; make > ../log.txt
Config complete
third-party/qpid/Mybuild:34:21: error: Couldn't resolve reference to ModuleType 'embox.lib.stl'.
third-party/qpid/Mybuild:8:15: error: Couldn't resolve reference to Type 'embox.lib.stl'.
mk/image2.mk:157: warning: undefined variable `file >>/Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/arch/arm/armlib/context.o.objcopy_options.txt, --rename-section .bss=.bss.module.embox__arch__arm__armlib__context  --rename-section .data=.data.module.embox__arch__arm__armlib__context  --rename-section .rodata=.rodata.module.embox__arch__arm__armlib__context  --rename-section .text=.text.module.embox__arch__arm__armlib__context'
/Users/kevinpeck/Documents/jessie/embox/mk/objcopy_helper.sh: line 10: /Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/arch/arm/armlib/context.o.objcopy_options.txt: No such file or directory
mk/image2.mk:157: warning: undefined variable `file >>/Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/arch/arm/mmu_small_page.o.objcopy_options.txt, --rename-section .bss=.bss.module.embox__arch__arm__mmu_small_page  --rename-section .data=.data.module.embox__arch__arm__mmu_small_page  --rename-section .rodata=.rodata.module.embox__arch__arm__mmu_small_page  --rename-section .text=.text.module.embox__arch__arm__mmu_small_page'
/Users/kevinpeck/Documents/jessie/embox/mk/objcopy_helper.sh: line 10: /Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/arch/arm/mmu_small_page.o.objcopy_options.txt: No such file or directory
mk/image2.mk:157: warning: undefined variable `file >>/Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/kernel/Kernel.o.objcopy_options.txt, --rename-section .bss=.bss.module.embox__kernel__Kernel  --rename-section .data=.data.module.embox__kernel__Kernel  --rename-section .rodata=.rodata.module.embox__kernel__Kernel  --rename-section .text=.text.module.embox__kernel__Kernel'
/Users/kevinpeck/Documents/jessie/embox/mk/objcopy_helper.sh: line 10: /Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/kernel/Kernel.o.objcopy_options.txt: No such file or directory
mk/image2.mk:157: warning: undefined variable `file >>/Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/kernel/task/resource/atexit.o.objcopy_options.txt, --rename-section .bss=.bss.module.embox__kernel__task__resource__atexit  --rename-section .data=.data.module.embox__kernel__task__resource__atexit  --rename-section .rodata=.rodata.module.embox__kernel__task__resource__atexit  --rename-section .text=.text.module.embox__kernel__task__resource__atexit'
/Users/kevinpeck/Documents/jessie/embox/mk/objcopy_helper.sh: line 10: /Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/kernel/task/resource/atexit.o.objcopy_options.txt: No such file or directory
mk/image2.mk:157: warning: undefined variable `file >>/Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/lib/debug/whereami.o.objcopy_options.txt, --rename-section .bss=.bss.module.embox__lib__debug__whereami  --rename-section .data=.data.module.embox__lib__debug__whereami  --rename-section .rodata=.rodata.module.embox__lib__debug__whereami  --rename-section .text=.text.module.embox__lib__debug__whereami'
/Users/kevinpeck/Documents/jessie/embox/mk/objcopy_helper.sh: line 10: /Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/lib/debug/whereami.o.objcopy_options.txt: No such file or directory
mk/image2.mk:157: warning: undefined variable `file >>/Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/arch/arm/head.o.objcopy_options.txt, --rename-section .bss=.bss.module.embox__arch__arm__head  --rename-section .data=.data.module.embox__arch__arm__head  --rename-section .text.bootcode=.text.bootcode.module.embox__arch__arm__head  --rename-section .text=.text.module.embox__arch__arm__head'
/Users/kevinpeck/Documents/jessie/embox/mk/objcopy_helper.sh: line 10: /Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/arch/arm/head.o.objcopy_options.txt: No such file or directory
mk/image2.mk:157: warning: undefined variable `file >>/Users/kevinpeck/Documents/jessie/embox/build/base/obj/module/embox/arch/arm/stackframe.o.objcopy_options.txt, --rename-section .bss=.bss.module.embox__arch__arm__stackframe  --rename-section .data=.data.module.embox__arch__arm__stackframe  --rename-section .text=.text.module.embox__arch__arm__stackframe'
[... cut ...] compat/atomic/pseudo_atomic/pseudo_atomic.o /Users/kevinpeck/Documents/jessie/embox/src/compat/atomic/pseudo_atomic/pseudo_atomic.c
```

My build pipeline version is this: 

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

anton-bondarev, 2021-09-19:
@kpishere Thank you
Could you try #2504 I hope it must fix the problem

kpishere, 2021-09-19:
@anton-bondarev Yep, that fixed it.  Thanks!

closed 2021-09-19 by kpishere