embox/embox #2530: thread_join() does not appear to work

kpishere, 2021-12-26:
Test environment is with a raspberry pi zero, only 16 threads are allowed in this configuration.  Thread priority test has been modified to join threads rather than create them detached.  The reason being that the next test failed due to not being able to create a thread.  It is suspected that threads aren't completing.

Indeed, with the following change to the thread_priority test, none of the threads join.

There appears to be a bug with thread completion. 

There is a PR with this code change for this ticket

anton-bondarev, 2021-12-27:
@kpishere Thanks
Could you give more details about the problem?
I see there are two different issues: one is RaPI is not able to create more than 16 threads, other is the problem with joinable threads on PaPi. Whether is it right?
It seems the first issue can be solved just pointed the option in mods.conf
```
include embox.test.kernel.thread.thread_priority_test(threads_quantity=16)
```
The second issue makes us think about the correctness of thread implementation on ARMv7

kpishere, 2021-12-27:
Yes, will reduce threads the way you show. First issue of importance is the joining for this ticket. Second could be as simple as stack size config but working with 16 for now is a start. will switch printf to logging statements.

kpishere, 2021-12-27:
So, just cleaning up the test, setting some logs, and showing the output below.

Could it possibly be breaking in the last three lines of this output?  Where the wakeup thread tries to jump to #1c00b24? Early on, it looks like the wakeup thread should be #1c00aa0. 

```
Interrupt controller: raspi_intc

Embox kernel start
        unit: initializing embox.kernel.task.task_resource: done
        unit: initializing embox.kernel.task.task_table: done
        unit: initializing embox.kernel.task.kernel_task: done
        unit: initializing embox.driver.periph_memory_mmap: done
        unit: initializing embox.arch.arm.mmu_small_page: set ctx 0xb00000
done 
        unit: initializing embox.mem.phymem:
                        start=0x0a5e6000 end=0x20008000 size=362946560
                done
        unit: initializing embox.driver.dma.pl330_dma: done
runlevel: init level is 0
        unit: initializing embox.mem.static_heap: done
        unit: initializing embox.profiler.tracing: done
        unit: initializing embox.kernel.sched.sched: debug: boot_thread_create: boot_schedee = 0x1c52000
debug: sched_start_init: boot_thread = 0x0x01c52000
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x1c52000, next: 0x1c52000
done 
        unit: initializing embox.driver.gpio.bcm283x_gpio: done
        test: running embox.test.math.math_test .. done
runlevel: init level is 1
        unit: initializing embox.fs.buffer_cache: done
        unit: initializing embox.driver.spi.core: done
        unit: initializing raspberry_pi.driver.jtag:  Opt 0 done
        unit: initializing embox.driver.flash.flash_nofs: done
        unit: initializing embox.driver.video.raspi_video: done
        unit: initializing embox.driver.serial.pl011: done
        unit: initializing embox.driver.spi.bcm283x_spi0: done
        unit: initializing embox.driver.tty.serial: debug: __schedule: prev: 0x1c52000, next: 0x1c52000
done 
        unit: initializing embox.driver.clock.raspi_systick: done
        test: running embox.test.kernel.thread.thread_join_test debug: __test_case_at_line_32:
Create threads:
debug: __test_case_at_line_32: .0!
debug: __sched_wakeup: schedee #9da5620
debug: __test_case_at_line_32: .1!
debug: __sched_wakeup: schedee #9dc5620
debug: __test_case_at_line_32: .2!
debug: __sched_wakeup: schedee #9de5620
debug: __test_case_at_line_32: .3!
debug: __sched_wakeup: schedee #9e05620
debug: __test_case_at_line_32: .4!
debug: __sched_wakeup: schedee #9e25620
debug: __test_case_at_line_32: .5!
debug: __sched_wakeup: schedee #9e45620
debug: __test_case_at_line_32: .6!
debug: __sched_wakeup: schedee #9e65620
debug: __test_case_at_line_32: .7!
debug: __sched_wakeup: schedee #9e85620
debug: __test_case_at_line_32: .8!
debug: __sched_wakeup: schedee #9ea5620
debug: __test_case_at_line_32: .9!
debug: __sched_wakeup: schedee #9ec5620
debug: __test_case_at_line_32: .10!
debug: __sched_wakeup: schedee #9ee5620
debug: __test_case_at_line_32: .11!
debug: __sched_wakeup: schedee #9f05620
debug: __test_case_at_line_32: .12!
debug: __sched_wakeup: schedee #9f25620
debug: __test_case_at_line_32: .13!
debug: __sched_wakeup: schedee #9f45620
debug: __test_case_at_line_32: .14!
[... 110 lines cut ...]
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x9da5620
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x9da5620
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x9da5620
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00aa0
debug: __schedule: prev: 0x9da5620, next: 0x9da5620
debug: __schedule: prev: 0x9da5620, next: 0x1c00aa0
debug: __sched_wakeup: schedee #1c00b24
debug: __sched_wakeup: schedee #1c00b24
debug: __sched_wakeup: schedee #1c00b24
```

kpishere, 2021-12-28:
Hmm well #1c00b24 is a vector to clock_handler_lt.  That vector is 0x9da5620 which is __pool_storagethread_pool.  The wakeup process at the beginning, #1c00aa0 is to idle() function.

Well, this is curious.  I'd set the $pc to start at main() again and it caused an assertion failure and listed all the processes and their current state.  It is .. 

```
Last login: Mon Dec 27 10:22:58 on ttys003
Mac-mini:embox kevinpeck$ !scr
-bash: !scr: event not found
Mac-mini:embox kevinpeck$ screen /dev/tty.
tty.Bluetooth-Incoming-Port  tty.usbmodemHIDP1            
Mac-mini:embox kevinpeck$ screen /dev/tty.w
tty.wchusbserial1410
Mac-mini:embox kevinpeck$ screen /dev/tty.wchusbserial1410 115200

  ______
 |  ____|                                            __          __
 | |___  _ __ ___            ____  ____  ____  _____/ /   _____ / /
 |  ___|| '_ ` _ \          / __ \/ __ \/ __ \/ ___/ /   |_____| |
 | |____| | | | | |_ _ _   / /_/ / /_/ / /_/ (__  )_/    |_____| |
 |______|_| |_| |_(_|_|_)  \____/\____/ .___/____(_)           | |
                                     /_/                        \_\
 ASSERTION FAILED on CPU 0
        at /Users/kevinpeck/Documents/jessie/embox/src/kernel/task/multi/multi.c:350
        in function task_start_exit

task_self() != task_kernel_task()

 --   00001000 * A R    thread 1  task 1 --------------------------------------

 24 0x00076a20 <__assertion_handle_failure+0x130> ...at/libc/assert/assert.c:45
 23 0x000143a8 <task_start_exit+0x3c>        /src/kernel/task/multi/multi.c:348
 22 0x0007a18c <_exit+0x34>               /src/compat/posix/proc/exit/exit.c:18
 21 0x0007a228 <logging_raw+0x0>                         /src/util/logging.c:22
 20 0x000565b4 <task_resource+0x0>       /src/include/kernel/task/resource.h:63
 19 0x0000f1f8 <diag_putc+0xc4>                     /src/drivers/diag/diag.c:67
 18 0x0004054c <sched_wakeup+0x58>                /src/kernel/sched/sched.c:367
 17 0x0003f15c <lthread_launch+0x44>           /src/kernel/lthread/lthread.c:89
 16 0x00042f0c <jiffies_update+0x78>           /src/kernel/time/clock_tick.c:34
 15 0x00042e74 <clock_tick_handler+0x30>       /src/kernel/time/clock_tick.c:26
 14 0x0000d6a4 <clock_handler+0x2c> ...s/clock/raspi_systick/raspi_systick.c:66
 13 0x00012680 <irq_dispatch+0x164>                   /src/kernel/irq/irq.c:155
 12 0x0000c604 <__raspi__dispatch_bank+0x60> ...nterrupt/raspi_intc/raspi.c:113
 11 0x0000c674 <interrupt_handle+0x54> ...vers/interrupt/raspi_intc/raspi.c:123
 10 0x000398e8 <irq_handler+0x58>    /src/arch/arm/armlib/exceptions/entry.S:18
  9 0x0003eb00 <ipl_init+0x20> ...clude/../../src/arch/arm/armlib/ipl_impl.h:20
  8 0x0003ef04 <lthread_do_process+0x68>       /src/kernel/lthread/lthread.c:29
  7 0x0003ef8c <lthread_do_process+0xf0>       /src/kernel/lthread/lthread.c:29
  6 0x00040858 <__schedule+0x1bc>                 /src/kernel/sched/sched.c:417
  5 0x000408f8 <sched_preempt+0x18>               /src/kernel/sched/sched.c:501
  4 0x000128b8 <critical_dispatch_pending+0x84>       /src/kernel/critical.c:21
  3 0x00015cd8 <sched_unlock+0x18>    /src/include/kernel/sched/sched_lock.h:34
  2 0x00015d14 <thread_ack_switched+0x34> /src/kernel/thread/thread_switch.c:21
  1 0x00014eb4 <thread_trampoline+0x60>            /src/kernel/thread/core.c:69

run 0x00031464 <thread_run+0x0>  /src/tests/kernel/thread/thread_join_test.c:27

--   00002000       W  thread 0  task 1 --------------------------------------

 64 0x00015dc8 <thread_context_switch+0x54> ...kernel/thread/thread_switch.c:36
 63 0x00040858 <__schedule+0x1bc>                 /src/kernel/sched/sched.c:417
 62 0x000408d4 <schedule+0x18>                    /src/kernel/sched/sched.c:494
 61 0x00049dec <sched_wait+0x10>      /src/kernel/thread/thread_sched_wait.c:35
 60 0x0004a020 <sched_wait_timeout+0x38> ...ernel/thread/thread_sched_wait.c:67
 59 0x00015860 <thread_join+0x130>                /src/kernel/thread/core.c:336
 58 0x000316b0 <__test_case_at_line_32+0x220> ...l/thread/thread_join_test.c:32
 57 0x00053dec <test_run+0x64>                    /src/framework/test/run.c:120
 56 0x00053d0c <test_case_run+0x74>                /src/framework/test/run.c:97
 55 0x00053bc8 <test_suite_run+0xdc>               /src/framework/test/run.c:56
 54 0x00053744 <test_mod_enable+0x4c>         /src/framework/test/registry.c:31
 53 0x0003cc80 <mod_enable+0xbc>                  /src/framework/mod/core.c:172
[... 210 lines cut ...]

 --   00001000       W  thread 3  task 1 --------------------------------------

  7 0x00015dc8 <thread_context_switch+0x54> ...kernel/thread/thread_switch.c:36
  6 0x00040858 <__schedule+0x1bc>                 /src/kernel/sched/sched.c:417
  5 0x000408d4 <schedule+0x18>                    /src/kernel/sched/sched.c:494
  4 0x0004a09c <sched_wait_timeout+0xb4> ...ernel/thread/thread_sched_wait.c:67
  3 0x00012b0c <ksleep+0x88>                 /src/kernel/time/timer/ksleep.c:16
  2 0x00031480 <thread_run+0x1c> /src/tests/kernel/thread/thread_join_test.c:27
  1 0x00014ef8 <thread_trampoline+0xa4>            /src/kernel/thread/core.c:69

run 0x00031464 <thread_run+0x0>  /src/tests/kernel/thread/thread_join_test.c:27

 --   00001000       W  thread 2  task 1 --------------------------------------

  7 0x00015dc8 <thread_context_switch+0x54> ...kernel/thread/thread_switch.c:36
  6 0x00040858 <__schedule+0x1bc>                 /src/kernel/sched/sched.c:417
  5 0x000408d4 <schedule+0x18>                    /src/kernel/sched/sched.c:494
  4 0x0004a09c <sched_wait_timeout+0xb4> ...ernel/thread/thread_sched_wait.c:67
  3 0x00012b0c <ksleep+0x88>                 /src/kernel/time/timer/ksleep.c:16
  2 0x00031480 <thread_run+0x1c> /src/tests/kernel/thread/thread_join_test.c:27
  1 0x00014ef8 <thread_trampoline+0xa4>            /src/kernel/thread/core.c:69

run 0x00031464 <thread_run+0x0>  /src/tests/kernel/thread/thread_join_test.c:27
```

anton-bondarev, 2021-12-28:
> Could it possibly be breaking in the last three lines of this output? Where the wakeup thread tries to jump to #1c00b24? Early on, it looks like the wakeup thread should be #1c00aa0.

Hm, It looks like that #1c00b24 is light_thread, it seems it works correctly

anton-bondarev, 2021-12-28:
> Well, this is curious. I'd set the $pc to start at main() again and it caused an assertion failure and listed all the processes and their current state. It is ..

How had you set the $PC? you call the test suite again with 'test' command?

when assert() is failed all threads stack traces is printed, It is normal behaviour

kpishere, 2021-12-28:
@anton-bondarev   Yes, normal behaviour.  But with my limited knowledge, it was was an Oh, that is what I want to see kind of moment.  I know running from main() again makes no sense but the side effect was useful.

Anyhow, this ticket can likely be close, I'd managed to make the test complete as expected.

The tests here execute and complete as expected.  Now, it is the stack protect test that is failing.  Actually, that is what it was before but I'd wanted to be sure I didn't hit the limit of open threads before focusing on that.  I'll explore it more before posting another ticket, will only do If I get stuck or there is in fact a change needed.

In any case, there is another test case in this branch now if you'd like to include it.

```
Embox kernel start
        unit: initializing embox.kernel.task.task_resource: done
        unit: initializing embox.kernel.task.task_table: done
        unit: initializing embox.kernel.task.kernel_task: done
        unit: initializing embox.driver.periph_memory_mmap: done
        unit: initializing embox.arch.arm.mmu_small_page: set ctx 0xb00000
done
        unit: initializing embox.mem.phymem:
                        start=0x0a5e6000 end=0x20008000 size=362946560
                done
        unit: initializing embox.driver.dma.pl330_dma: done
runlevel: init level is 0
        unit: initializing embox.mem.static_heap: done
        unit: initializing embox.profiler.tracing: done
        unit: initializing embox.kernel.sched.sched: done
        unit: initializing embox.driver.gpio.bcm283x_gpio: done
        test: running embox.test.math.math_test .. done
runlevel: init level is 1
        unit: initializing embox.fs.buffer_cache: done
        unit: initializing embox.driver.spi.core: done
        unit: initializing raspberry_pi.driver.jtag:  Opt 0 done
        unit: initializing embox.driver.flash.flash_nofs: done
        unit: initializing embox.driver.video.raspi_video: done
        unit: initializing embox.driver.serial.pl011: done
        unit: initializing embox.driver.spi.bcm283x_spi0: done
        unit: initializing embox.driver.tty.serial: done
        unit: initializing embox.driver.clock.raspi_systick: done
        test: running embox.test.kernel.thread.thread_join_test debug: __test_case_at_line_36: Create thread 0
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
debug: thread_exit: thread_exit.'after joined' joining:0 ret:0
debug: thread_exit: thread_exit.'after delete' state:2
debug: __test_case_at_line_36: Create thread 1
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
debug: thread_exit: thread_exit.'after joined' joining:0 ret:0
debug: thread_exit: thread_exit.'after delete' state:2
debug: __test_case_at_line_36: Create thread 2
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
debug: thread_exit: thread_exit.'after joined' joining:0 ret:0
debug: thread_exit: thread_exit.'after delete' state:2
debug: __test_case_at_line_36: Create thread 3
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
debug: thread_exit: thread_exit.'after joined' joining:0 ret:0
debug: thread_exit: thread_exit.'after delete' state:2
debug: __test_case_at_line_36: Create thread 2
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
debug: thread_exit: thread_exit.'after joined' joining:0 ret:0
debug: thread_exit: thread_exit.'after delete' state:2
debug: __test_case_at_line_36: Create thread 3
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
debug: thread_exit: thread_exit.'after joined' joining:0 ret:0
debug: thread_exit: thread_exit.'after delete' state:2
debug: __test_case_at_line_36: Create thread 4
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
[... 98 lines cut ...]
debug: thread_exit: thread_exit.'after delete' state:2
debug: thread_delete: thread_delete()
. done
        test: running embox.test.kernel.thread.thread_priority_test . done
        test: running embox.test.kernel.thread.stack_protect_test set ctx 0xb00000

  ______
 |  ____|                                            __          __
 | |___  _ __ ___            ____  ____  ____  _____/ /   _____ / /
 |  ___|| '_ ` _ \          / __ \/ __ \/ __ \/ ___/ /   |_____| |
 | |____| | | | | |_ _ _   / /_/ / /_/ / /_/ (__  )_/    |_____| |
 |______|_| |_| |_(_|_|_)  \____/\____/ .___/____(_)           | |
                                     /_/                        \_\
 ASSERTION FAILED on CPU 0
        at /Users/kevinpeck/Documents/jessie/embox/src/tests/kernel/thread/stack_protect_test.c:50
        in function __test_case_at_line_42

!pthread_create(&thread, NULL, test_func, NULL)
Data abort exception!
regs:
00050078 01c71c54 db6f3faf db6f3fab
000f4034 00000000 00000000 00000000
00000000 01c51060 bdfa0b4b 01c71c54
01c71c58 0000b218
```

kpishere, 2021-12-28:
@anton-bondarev Actually, I'd just run the stack protect test without the other thread tests and it was able to create the thread this time.  So, it looks like there may be a bug?  Something thinks that the threads still exist even though they appear to be exited correctly above.

```
Interrupt controller: raspi_intc

Embox kernel start
        unit: initializing embox.kernel.task.task_resource: done
        unit: initializing embox.kernel.task.task_table: done
        unit: initializing embox.kernel.task.kernel_task: done
        unit: initializing embox.driver.periph_memory_mmap: done
        unit: initializing embox.arch.arm.mmu_small_page: set ctx 0xb00000
done
        unit: initializing embox.mem.phymem: 
                        start=0x0a5e6000 end=0x20008000 size=362946560
                done
        unit: initializing embox.driver.dma.pl330_dma: done
runlevel: init level is 0
        unit: initializing embox.mem.static_heap: done
        unit: initializing embox.profiler.tracing: done
        unit: initializing embox.kernel.sched.sched: done
        unit: initializing embox.driver.gpio.bcm283x_gpio: done
        test: running embox.test.math.math_test .. done
runlevel: init level is 1
        unit: initializing embox.fs.buffer_cache: done
        unit: initializing embox.driver.spi.core: done
        unit: initializing raspberry_pi.driver.jtag:  Opt 0 done
        unit: initializing embox.driver.flash.flash_nofs: done
        unit: initializing embox.driver.video.raspi_video: done
        unit: initializing embox.driver.serial.pl011: done
        unit: initializing embox.driver.spi.bcm283x_spi0: done
        unit: initializing embox.driver.tty.serial: done
        unit: initializing embox.driver.clock.raspi_systick: done
        test: running embox.test.kernel.thread.stack_protect_test set ctx 0xb00000
debug: thread_exit: thread_exit()
debug: thread_exit: thread_exit.sched_lock()
debug: thread_exit: thread_exit.'after joined' joining:0 ret:0
debug: thread_exit: thread_exit.'after delete' state:2
debug: thread_delete: thread_delete()

        failure at /Users/kevinpeck/Documents/jessie/embox/src/tests/kernel/thread/stack_protect_test.c : 54, in function __test_case_at_line_42
                test_assert_equal(exception_flag, 1)
           case at /Users/kevinpeck/Documents/jessie/embox/src/tests/kernel/thread/stack_protect_test.c : 42
                "Stack overflow causes exception"

        testing stack_protect_test (Stack protection test suite) failed
                1/1 failures
Failed to get into level 2, current level 1
Default IO device[diag]
>export PWD=/
>export HOME=/
```

anton-bondarev, 2021-12-28:
Hm, try to use `thread -s` to show all threads in the system. Maybe you have found a bug and the threads are not finished

kpishere, 2021-12-28:
Yes, OK.  I'd added back thread_priority_test.c test back and the original error appears again so will dig into that.

anton-bondarev, 2021-12-29:
> OK. I'd added back thread_priority_test.c test back and the original error appears again so will dig into that.

Ok. Thanks. I'll try to investigate the problem on another platform.

anton-bondarev, 2021-12-29:
@kpishere I have reproduced the problem.
But in my case, it means that the quantity of threads is not enough.
Please try to set the following
set tested thread as 16
`@Runlevel(1) include embox.test.kernel.thread.thread_priority_test(threads_quantity=16)`
and increase pool threads up to 32
` include embox.kernel.thread.core(thread_pool_size=32, ...)`

kpishere, 2022-01-02:
So, I've resolved issues with tests around joining and the thread priority tests.  Essentially, setting a thread priority at zero causes problems.  I've modified the start-up test to adjust thread priority by N count below the default/current thread priority.  Default value of 127, then minus one for each new thread.  That worked fine.  Added a command line version of that test, and had thread priority increment from 127, that worked fine too.  When testing with setting thread priority to 0, it appeared that the 0 thread priority would not end.  Is not getting scheduled?

There may be a fix to make there or not, what do you think?  For my purposes, I'll just make sure VC4 driver isn't setting that value in thread priority.

So, the other issue remains with the stack protection tests.  Now, it does not get a thread creation error, instead, it appears the thread exception handler isn't getting called.  Here is the output. 

````
.test: running embox.test.kernel.thread.stack_protect_test set ctx 0xb00000
debug: thread_exit: current priority 7F007F
debug: thread_exit: 'after joined' joining:0 ret:0
debug: thread_exit: 'after delete' state:2
debug: thread_join: current thread priority: 7F007F
debug: thread_delete: priority 7F007F

.failure at /Users/kevinpeck/Documents/jessie/embox/src/tests/kernel/thread/stack_protect_test.c : 54, in function __test_case_at_line_42
..test_assert_equal(exception_flag, 1)
.   case at /Users/kevinpeck/Documents/jessie/embox/src/tests/kernel/thread/stack_protect_test.c : 42
.."Stack overflow causes exception"
.
.testing stack_protect_test (Stack protection test suite) failed
..1/1 failures
Failed to get into level 2, current level 1
Default IO device[diag]
>export PWD=/
>export HOME=/
>fb_direct_access
>tish
cmd tish failed with error (-2)
````

kpishere, 2022-01-02:
@anton-bondarev I've looked at this test some, I just don't see an issue.  It looks like the interrupt vector (?) never gets called at 0x8030 for the stack overflow handler.  I'm thinking this just isn't supported on ARMV6 Raspberry Pi Zero.

From my point of view, this ticket can be closed.

anton-bondarev, 2022-01-03:
> @anton-bondarev I've looked at this test some, I just don't see an issue. It looks like the interrupt vector (?) never gets called at 0x8030 for the stack overflow handler. I'm thinking this just isn't supported on ARMV6 Raspberry Pi Zero.
> 

Yes, it seems you are right.

> From my point of view, this ticket can be closed.

I agree. However, I want to take a look at the PR (I would be glad if you suggested a fix for the zero priority threads).

kpishere, 2022-01-07:
@anton-bondarev The problem is with the function priolist_add() in the util/priolist.c code.  I don't see any tests for this code so I think I'll try writing some without relying on the scheduler to call it (a real pain that way).  It is definitely related to trying to add a priority zero item to a list with lower and upper bounds of 0 and 255.

kpishere, 2022-01-08:
@anton-bondarev Ok, confusing as doubly linked lists with a second priority list are, I'm satisfied the issue isn't there (you likely knew that).

It is something with the selection within the list.  Probably just with the priority scheduler.

This log point below is in sched.c at 

		schedee_set_current(next);
		log_debug("prev: %#x, next: %#x", prev->priority.current_priority, next->priority.current_priority);

```
# Test program starts, adding three threads, then waits to join
# 
debug: __schedule: prev: 0x7f, next: 0x7f
debug: main: Main process priority 127
debug: main: Thread create 0 7F
debug: main: Thread re-prioritize 0 0
debug: main: Thread create 1 7F
debug: main: Thread re-prioritize 1 1
debug: main: Thread create 2 7F
debug: main: Thread re-prioritize 2 2
debug: thread_join: thread joining priority: 0, joining 7F
debug: __schedule: prev: 0x7f, next: 0x7f
debug: __schedule: prev: 0x7f, next: 0x2
debug: __schedule: prev: 0x2, next: 0x1
debug: __schedule: prev: 0x1, next: 0x0
debug: __schedule: prev: 0x1, next: 0x0
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0x0
... # stuck looping here
debug: __schedule: prev: 0x0, next: 0x0
# this shows up every once in a while, the system tick?
debug: __schedule: prev: 0x0, next: 0xc8
debug: __schedule: prev: 0x0, next: 0xc8
debug: __schedule: prev: 0x0, next: 0x0
...
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0x0
# I press control-C here and it cancels the program, thread with priority 80 is the tsch? 
debug: __schedule: prev: 0x0, next: 0x80
debug: __schedule: prev: 0x0, next: 0x7f
debug: __schedule: prev: 0x7f, next: 0x7f
debug: thread_delete: priority 20002
debug: thread_delete: priority 10001
debug: thread_delete: priority 0
debug: __schedule: prev: 0x7f, next: 0x7f
debug: thread_delete: priority 7F007F
# Test program has exited here 
root@embox:/# 
root@embox:/#debug: __schedule: prev: 0x7f, next: 0x0
# And this is normal waiting in the tsch process. ... but I don't see that regular 0xc8 process appearing as above  so what was that?
debug: __schedule: prev: 0x7f, next: 0x0
debug: __schedule: prev: 0x7f, next: 0x0
debug: __schedule: prev: 0x7f, next: 0x0
```

Such a pain this.  I appreciate the encouragement, but my time is getting near the end on this one.

kpishere, 2022-01-08:
@anton-bondarev Well this has been a learning opportunity.  I'm gonna call it quits now though.  I can very clearly see that the thread with priority 0 is the idle thread "arch_idle()".  Ideally, you'd think the scheduler would round-robin a user created thread with same priority but it does not do that.  Simply increasing the minimum thread priority value to 1 does not help either.

Did some theory review and this is a classic problem with this kind of pre-emptive priority scheduler so ... not perfect but nearly so?

This is my best visualization of the problem ... I have the log_debug() message in the scheduler only printing with either of the two values have changed.

```
## Test run where with lowest priority at 1 (but macro allowing zero priority)
#
# Notice that one of the priority values is 0x0 -- scheduler appears to be using/setting this value
# this 0x0 priority thread is the arch_idle() thread
#
debug: main: Main process priority 127
debug: main: Thread create 0 7F
debug: main: Thread re-prioritize 0 1
debug: main: Thread create 1 7F
debug: main: Thread re-prioritize 1 2
debug: main: Thread create 2 7F
debug: main: Thread re-prioritize 2 3
debug: thread_join: thread joining priority: 1, joining 7F
debug: __schedule: prev: 0x7f, next: 0x3
debug: __schedule: prev: 0x3, next: 0x2
debug: __schedule: prev: 0x2, next: 0x1
debug: __schedule: prev: 0x1, next: 0x0
debug: __schedule: prev: 0x1, next: 0xc8
debug: __schedule: prev: 0x1, next: 0x3
debug: __schedule: prev: 0x3, next: 0xc8
debug: __schedule: prev: 0x3, next: 0x3
debug: __schedule: prev: 0x3, next: 0x2
debug: __schedule: prev: 0x2, next: 0x1
debug: __schedule: prev: 0x1, next: 0x7f
debug: thread_delete: priority 10001
debug: __schedule: prev: 0x7f, next: 0x7f
debug: main: Thread joined 0
debug: thread_delete: priority 20002
debug: main: Thread joined 1
debug: thread_delete: priority 30003
debug: main: Thread joined 2
debug: thread_delete: priority 7F007F
root@embox:/#debug: __schedule: prev: 0x7f, next: 0x0

## With first thread at priority zero, it looks as if idle thread runs in place of 
# user's zero priority thread
#
debug: main: Main process priority 127
debug: main: Thread create 0 7F
debug: main: Thread re-prioritize 0 0
debug: main: Thread create 1 7F
debug: main: Thread re-prioritize 1 1
debug: main: Thread create 2 7F
debug: main: Thread re-prioritize 2 2
debug: thread_join: thread joining priority: 0, joining 7F
debug: __schedule: prev: 0x7f, next: 0x2
debug: __schedule: prev: 0x2, next: 0x1
debug: __schedule: prev: 0x1, next: 0x0
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0xc8
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0xc8
.. about 40 repeats ..
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0xc8
debug: __schedule: prev: 0x0, next: 0x2
debug: __schedule: prev: 0x2, next: 0x2
debug: __schedule: prev: 0x2, next: 0x1
debug: __schedule: prev: 0x1, next: 0x0
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0xc8
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0xc8
.. repeats of above two lines until CTRL-C pressed ..
debug: __schedule: prev: 0x0, next: 0x0
debug: __schedule: prev: 0x0, next: 0x80
debug: __schedule: prev: 0x0, next: 0x7f
debug: __schedule: prev: 0x7f, next: 0x7f
debug: thread_delete: priority 20002
debug: thread_delete: priority 10001
debug: thread_delete: priority 0
debug: thread_delete: priority 7F007F
root@embox:/#
root@embox:/#
root@embox:/#debug: __schedule: prev: 0x7f, next: 0x0
```

kpishere, 2022-01-08:
BTW: I pushed the code as I'm leaving it for this branch just in case anybody is interested.

anton-bondarev, 2022-01-16:
@kpishere Thanks. I research your results and make some comments in the PR.