embox/embox #2545: Can't build with opencv

jagauthier, 2022-02-09:
Using Fedora 35, and 

`make confload-project/opencv/stm32h745i-discovery`

```
cc1plus: warning: '-Wabi' won't warn about anything [-Wabi]
cc1plus: note: '-Wabi' warns about differences from the most up-to-date ABI, which is also used by default
cc1plus: note: use e.g. '-Wabi=11' to warn about changes from GCC 7
make[10]: *** No rule to make target '../src/c++11/libc++11convenience.la', needed by 'libstdc++.la'.  Stop.
make[10]: Leaving directory '/home/jgauthier/embox/build/extbld/third_party/gcc/gcc_build/build/arm-none-eabi/libstdc++-v3/src'
make[9]: *** [Makefile:638: all-recursive] Error 1
make[9]: Leaving directory '/home/jgauthier/embox/build/extbld/third_party/gcc/gcc_build/build/arm-none-eabi/libstdc++-v3/src'
make[8]: *** [Makefile:507: all-recursive] Error 1
make[8]: Leaving directory '/home/jgauthier/embox/build/extbld/third_party/gcc/gcc_build/build/arm-none-eabi/libstdc++-v3'
make[7]: *** [Makefile:414: all] Error 2
make[7]: Leaving directory '/home/jgauthier/embox/build/extbld/third_party/gcc/gcc_build/build/arm-none-eabi/libstdc++-v3'
make[6]: *** [Makefile:10276: all-target-libstdc++-v3] Error 2
make[6]: Leaving directory '/home/jgauthier/embox/build/extbld/third_party/gcc/gcc_build/build'
make[5]: *** [Makefile:99: /home/jgauthier/embox/build/extbld/third_party/gcc/gcc_build/.builded] Error 2
make[4]: *** [/home/jgauthier/embox/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: /home/jgauthier/embox/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:30: build] Error 2
make: *** [Makefile:40: all] Error 2

```

anton-bondarev, 2022-02-10:
@jagauthier What cross-compiler version do you use?
`arm-none-eabi-gcc --version`

anton-bondarev, 2022-02-10:
@jagauthier I've fixed build the template in  #2546. You should update Embox master

jagauthier, 2022-02-12:
@anton-bondarev I wondered if that was the case.  I tried another build on Debian 10.11, which uses :
`gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (15:7-2018-q2-6)`

And I got a completely different error:
`/tmp/ccAJrqBt.s: Assembler messages:
/tmp/ccAJrqBt.s:3505: Error: selected FPU does not support instruction -- `vcvtb.f64.f16 d7,s15'`

Is 7.3.1 too old?

anton-bondarev, 2022-02-13:
> Is 7.3.1 too old?

Hmm, it is surprising for me. gcc 7.3.1 exactly works but may be failed on this template.

anton-bondarev, 2022-02-14:
@jagauthier I prefer to use the official ARM cross-toolchain from here ([https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads))

jagauthier, 2022-02-14:
@anton-bondarev Using the official ARM cross-toolchain solved the build issue on my fedora system.
Upgrading from buster to bullseye on my other system, which upgraded the compiler to 8.3.1, also worked.

So I have two functional paths now.  Thanks!

closed 2022-02-14 by jagauthier