embox/embox #2549: driver/block_dev.h: no such file or directory

zhouminglong, 2022-03-20:
make confload-project/pjsip/stm32f746g-discovery
make 

fatal error ：driver/block_dev.h: no such file or directory
include <driver/block_dev.h>

anton-bondarev, 2022-03-21:
Thanks!
To fix the problem just add `include embox.fs.driver.devfs_stub` into 'conf/mods.conf' as in #2550
You can also get repo master
Don't remember to modify 'conf/simple_pjsua_sip_account.inc`
```
#define SIP_DOMAIN     "sip.linphone.org"
#define SIP_USER       "yyyyy"
#define SIP_PASSWD     "xxx"
```

wawuwo, 2023-08-11:
Hi @anton-bondarev! Shouldn't this issue be closed? It look likes like you fixed it in #2550

anton-bondarev, 2023-08-12:
> Hi @anton-bondarev! Shouldn't this issue be closed? It look likes like you fixed it in https://github.com/embox/embox/pull/2550

@wawuwo Thanks! I've closed it

closed 2023-08-12 by anton-bondarev