embox/embox #2558: Guidance to perform minimal port (with basically just 'tish' and file system) for STM32F7508-DK

bl0wfi-sh, 2022-04-02:
Would like to get Embox running on my STM32F7508-DK board, but not sure where to start.

Tried compiling stm32f746g_discovery templates and changed lds.conf to use 64Kbyte flash available on STM32F7508-DK, but build keeps coming out too large.

`/opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region 'ROM' overflowed by 661412 bytes`

anton-bondarev, 2022-04-03:
@bl0wfi-sh Hello
I recently update wiki page for [porting to a new STM32 board](https://github.com/embox/embox/wiki/How-to-port-to-a-new-STM32-board)
As for the compiler error. You need decrise functionality. I think you do not need some features described in mods.conf web site for example. just delete following lines
```
	include embox.cmd.net.httpd(use_real_cmd=true)
	include embox.cmd.net.httpd_cgi
	include embox.service.http_admin
	include embox.service.http_admin_iface_list(is_readonly=true)
	include embox.service.http_admin_iface_html
	include embox.demo.website
```

64кB is rather restriction for all features in this templates (AUDIO, TCP/IP, SD card, Video ...)

If you want I try to prepare a template with basically just 'tish' and file system (only rootfs) for stm32f746g_discovery and we measure how much flash it needs

bl0wfi-sh, 2022-04-03:
Hi Anton!

Thank you for pointing me in the right direction with the Wiki.

It would indeed be very helpful if you could whip up a quick mods.conf with minimal functionality.
Is there a documentation area where I can read what is needed to get certain technologies up and running.

For example if I want to add support for working with my boards display, include x,y,z into mods.conf.

anton-bondarev, 2022-04-06:
@bl0wfi-sh Hi, sorry for the long response.

I've created a very restricted template for F7 with 64kB flash mods.conf in attachment and you have to add empty file into 'conf/' folder with name 'start_script.inc'
[mods.conf.txt](https://github.com/embox/embox/files/8426340/mods.conf.txt)

As for documentation/ Unfortunatly today we have rather weak manuals for system descriptions (templates). Usually we suggest to look at ready templates and ask questions.

For example to use board display you must add particular driver . for STM32f746-discovery it is
`@Runlevel(1) include embox.driver.video.stm32f7_lcd(fb_base=0x60000000)`