embox/embox #2612: Unable to Build Opencv

pkuppusamy77, 2022-10-03:
Hi, 
I tried to build  the opencv with stm32f746g-discovery template in Ubuntu 20.04 
I  am using the release version of embox-0.5.6

Its downloading gcc version 6.3 and fails to compile it
`make confload-project/opencv/stm32f746g-discovery`
`Config complete`
`make -j4`

Failed as below 
`checking for the correct version of gmp.h... no`
`configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.`
`Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify`

then I ran this command to download those sources

`embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0$ ./contrib/download_prerequisites`

and tried to compile gcc manually it fails at 

`0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/isl/include  -o ubsan.o -MT ubsan.o -MMD -MP -MF ./.deps/ubsan.TPo ../.././gcc/ubsan.c`
`../.././gcc/ubsan.c: In function ‘bool ubsan_use_new_style_p(location_t)’:`
`../.././gcc/ubsan.c:1474:23: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]`

then after correcting this It got failed again at

`g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././gcc -I../.././gcc/. -I../.././gcc/../include -I../.././gcc/../libcpp/include -I/home/prabhumk/Downloads/embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/host-x86_64-pc-linux-gnu/gmp -I/home/prabhumk/Downloads/embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/gmp -I/home/prabhumk/Downloads/embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/host-x86_64-pc-linux-gnu/mpfr -I/home/prabhumk/Downloads/embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/mpfr -I/home/prabhumk/Downloads/embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/mpc/src  -I../.././gcc/../libdecnumber -I../.././gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/../libbacktrace -I/home/prabhumk/Downloads/embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/host-x86_64-pc-linux-gnu/isl/include -I/home/prabhumk/Downloads/embox-0.5.6/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/isl/include  -o cc1-checksum.o -MT cc1-checksum.o -MMD -MP -MF ./.deps/cc1-checksum.TPo cc1-checksum.c`
`g++: error: cc1-checksum.c: No such file or directory`
`g++: fatal error: no input files`
`compilation terminated.`

Is there anyway to skip the gcc compilation and use the inbuilt gcc compiler.

Below the  version of gcc and arm-gcc compilers in the system
`gcc -v`
`Using built-in specs.`
`COLLECT_GCC=gcc`
`COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper`
`OFFLOAD_TARGET_NAMES=nvptx-none:hsa`
`OFFLOAD_TARGET_DEFAULT=1`
`Target: x86_64-linux-gnu`
`Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-1ubuntu1~20.04.1' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-Av3uEd/gcc-9-9.4.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu`
`Thread model: posix`
`gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)` 

`arm-none-eabi-gcc -v`
`Using built-in specs.`
`COLLECT_GCC=arm-none-eabi-gcc`
`COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/9.2.1/lto-wrapper`
`Target: arm-none-eabi`
`Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --enable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:9-2019-q4-0ubuntu1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip`
`Thread model: single`
`gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1)`

anton-bondarev, 2022-10-04:
@pkuppusamy77 Hi, thank you for the report
I hope I fix the trouble in #2614
You could fix it locally, just add
`include embox.driver.block_dev`
to conf/mods.conf file
then clean and rebase the project

```
make clean cacheclean
make
```

pkuppusamy77, 2022-10-05:
Hi,

I tried to build  after changing the mods.conf as suggested, but still i get the same error in gcc compilation 

I get this at the start of build
```
BUILDGEN /home/prabhumk/Downloads/embox/build
Error: modules dependency cycle:
	embox.driver.block_dev
	embox.fs.buffer_cache

```
and continues 

then it stops at

```
make -C third-party/gcc/ MAKEFLAGS=  ROOT_DIR=/home/prabhumk/Downloads/embox  EMBOX_ROOT_DIR=/home/prabhumk/Downloads/embox  CONF_DIR=/home/prabhumk/Downloads/embox/conf  TEMPLATES_DIR=/home/prabhumk/Downloads/embox/templates  SRC_DIR=/home/prabhumk/Downloads/embox/src  THIRDPARTY_DIR=/home/prabhumk/Downloads/embox/third-party  PLATFORM_DIR=/home/prabhumk/Downloads/embox/platform  PROJECT_DIR=/home/prabhumk/Downloads/embox/project  SUBPLATFORM_TEMPLATE_DIR=/home/prabhumk/Downloads/embox/templates  EXTERNAL_BUILD_DIR=/home/prabhumk/Downloads/embox/build/extbld  LOADABLE_DIR=/home/prabhumk/Downloads/embox/build/loadable  DOC_DIR=/home/prabhumk/Downloads/embox/build/doc  BIN_DIR=/home/prabhumk/Downloads/embox/build/base/bin  OBJ_DIR=/home/prabhumk/Downloads/embox/build/base/obj  GEN_DIR=/home/prabhumk/Downloads/embox/build/base/gen  SRCGEN_DIR=/home/prabhumk/Downloads/embox/build/base/gen  MKGEN_DIR=/home/prabhumk/Downloads/embox/build/base/gen  AUTOCONF_DIR=/home/prabhumk/Downloads/embox/build/base/gen  ROOTFS_DIR=/home/prabhumk/Downloads/embox/build/base/obj/rootfs  INCUDE_INSTALL_DIR=/home/prabhumk/Downloads/embox/build/base/include  USER_ROOTFS_DIR=/home/prabhumk/Downloads/embox/conf/rootfs  DOT_DIR=/home/prabhumk/Downloads/embox/build/doc  DOCS_OUT_DIR=/home/prabhumk/Downloads/embox/build/doc  CACHE_DIR=/home/prabhumk/Downloads/embox/mk/.cache BUILD_DIR=/home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build COMPILER=gcc EMBOX_ARCH='arm' EMBOX_CROSS_COMPILE='arm-none-eabi-' EMBOX_MAKEFLAGS='rR --no-print-directory --warn-undefined-variables -- MAKEFILES=' EMBOX_IMPORTED_MAKEFLAGS='' EMBOX_CFLAGS='-pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -fno-stack-protector -fno-pic -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts -Wformat -std=gnu99 -Os -g -mthumb -mlittle-endian -mcpu=cortex-m7 -ffreestanding -fomit-frame-pointer -mfpu=fpv5-sp-d16 -mfloat-abi=hard -Wno-misleading-indentation -Wno-nonnull-compare' EMBOX_CXXFLAGS='-pipe --debug-prefix-map=`pwd`= --debug-prefix-map=./= -fno-strict-aliasing -fno-common -fno-stack-protector -fno-pic -Wall -Werror -Wundef -Wno-trigraphs -Wno-char-subscripts -Wformat -I/home/prabhumk/Downloads/embox/src/compat/cxx/include -DSTDBOOL_H_ -Os -g -mthumb -mlittle-endian -mcpu=cortex-m7 -ffreestanding -fomit-frame-pointer -mfpu=fpv5-sp-d16 -mfloat-abi=hard -std=gnu++11' EMBOX_DEPS_CPPFLAGS_BEFORE='' EMBOX_DEPS_CPPFLAGS_AFTER='' EMBOX_CPPFLAGS='-U__linux__ -Ulinux -U__linux -D__EMBOX__ -D__unix -imacros /home/prabhumk/Downloads/embox/build/base/gen/config.lds.h -I/home/prabhumk/Downloads/embox/build/base/include -I/home/prabhumk/Downloads/embox/src/include -I/home/prabhumk/Downloads/embox/src/arch/arm/include -I/home/prabhumk/Downloads/embox/build/base/gen/include -I/home/prabhumk/Downloads/embox/build/base/gen/src/include -I/home/prabhumk/Downloads/embox/build/base/gen/src/include -I/home/prabhumk/Downloads/embox/build/base/gen/src/arch/arm/include -I/home/prabhumk/Downloads/embox/platform/stm32f746g_discovery/include -I/home/prabhumk/Downloads/embox/src/compat/linux/include -I/home/prabhumk/Downloads/embox/src/compat/posix/include -I/home/prabhumk/Downloads/embox/src/compat/bsd/include -I/home/prabhumk/Downloads/embox/src/compat/libc/include -I/home/prabhumk/Downloads/embox/src/compat/cxx/include -nostdinc -MMD -MP -DNDEBUG' EMBOX_LDFLAGS='-static -nostdlib  '
mkdir -p /home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/build && cd /home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/build && ( \
	. /home/prabhumk/Downloads/embox/build/base/gen/embox_gcc_env.sh && /home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/build /home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/configure \
		--target=arm-none-eabi \
		--with-cross-host=embox \
		--prefix=/home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/install \
		--disable-multilib \
		--disable-shared \
		--disable-tls \
		--disable-nls \
		--disable-decimal-float \
		--disable-libffi \
		--disable-libgomp \
		--disable-libmudflap \
		--disable-libquadmath \
		--disable-libstdcxx-pch \
		--disable-libssp \
		--disable-bootstrap \
		--without-headers \
		--without-newlib \
		--enable-languages=c,c++ \
		--enable-soft-float \
		CC_FOR_TARGET=${EMBOX_CROSS_COMPILE}gcc \
		CXX_FOR_TARGET=${EMBOX_CROSS_COMPILE}g++ \
		CFLAGS_FOR_TARGET="${EMBOX_IMPORTED_CPPFLAGS} -I/home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/build/gcc/include -DLIBGCC2_HAS_XF_MODE=0 -DDO_GLOBAL_CTORS_BODY -DDO_GLOBAL_DTORS_BODY ${EMBOX_IMPORTED_CFLAGS}" \
		CXXFLAGS_FOR_TARGET="${EMBOX_IMPORTED_CPPFLAGS} -I/home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/build/gcc/include -fno-strict-aliasing -fno-common -fno-stack-protector -fno-pic -Wall -Os -g -mthumb -mlittle-endian -mcpu=cortex-m7 -ffreestanding -fomit-frame-pointer -mfpu=fpv5-sp-d16 -mfloat-abi=hard -fpermissive" \
		LDFLAGS_FOR_TARGET="${EMBOX_IMPORTED_LDFLAGS}"; \
)
/bin/sh: 2: /home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/build: Permission denied
make[5]: *** [Makefile:69: /home/prabhumk/Downloads/embox/build/extbld/third_party/gcc/gcc_build/.configured] Error 126
make[4]: *** [/home/prabhumk/Downloads/embox/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: /home/prabhumk/Downloads/embox/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2
```

anton-bondarev, 2022-10-05:
Hi,

> configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try to install GMP MPFR MPC 
`sudo apt-get install libgmp-dev libmpfr-dev libmpc-dev`

and rebuild again
```
make clean
make
```

pkuppusamy77, 2022-10-06:
Thank you. I got it compiled..

anton-bondarev, 2022-10-22:
@pkuppusamy77 Can I close the issue?

Feel free to open issues if something is wrong or unclear

closed 2024-07-05 by anton-bondarev