embox/embox #2615: x86/qemu constant crash

ioannessh, 2022-10-04:
x86/qemu always crash on master

```
commit f6f70e4e2496d4e81fd4b46a73189c4794d71b38 (HEAD -> master, origin/master, origin/HEAD)
Merge: d713a61e0 f37185eb3
Author: Anton Bondarev <anton.bondarev2310@gmail.com>
Date:   Sun Oct 2 17:48:42 2022 +0300

    Merge pull request #2611 from ioannessh/test-memchr-&-memrchr
    
    Add tests for the memchr and memrchr functions

$ make confload-x86/qemu
$ make
$ ./scripts/qemu/auto_qemu
	unit: initializing embox.fs.rootfs: EXCEPTION [0x6]: error = 00000000
EAX=0a55eac0    EBX=001df614 ECX=00000010 EDX=0a55eac0
 GS=09c90010     FS=00290010  ES=00000010  DS=01b90010
EDI=00001000    ESI=00000000 EBP=01b98e58 EIP=00189f5e
 CS=00000008 EFLAGS=00010202 ESP=01b98e64  SS=00000004

 --   00001000 * A R    thread 0  task 1 --------------------------------------

 11 0x0014d14f <exception_handler+0x113> ...h/x86/kernel/exception_handler.c:20
 10 0x00100079 <excep_stub+0x1c> 
  9 0x00189f5e <super_block_alloc+0x75>                /src/fs/super_block.c:21
  8 0x0018bf94 <mount+0x17>                            /src/fs/syslib/fsop.c:19
  7 0x0010e1dd <rootfs_mount+0x19>                   /src/fs/rootfs/rootfs.c:26
  6 0x0010e2bd <unit_init+0x22>                      /src/fs/rootfs/rootfs.c:51
  5 0x0014c686 <unit_mod_enable+0x57>                  /src/framework/unit.c:25
  4 0x0014cd60 <mod_enable+0x7f>                  /src/framework/mod/core.c:172
  3 0x0014c584 <runlevel_set+0xa4>                 /src/framework/runlevel.c:54
  2 0x0010114e <init+0x28>                                /src/kernel/init.c:59
  1 0x001010d5 <kernel_start+0x10>                        /src/kernel/init.c:24

run 0x00185224 <boot_stub+0x0>               /src/kernel/sched/boot_thread.c:20
```

anton-bondarev, 2022-10-04:
@ioannessh Please give information about gcc and qemu version

ioannessh, 2022-10-04:
```
$ gcc --version
gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.

$ qemu-system-i386 --version
QEMU emulator version 7.1.0
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers
```

ioannessh, 2022-10-12:
Invalid opcode exception appears when it encounters ```sse``` instructions

PR #2620 (2022-10-13): Update CFLAGS
This PR add CFLAGS to fix x86/qemu crash

Fixes #2615
closed 2022-10-13 by anton-bondarev