embox/embox #2654: Build is breaking while compiling for stm32f764g-discovery

kumarchaitanyab, 2023-01-13:
I am getting below errors while compiling for stm32f764g-discovery board. Can you please suggest what could be the reason?

I also tried downloading below zip file through wget but still no luck

"**https://github.com/STMicroelectronics/STM32CubeF7/archive/refs/tags/v1.16.0.zip**"

Errors getting while compilation:

error1: 
 BUILDGEN /home/snanduri/opencv/embox/build
Error: modules dependency cycle:
        embox.driver.block_dev
        embox.fs.buffer_cache

error2: 

config.status: creating testsuite/Makefile
config.status: creating config.h
config.status: creating config.h
config.status: executing default commands
config.status: executing default commands
make[6]: Leaving directory '/home/snanduri/opencv/embox/build/extbld/third_party/gcc/gcc_build/build'
make[5]: *** [Makefile:99: /home/snanduri/opencv/embox/build/extbld/third_party/gcc/gcc_build/.builded] Error 2
make[4]: *** [/home/snanduri/opencv/embox/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: /home/snanduri/opencv/embox/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[4]: *** Waiting for unfinished jobs....
100  594M    0  594M    0     0   473k      0 --:--:--  0:21:24 --:--:--  408k
touch /home/snanduri/opencv/embox/build/extbld/third_party/bsp/stmf7cube/cube/.downloaded
if [ "https://github.com/STMicroelectronics/STM32CubeF7/archive/refs/tags/v1.16.0.zip" ] ; then \
        cd /home/snanduri/opencv/embox/download && ( \
                        md5sum stm32cubef7-v1.16.0.zip | awk '{print $1}' | grep 9586ec0996e154c21dbb31f92834c3aa 2>&1 >/dev/null; \
                ); \
    fi;
unzip -q /home/snanduri/opencv/embox/download/stm32cubef7-v1.16.0.zip -d /home/snanduri/opencv/embox/build/extbld/third_party/bsp/stmf7cube/cube;
md5sum: stm32cubef7-v1.16.0.zip: No such file or directory
unzip:  cannot find or open /home/snanduri/opencv/embox/download/stm32cubef7-v1.16.0.zip, /home/snanduri/opencv/embox/download/stm32cubef7-v1.16.0.zip.zip or /home/snanduri/opencv/embox/download/stm32cubef7-v1.16.0.zip.ZIP.
make[5]: *** [/home/snanduri/opencv/embox/mk/extbld/lib.mk:86: /home/snanduri/opencv/embox/build/extbld/third_party/bsp/stmf7cube/cube/.download_checked] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [/home/snanduri/opencv/embox/mk/extbld/lib.mk:98: /home/snanduri/opencv/embox/build/extbld/third_party/bsp/stmf7cube/cube/.extracted] Error 9
make[4]: *** [/home/snanduri/opencv/embox/build/base/gen/third_party/bsp/stmf7cube/cube.extbld_rule.mk:26: /home/snanduri/opencv/embox/build/base/gen/third_party/bsp/stmf7cube/cube.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2

anton-bondarev, 2023-01-13:
Hello @kumarchaitanyab , Do you mean stm32f746g-discovery?
Did you configured Embox with following
`make confload-platform/stm32/f7/stm32f746g-discovery`

First error 
> error1:
> BUILDGEN /home/snanduri/opencv/embox/build
> Error: modules dependency cycle:
> embox.driver.block_dev
> embox.fs.buffer_cache
>
is not significant. It os just internal embox message

Is there download/stm32cubef7-v1.16.0.zip
if yes please show md5summ
```
md5sum download/stm32cubef7-v1.16.0.zip
9586ec0996e154c21dbb31f92834c3aa  download/stm32cubef7-v1.16.0.zip
```
if no try to 
download manually form [https://github.com/STMicroelectronics/STM32CubeF7/archive/refs/tags/v1.16.0.zip](https://github.com/STMicroelectronics/STM32CubeF7/archive/refs/tags/v1.16.0.zip)
move (and rename) **v1.16.0.zip** to **download/stm32cubef7-v1.16.0.zip**

Check its md5 and rebuild embox again
```
make clean
make j5
```

kumarchaitanyab, 2023-01-13:
Thanks for you reply, I have downloaded the zip file manually but now I am getting below issue.

config.status: executing default commands
make[6]: Leaving directory '/home/snanduri/opencv/embox/build/extbld/third_party/gcc/gcc_build/build'
make[5]: *** [Makefile:99: /home/snanduri/opencv/embox/build/extbld/third_party/gcc/gcc_build/.builded] Error 2
make[4]: *** [/home/snanduri/opencv/embox/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: /home/snanduri/opencv/embox/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2

anton-bondarev, 2023-01-13:
@kumarchaitanyab Yes, I reproduce the problem,  Now it seems that the problem in wrong archive on github , but I am not fully sure and continue investigating

kumarchaitanyab, 2023-01-13:
Below is the output:
(ak_226) snanduri:2@ ~/opencv/embox$ md5sum download/stm32cubef7-v1.16.0.zip
9586ec0996e154c21dbb31f92834c3aa  download/stm32cubef7-v1.16.0.zip

Attaching the issue again, not seeing the specific error message though.   
ar rc ./libiberty.a \
    ./regex.o ./cplus-dem.o ./cp-demangle.o ./md5.o ./sha1.o ./alloca.o ./argv.o ./choose-temp.o ./concat.o ./cp-demint.o ./crc32.o ./d-demangle.o ./dwarfnames.o ./dyn-string.o ./fdmatch.o ./fibheap.o ./filename_cmp.o ./floatformat.o ./fnmatch.o ./fopen_unlocked.o ./getopt.o ./getopt1.o ./getpwd.o ./getruntime.o ./hashtab.o ./hex.o ./lbasename.o ./lrealpath.o ./make-relative-prefix.o ./make-temp-file.o ./objalloc.o ./obstack.o ./partition.o ./pexecute.o ./physmem.o ./pex-common.o ./pex-one.o ./pex-unix.o ./vprintf-support.o ./safe-ctype.o ./simple-object.o ./simple-object-coff.o ./simple-object-elf.o ./simple-object-mach-o.o ./simple-object-xcoff.o ./sort.o ./spaces.o ./splay-tree.o ./stack-limit.o ./strerror.o ./strsignal.o ./timeval-utils.o ./unlink-if-ordinary.o ./xasprintf.o ./xatexit.o ./xexit.o ./xmalloc.o ./xmemdup.o ./xstrdup.o ./xstrerror.o ./xstrndup.o ./xvasprintf.o  ./setproctitle.o; \
  ranlib ./libiberty.a; \
  cd ..; \
else true; fi
make[7]: Leaving directory '/home/snanduri/opencv/embox/build/extbld/third_party/gcc/gcc_build/build/build-x86_64-pc-linux-gnu/libiberty'
make[6]: Leaving directory '/home/snanduri/opencv/embox/build/extbld/third_party/gcc/gcc_build/build'
make[5]: *** [Makefile:99: /home/snanduri/opencv/embox/build/extbld/third_party/gcc/gcc_build/.builded] Error 2
make[4]: *** [/home/snanduri/opencv/embox/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: /home/snanduri/opencv/embox/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2

anton-bondarev, 2023-01-13:
> md5sum download/stm32cubef7-v1.16.0.zip
> 9586ec0996e154c21dbb31f92834c3aa download/stm32cubef7-v1.16.0.zip

Yes, it is the correct archive

try to make clean and build again
```
make clean cacheclean
make
```

if there is some errors show me output
you can save it into a file
`make > embox_build.log`

anton-bondarev, 2023-01-14:
> After redirecting the logs to a file, I could not see any errors getting logged in the log file, however some are shown up on terminal. Please find the error logs which were seen on terminal.

Could you attach the log file?
I want to see full build log with the errors/

kumarchaitanyab, 2023-01-16:
Please find the log file attachment. The errors were copied in the previous chat.
[build_log_13_jan.txt](https://github.com/embox/embox/files/10422670/build_log_13_jan.txt)

kumarchaitanyab, 2023-01-16:
I have one more question, the open CV code which gets compiled here will be using arm_none_eabi tool chain or gcc tool chain? While looking at the mk files in path (embox/build/extbld/third_party/lib/opencv/build/install/lib/) are suggesting me that the openCV gets compiled using gcc and not arm_none_eabi tool chain? please let me know.

anton-bondarev, 2023-01-16:
> Please find the log file attachment. The errors were copied in the previous chat. [build_log_13_jan.txt](https://github.com/embox/embox/files/10422670/build_log_13_jan.txt)

Thanks, now I see the error 
```md5sum: stm32cubef7-v1.16.0.zip: No such file or directory```
it still links with stm32cubef7-v1.16.0.zip downloading. It is weird because you show the md5sum
```md5sum download/stm32cubef7-v1.16.0.zip
9586ec0996e154c21dbb31f92834c3aa download/stm32cubef7-v1.16.0.zip
```
Is there any changes in your code?

kumarchaitanyab, 2023-01-16:
I did not do any code changes and not sure how to come out of this issue. Can you also tell me about the other question I had on openCV compilation toolchain?

kumarchaitanyab, 2023-01-17:
I cloned a fresh repo and compiled again. I had to manually wget stm32cubef7-v1.16.0.zip file in order to go to next compilation process. Now I see a below error. Do you have any recommendation to fix this?

below commands are given for compilation:

make confload-project/opencv/stm32f746g-discovery
make

error log:

configure: error: in `/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build/build/gcc':
configure: error: C++ preprocessor "/lib/cpp" fails sanity check
See `config.log' for more details.
make[6]: *** [Makefile:4077: configure-gcc] Error 1
make[6]: Leaving directory '/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build/build'
make[5]: *** [Makefile:99: /home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build/.builded] Error 2
make[4]: *** [/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: /home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2

kumarchaitanyab, 2023-01-17:
[build_log.log](https://github.com/embox/embox/files/10430542/build_log.log)
Attaching the log file again for reference, I see that compilation process downloading gcc-6.3.0, just wanted to know is it difference from arm_none_eabi version?

anton-bondarev, 2023-01-17:
> make confload-project/opencv/stm32f746g-discovery
> make

Could we start with building base stm32f746g-discovery template (without OpenCV and other packages)
```
make confload-platform/stm32/f7/stm32f746g-discovery
make
```
It help to localize the problem

kumarchaitanyab, 2023-01-17:
with below command, build is successful. Pleas find the log file.
make confload-platform/stm32/f7/stm32f746g-discovery
make
[stm32.log](https://github.com/embox/embox/files/10433860/stm32.log)

The issue is occurring when we compile for openCV.
For OpenCV compilation, I created .builded and .installed files manually in /home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build then compiled it again. The build did go through further but later it is failing due to libgcc.a file and it is unable to create it may be.

arm-none-eabi-objdump: '/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/obj/../../extbld/third_party/gcc/gcc_build/install/libs/libgcc.a': No such file
arm-none-eabi-objdump: '/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/obj/../../extbld/third_party/gcc/gcc_build/install/libs/libgcc.a': No such file
arm-none-eabi-objdump: '/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/obj/../../extbld/third_party/gcc/gcc_build/install/libs/libgcc.a': No such file
arm-none-eabi-objdump: '/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/obj/../../extbld/third_party/gcc/gcc_build/install/libs/libgcc.a': No such file
arm-none-eabi-objdump: '/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/obj/../../extbld/third_party/gcc/gcc_build/install/libs/libgcc.a': No such file
arm-none-eabi-objdump: '/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/obj/../../extbld/third_party/gcc/gcc_build/install/libs/libgcc.a': No such file
arm-none-eabi-ar: build/extbld/third_party/gcc/gcc_build/install/libs/libgcc.a: No such file or directory
mk/arhelper.mk:69: *** Can't get members of the archive 'build/extbld/third_party/gcc/gcc_build/install/libs/libgcc.a'.  Stop.
make[4]: *** [mk/image2.mk:146: /home/chaitanya/Chaitanya/mbox_Jan16/embox/build/base/obj/module/third_party/gcc/libgcc.a] Error 2

anton-bondarev, 2023-01-17:
> with below command, build is successful. Pleas find the log file.

Great! Thanks.

> I created .builded and .installed files manually in /home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build then compiled it again.

I think it can not work. You must build all environment in embox

> configure: error: in /home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build/build/gcc': > > configure: error: C++ preprocessor "/lib/cpp" fails sanity check See config.log' for more details.

Is there g++ in your host system?
If yes, please attach 
/home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build/build/gcc/config.log

kumarchaitanyab, 2023-01-17:
Please find the config.log file
[config.log](https://github.com/embox/embox/files/10434668/config.log)

anton-bondarev, 2023-01-17:
> Please find the config.log file [config.log](https://github.com/embox/embox/files/10434668/config.log)

As I see there is no g++ compiler

> /home/chaitanya/Chaitanya/mbox_Jan16/embox/build/extbld/third_party/gcc/gcc_build/gcc-6.3.0/configure: line 4954: g++: command not found

try `g++ --version`
You need to install g++ to build embox internal cross-tools

kumarchaitanyab, 2023-01-17:
When I tried installing g++, I was getting below issue, 

sudo apt-get install g++

**The following packages have unmet dependencies:
 g++ : Depends: g++-9 (>= 9.3.0-3~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.**

Then I tried installing g++-9 using : sudo apt-get install g++
I am getting below issue:

he following packages have unmet dependencies:
 **g++-9 : Depends: gcc-9-base (= 9.3.0-10ubuntu2) but 9.4.0-1ubuntu1~20.04.1 is to be installed
         Depends: gcc-9 (= 9.3.0-10ubuntu2) but 9.4.0-1ubuntu1~20.04.1 is to be installed
         Depends: libstdc++-9-dev (= 9.3.0-10ubuntu2) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.**

kumarchaitanyab, 2023-01-17:
unfortunaltrely, I am unable to install g++ even with sudo apt-get install build-essential. However I found another PC where the g++ is already installed.

I was able to compile the build with below command 
**make confload-platform/stm32/f7/stm32f746g-discovery
make**

but getting different issue with openCV build
**configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
ftp://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
make[5]: *** [Makefile:69: /home/chaitanya/Chaitanya/mbox_Jan17/embox/build/extbld/third_party/gcc/gcc_build/.configured] Error 1
make[4]: *** [/home/chaitanya/Chaitanya/mbox_Jan17/embox/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: /home/chaitanya/Chaitanya/mbox_Jan17/embox/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2**

please find the build log
[build_log.log](https://github.com/embox/embox/files/10436760/build_log.log)

kumarchaitanyab, 2023-01-18:
Finally after giving **sudo apt-get install libgmp-dev libmpfr-dev libmpc-dev** on the new PC, build compiled without any issues.

kumarchaitanyab, 2023-01-18:
To run the example, we executed the application using openocd and gdb. It appears that build is getting exception and nothing is displayed on the LCD. Please find the output logs.
[execution_log.log](https://github.com/embox/embox/files/10443273/execution_log.log)

anton-bondarev, 2023-01-18:
> To run the example, we executed the application using openocd and gdb. It appears that build is getting exception and nothing is displayed on the LCD. Please find the output logs. [execution_log.log](https://github.com/embox/embox/files/10443273/execution_log.log)

Yes, you are right there is an exception. I'll try to reproduce the problem on my board ASAP and tell you results.

anton-bondarev, 2023-01-18:
@kumarchaitanyab Can you also attach your embox image here. I'll check it on my board.

kumarchaitanyab, 2023-01-18:
[embox.zip](https://github.com/embox/embox/files/10445051/embox.zip)

Please find the zipped elf file

kumarchaitanyab, 2023-01-19:
Any update from you? do you have any inputs for me?

anton-bondarev, 2023-01-20:
> Any update from you? do you have any inputs for me?

Sorry, for the slow response

I've checked your embox image it works on my board

```
Interrupt controller: cortexm_nvic

Embox kernel start
        unit: initializing embox.mem.static_heap: done
        unit: initializing embox.kernel.task.task_resource: done
        unit: initializing embox.kernel.task.task_table: done
        unit: initializing embox.kernel.task.kernel_task: done
        unit: initializing embox.arch.arm.fpu.fpv5: 
                        FPv4/FPv5 info:
                         MVFR0 (Media and VFP Feature Register 0) = 0x10110021
                                 Single precision support = yes
                                 Double precision support = no
                                 Size of FP register bank = 16 x 64-bit registes
                         MVFR1 (Media and VFP Feature Register 1) = 0x11000011  
                                                                                
                        FPU registers:                                          
                         CPACR (Coprocessor Access Control Register)     = 0x000
                         FPCCR (Floating-point Context Control Register) = 0x000
                                 Automatic hardware state preservation and restn
                                 on exception entry and exit       = no         
                                 Automatic lazy state preservation = no         
                done                                                            
        unit: initializing embox.kernel.sched.sched: done                       
        unit: initializing embox.driver.interrupt.cortexm_nvic: done            
        unit: initializing embox.arch.arm.armmlib.armv7m_cpu_cache: debug: arm_ 
SRAM section non-cacheble: start=0x20008000, region size=0x4000                 
debug: arm_mpu_init_nocache_regions:                                            
SRAM device memory: start=0x20008000, region size=0x200                         
debug: arm_mpu_init_nocache_regions:                                            
Region0 non-cacheble: start=0x60000000, region size=0x100000                    
done                                                                            
        unit: initializing embox.driver.flash.stm32f7_qspi: done                
        unit: initializing third_party.bsp.stmf7cube.sdram: done                
runlevel: init level is 0                                                       
        unit: initializing embox.driver.clock.cortexm_systick: done             
        unit: initializing embox.mem.fixed_heap: done                           
runlevel: init level is 1                                                       
        unit: initializing embox.mem.phymem:                                    
                        start=0x20040540 end=0x20050000 size=64192              
                done                                                            
        unit: initializing embox.fs.buffer_cache: done                          
        unit: initializing embox.driver.video.stm32f7_lcd: done                 
        unit: initializing embox.driver.net.stm32f7cube_eth: done               
        unit: initializing embox.driver.tty.serial: done                        
        unit: initializing embox.fs.rootfs_dvfs: done                           
runlevel: init level is 2                                                       
        unit: initializing embox.init.start_script:                             
                                                                                
Started shell [tish] on device [diag]                                           
loading start script:                                                           
> edges fruits.png 20                                                           
Image: 512x269; Threshold=20                                                    
Detection time: 0 s 324 ms                                                      
Framebuffer: 480x272 32bpp                                                      
root@embox:(null)#
```
I see fruit's borders image on the display

kumarchaitanyab, 2023-01-20:
Can you please let me know the procedure you used to load the image? I am using stm32f746 based discovery board to test the image.

anton-bondarev, 2023-01-21:
> Can you please let me know the procedure you used to load the image? I am using stm32f746 based discovery board to test the image.

Oh, yes, I forget that it is not a common sequence You should use './scripts/gdb_run_stm32f7_qspi.sh' script instead of using gdb

1. Connect to the board through openocd. I use `sudo ../src/openocd -f board/stm32f746g-disco.cfg`
2. run `./scripts/gdb_run_stm32f7_qspi.sh` it load all required section into in-chip flash and QSPI flash 

PS sometimes it needs to erase in-chip flash and I use st-tools for it `./st-flash --connect-under-reset erase`

kumarchaitanyab, 2023-01-21:
I see that there are two bins used in the ./scripts/gdb_run_stm32f7_qspi.sh file. Please let me know what are they?
is this embox_XXXX.bin corresponds to bin generated using **make confload-arm/stm32f746g-discovery**? and qspi_XXXX.bin using **make confload-project/opencv/stm32f746g-discovery**

**EMBOX_BIN=$(mktemp --tmpdir=$PWD embox_XXXX.bin)
QSPI_BIN=$(mktemp --tmpdir=$PWD qspi_XXXX.bin)**

anton-bondarev, 2023-01-21:
> I see that there are two bins used in the ./scripts/gdb_run_stm32f7_qspi.sh file. Please let me know what are they?

If you look into the lds.conf file in 'project/opencv/stm32f746g-discovery' template you see 3 memory regions
```
/* region (origin, length) */
ROM (0x08000000, 1M)
 /* SRAM: 320K (256K RAM + 64K for TCM RAM) */
RAM (0x20000000, 320K)
region (QSPI,  0x90000000, 16M)
```
and also a couple of sections 
```
/* section (region[, lma_region]) */
text   (ROM)
rodata (ROM)
data   (RAM, ROM)
bss    (RAM)

...
/* InitFS */
section (initfs, QSPI, QSPI)
phdr    (initfs, PT_LOAD, FLAGS(5))

/* OpenCV core */
section (cv_core_text, QSPI, QSPI)
phdr    (cv_core_text, PT_LOAD, FLAGS(5))

section (cv_core_rodata, QSPI, QSPI)
phdr    (cv_core_rodata, PT_LOAD, FLAGS(5))

section (cv_core_arm_exidx, QSPI, QSPI)
phdr    (cv_core_arm_exidx, PT_LOAD, FLAGS(5))

section (cv_core_arm_extab, QSPI, QSPI)
phdr    (cv_core_arm_extab, PT_LOAD, FLAGS(5))

section (cv_core_data, RAM, QSPI)
phdr    (cv_core_data, PT_LOAD, FLAGS(6))

section (cv_core_bss, RAM, QSPI)
phdr    (cv_core_bss, PT_LOAD, FLAGS(6))

/* OpenCV imgcodecs */
....

section (libsupcxx_bss, RAM, QSPI)
phdr    (libsupcxx_bss, PT_LOAD, FLAGS(6))
```

so there are sections for in-chip and QSPI memory in target elf file. You can use command 'load' in gdb+openocd only for in-chip memory sections. It needs special sequence for QSPI flash, so we split embox (elf file with all sections) for two binary files (EMBOX_BIN contains in-chip sections and QSPI_BIN contains QSPI sections).

> Also do let me know if we need to connect ethernet cable to the board?

No, you can use it without an ethernet connection.

kumarchaitanyab, 2023-01-23:
I am getting below, not sure if this is working correctly.

I dont have **board/stm32f746g-disco.cfg** file, I am using /usr/share/openocd/scripts/board/stm32f7discovery.cfg and I hope both are same files.

**(ak_226) snanduri:2@ ~/opencv/embox_13_jan_org/new/embox$ sudo openocd -f /usr/share/openocd/scripts/board/stm32f7discovery.cfg
[sudo] password for snanduri:
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v37 API v2 SWIM v26 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.221551
Warn : Silicon bug: single stepping will enter pending exception handler!
Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
./scripts/gdb_run_stm32f7_qspi.sh
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x10016449
Info : flash size = 1024kbytes
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
#0 : stm32f7x.flash (stm32f2x) at 0x08000000, size 0x00100000, buswidth 0, chipwidth 0
stm32x mass erase complete
  flash write_bank bank_id filename offset

Error: flash bank 3 does not exist

  flash write_bank bank_id filename offset

  flash verify_bank bank_id filename offset

target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc**

When I tried to erase the flash using **st-flash --connect-under-reset erase**, ia m getting below
I followed the proceddure given in link **https://freeelectron.ro/installing-st-link-v2-to-flash-stm32-targets-on-linux/** to install st-link properly. Can you please help me here, at this point I am stuck due to this.

st-flash 1.7.0-237-g1bec78c
/usr/share//stlink/chips: No such file or directory
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/007: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
2023-01-23T11:19:05 WARN usb.c: Couldn't find any ST-Link devices

kumarchaitanyab, 2023-01-23:
After running **sudo ldconfig** command, I was able to run **"st-flash --connect-under-reset erase"** this command and full chip erase was successful. But even then am getting the same issue after running **./scripts/gdb_run_stm32f7_qspi.sh**

**Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v37 API v2 SWIM v26 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.229005
Warn : Silicon bug: single stepping will enter pending exception handler!
Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x10016449
Info : flash size = 1024kbytes
undefined debug reason 7 - target needs reset
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
#0 : stm32f7x.flash (stm32f2x) at 0x08000000, size 0x00100000, buswidth 0, chipwidth 0
stm32x mass erase complete
  flash write_bank bank_id filename offset

Error: flash bank 3 does not exist

  flash write_bank bank_id filename offset

  flash verify_bank bank_id filename offset

target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc**

anton-bondarev, 2023-01-23:
@kumarchaitanyab 
I checked the flashing on my board

1. start openocd

```
sudo ../src/openocd -f board/stm32f746g-disco.cfg 
[sudo] password for anton: 
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J24M11 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.211534
Warn : Silicon bug: single stepping may enter pending exception handler!
Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f7x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
```

start './scripts/gdb_run_stm32f7_qspi.sh' script (another console)
```
./scripts/gdb_run_stm32f7_qspi.sh 
GDB script:
target ext :3333
monitor reset halt
monitor reset init
monitor flash banks
monitor stm32f2x mass_erase 0
monitor flash write_bank 0 /home/anton/embox/embox_Co9U.bin
monitor flash erase_sector 3 0 13
monitor flash write_bank 3 /home/anton/embox/qspi_fyf7.bin
monitor flash verify_bank 3 /home/anton/embox/qspi_fyf7.bin
monitor reset halt
continue

GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/base/bin/embox...
exp_default_entry () at /src/arch/arm/armmlib/exception_table.S:48
48		mrs r1, IPSR
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
#0 : stm32f7x.flash (stm32f2x) at 0x08000000, size 0x00100000, buswidth 0, chipwidth 0
#1 : stm32f7x.otp (stm32f2x) at 0x1ff0f000, size 0x00000400, buswidth 0, chipwidth 0
#2 : stm32f7x.itcm-flash.alias (virtual) at 0x00200000, size 0x00100000, buswidth 0, chipwidth 0
[... 52 lines cut ...]
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
#0 : stm32f7x.flash (stm32f2x) at 0x08000000, size 0x00100000, buswidth 0, chipwidth 0
#1 : stm32f7x.otp (stm32f2x) at 0x1ff0f000, size 0x00000400, buswidth 0, chipwidth 0
#2 : stm32f7x.itcm-flash.alias (virtual) at 0x00200000, size 0x00100000, buswidth 0, chipwidth 0
#3 : stm32f7x.qspi (stmqspi) at 0x90000000, size 0x00000000, buswidth 0, chipwidth 0

stm32x mass erase complete

wrote 519676 bytes from file /home/anton/embox/embox_Co9U.bin to flash bank 0 at offset 0x00000000 in 3.603401s (140.838 KiB/s)

Info : flash1 'micron n25q128' id = 0x18ba20 size = 16384kbytes
erased sectors 0 through 13 on flash bank 3 in 4.225213s

wrote 829205 bytes from file /home/anton/embox/qspi_fyf7.bin to flash bank 3 at offset 0x00000000 in 5.223377s (155.028 KiB/s)

read 829205 bytes from file /home/anton/embox/qspi_fyf7.bin and flash bank 3 at offset 0x00000000 in 5.158771s (156.970 KiB/s)
contents match

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000220 msp: 0x200008a8
```

anton-bondarev, 2023-01-23:
@kumarchaitanyab I see you use openocd 0.10.0. Try to use 0.11.0 version You can build it from source
[https://github.com/embox/embox/wiki/How-to-flash-and-run-STM32-boards#building-openocd-from-sources](https://github.com/embox/embox/wiki/How-to-flash-and-run-STM32-boards#building-openocd-from-sources)

kumarchaitanyab, 2023-01-23:
There is an issue in getting the latest version

[image]

I have one more question on gcc which gets downloaded as part of embox compilation process. I could see that the gcc 9.3.0 tar file getting downloaded (I have specified 9.3.0 instead of 6.3.0))  and as part of cross compilation 9.3.0 source is getting compiled to support pthread_ *’ and ‘pthread_mutex_ *’ (per https://www.embedded.com/running-advanced-c-software-on-mcus/). However, I could not see the bin files getting generated? will it not generate the bin files as part of compilation itself? 
I have 9.3.1 installed in the PC and the embox build environment is having dependency with the installed 9.3.1 (may be for bin files). If I remove the 9.3.1 path from my environment then embox build is not getting compiled. 

I am not sure if I am missing something or it is expected that bin files will not be generated per the embox compilation processes?

anton-bondarev, 2023-01-23:
> There is an issue in getting the latest version

Try this link [https://github.com/openocd-org/openocd/releases/tag/v0.11.0](https://github.com/openocd-org/openocd/releases/tag/v0.11.0)

anton-bondarev, 2023-01-23:
> I have one more question on gcc

You can use different cross-tools version on host and in embox. You need host gcc (native x86 for x86 host) and arm-none-eabi-gcc
* arm-none-eabi-gcc is used
* host gcc is used for embox build system and **building internal embox gcc and g++ (c++ standard) libraries**. It does not matter which internal version you use (6.3.0 or 9.3.1). This article describes wich it needs [https://www.embedded.com/running-advanced-c-software-on-mcus/](https://www.embedded.com/running-advanced-c-software-on-mcus/)

kumarchaitanyab, 2023-01-23:
After using v0.11.0, am getting below (with SWD):

on terminal 1: sudo ../src/openocd -f board/stm32f746g-disco.cfg
**Open On-Chip Debugger 0.11.0-dirty (2023-01-23-21:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J37M26 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.233202
Info : stm32f7x.cpu: hardware has 0 breakpoints, 10 watchpoints
Info : starting gdb server for stm32f7x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0xa05f0000
Warn : Cannot identify target as a STM32 family.
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0xa05f0000
Warn : Cannot identify target as a STM32 family.
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0xa05f0000
Warn : Cannot identify target as a STM32 family.
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected**

on terminal 2: ./scripts/gdb_run_stm32f7_qspi.sh
**GDB script:
target ext :3333
monitor reset halt
monitor reset init
monitor flash banks
monitor stm32f2x mass_erase 0
monitor flash write_bank 0 /home/chaitanya/Chaitanya/mbox_ocv_jan20/embox/embox_WE5s.bin
monitor flash erase_sector 3 0 13
monitor flash write_bank 3 /home/chaitanya/Chaitanya/mbox_ocv_jan20/embox/qspi_Sd2D.bin
monitor flash verify_bank 3 /home/chaitanya/Chaitanya/mbox_ocv_jan20/embox/qspi_Sd2D.bin
monitor reset halt
continue

GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update) 8.3.1.20191211-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/base/bin/embox...
/home/chaitanya/Chaitanya/mbox_ocv_jan20/embox/embox_gdb_stm32f7_qspi.Rivo:1: Error in sourced command file:
Remote connection closed**

anton-bondarev, 2023-01-23:
> Info : stm32f7x.cpu: hardware has 0 breakpoints, 10 watchpoints
> Info : starting gdb server for stm32f7x.cpu on 3333
> Info : Listening on port 3333 for gdb connections
> Info : accepting 'gdb' [...]

This is wrong connection. Try to erase you chip before openocd connecting. You must have a correct connection with openocd before using ./scripts/gdb_run_stm32f7_qspi.sh

kumarchaitanyab, 2023-01-24:
Thanks for your support, now am able to see the same output on LCD screen after the chip full erase. 

I need to run the face detection application and looking into the opencv_samples repo and followed the steps but am not sure how to test the application and not sure in which terminal we need to specify the commands present conf/start_script.inc.

[image]

anton-bondarev, 2023-01-24:
> am not sure how to test the application and not sure in which terminal we need to specify the commands present conf/start_script.inc.

You can setup test application in conf/start_script.inc before you do 'make'

do following
make confload-ext_project/opencv_samples/stm32f746g-discovery
modify conf/start_script.inc as you wish by default it start face detection with 5 girl's faces as described here
[https://github.com/embox/opencv_samples/blob/main/opencv_samples/templates/stm32f746g-discovery/start_script.inc](https://github.com/embox/opencv_samples/blob/main/opencv_samples/templates/stm32f746g-discovery/start_script.inc)
build embox with `make`
and flash embox into the board as you did before

kumarchaitanyab, 2023-01-24:
Thanks a lot for you support. I am able to see different pictures now. One question though on detection timings. 
Per benchmark numbers provided in link **https://www.embedded.com/benchmarking-opencv-on-stm32-mcus/**, I expected the detection time is some what closer to 3 or 4 seconds per the table in the link. But I see that detection time is 12.325s in the print messages. Not sure if the numbers what I see are correct? I captured the snap shot of the number.
[image]

anton-bondarev, 2023-01-24:
> I expected the detection time is some what closer to 3 or 4 seconds per the table in the link. But I see that detection time is 12.325s in the print messages. 

Hm, It's weird for me too. However you have to take account of you use another type of board. I see that on stm32f769i_discovery we use in-chip memories (ROM and RAM). We place only initfs (images for demo) into QSPI. While on stm32f746g_discovery you use external memories including for OpenCV.  Exept this maybe this chip has smaller caches or may be we do not use some features of it. I mean we didn't measure OpenCV on this board, but if you check this demos on stm32f769i_discovery you get the same results as in the article

anton-bondarev, 2023-06-30:
@kumarchaitanyab Hello.
I've close this issue.
Feel free to open any issue if a question is raised.

closed 2024-07-05 by anton-bondarev