embox/embox #2663: Fail Load Embox with openocd

landoffree0412, 2023-02-06:
Referring to the site below, I put OpenCV on the STM32H747-disco board.

https://github.com/embox/embox/wiki/OpenCV-on-STM32

The order of execution is as follows.
1. Compile this template
    make confload-project/opencv/stm32f746g-discovery
    make
2. Load Embox with openocd
    arm-none-eabi-gdb build/base/bin/embox

As a result, the following error occurs. Could you please help?

/usr/share/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: build/base/bin/embox: in function `__module_embox__lib__cxx__DestructionStatic_bss_vma':
/src/compat/cxx/core/cxx_destruction_static.c:25: multiple definition of `__dso_handle'; /usr/share/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/crtbegin.o:(.data+0x0): first defined here
/usr/share/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: cannot use executable file 'build/base/bin/embox' as input to a link
collect2: error: ld returned 1 exit status

Thanks,

anton-bondarev, 2023-02-06:
Where does the error appear, during the compilation or gdb loading?
You should use this instraction for the loading
[https://github.com/embox/embox/issues/2654#issuecomment-1399228416](https://github.com/embox/embox/issues/2654#issuecomment-1399228416)

landoffree0412, 2023-02-07:
Thanks for your reply.

The error is occurred in gdb loading.
Actually I don't have the stm32f746g-discovery board, only the STM32H747-disco board.
And the configuration is used stm32f746g-discovery.
And when I did it the way you guided me, the following error occurred.

bccho@bccho:~/embox$ ./scripts/gdb_run_stm32f7_qspi.sh 
GDB script:
target ext :3333
monitor reset halt
monitor reset init
monitor flash banks
monitor stm32f2x mass_erase 0
monitor flash write_bank 0 /home/bccho/embox/embox_JGnf.bin
monitor flash erase_sector 3 0 13
monitor flash write_bank 3 /home/bccho/embox/qspi_M8uF.bin
monitor flash verify_bank 3 /home/bccho/embox/qspi_M8uF.bin
monitor reset halt
continue

arm-none-eabi-gdb: error: language /home/bccho/embox/embox_gdb_stm32f7_qspi.WU3k not recognized
arm-none-eabi-gdb: error: language /home/bccho/embox/embox_gdb_stm32f7_qspi.WU3k not recognized

Thanks,

landoffree0412, 2023-02-07:
I proceeded again as follows. (My board : **STM32H747-disco**)

**1) configuration setting with stm32h745h & make**
(The board I have is STM32H747-discovery, but there is no corresponding configuration, so I used stm32h745i-discovery)

bccho@bccho: ~/embox$ make confload-project/opencv/**stm32h745i-discovery**
Config complete
bccho@bccho: ~/embox$ make

**2) Connect with OpenOCD**
(There is 'stm32h747i-disco.cfg' in openocd and it is used)
bccho@bccho:~/openocd-0.11.0/tcl$ sudo openocd -f board/**stm32h747i-disco.cfg** 
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1800 kHz
Info : STLINK V3J9M3 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.277351
Info : stm32h747xih6.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h747xih6.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

**3) Load Embox**
**(Error occured)**
bccho@bccho:~/embox$ ./scripts/gdb_run_stm32h7_qspi.sh 
**arm-none-eabi-objcopy: 'build/base/bin/embox': No such file
arm-none-eabi-objcopy: 'build/base/bin/embox': No such file
dd: failed to open '/home/bccho/embox/embox_URJ5.bin': No such file or directory
dd: failed to open '/home/bccho/embox/embox_URJ5.bin': No such file or directory
stat: cannot stat '/home/bccho/embox/qspi_Pipy.bin': No such file or directory**
GDB script:
target ext :3333
monitor reset init
monitor flash banks
monitor stm32h7x mass_erase 0
monitor flash write_bank 0 /home/bccho/embox/embox_bank0_ntk2.bin
monitor stm32h7x mass_erase 1
monitor flash write_bank 1 /home/bccho/embox/embox_bank1_N9CH.bin
monitor flash erase_sector 2 0 0
monitor flash write_bank 2 /home/bccho/embox/qspi_Pipy.bin
monitor flash verify_bank 2 /home/bccho/embox/qspi_Pipy.bin
monitor reset halt
continue

**arm-none-eabi-gdb: error: build/base/bin/embox: No such file or directory
arm-none-eabi-gdb: fatal error: no input files
compilation terminated.**
rm: cannot remove '/home/bccho/embox/embox_URJ5.bin': No such file or directory
rm: cannot remove '/home/bccho/embox/qspi_Pipy.bin': No such file or directory

anton-bondarev, 2023-02-07:
> arm-none-eabi-gdb: error: language /home/bccho/embox/embox_gdb_stm32h7_qspi.pWPS not recognized

Is there `arm-none-eabi-gdb` command
```arm-none-eabi-gdb --version```

Also as I remember you must erase './scripts/gdb_erase_stm32h7.sh' with before loading

landoffree0412, 2023-02-07:
arm-none-eabi-gdb --version is as follows.

bccho@bccho:~/embox$ arm-none-eabi-gdb --version
**arm-none-eabi-gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)**
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

As a result of executing **./scripts/gdb_erase_stm32h7.sh**, the following error occurs.

**bccho@bccho:~/embox$ ./scripts/gdb_erase_stm32h7.sh**
GDB script:
target ext :3333
monitor reset halt
monitor flash banks
monitor stm32h7x mass_erase 0
monitor stm32h7x mass_erase 1

**arm-none-eabi-gdb: error: unrecognized command-line option '-batch'
arm-none-eabi-gdb: fatal error: no input files**

landoffree0412, 2023-02-08:
I found the comment below.

**Ubuntu 18.04(Bionic) does not support arm-none-eabi-gdb,
However, we can use gdb-multiarch for remote debugging.**

So I tried the following. Is it correct to do it this way?
1. Install gdb-multiarch
2. gdb-multiarch build/base/embox

bccho@bccho:~/embox$ gdb-multiarch build/base/bin/embox

GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/base/bin/embox...
(gdb) target remote :3333
Remote debugging using :3333
0x08004cd0 in dlist_add_prev (list=0x10, _new=0x0)
    at /src/include/../../src/util/dlist/dlist_debug.h:150
150		__dlist_debug_check(list);
(gdb) monitor reset halt
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080006b8 msp: 0x24080000
(gdb) load
Loading section .text, size 0x13af88 lma 0x8000000
Loading section .ARM.exidx, size 0x70e8 lma 0x813af88
Loading section .ARM.extab, size 0x1225f lma 0x8142070
Loading section .rodata, size 0x86c00 lma 0x8154300
Loading section .data, size 0x14a8 lma 0x81daf00
Start address 0x08000220, load size 1950583
Transfer rate: 66 KB/sec, 15480 bytes/write.

anton-bondarev, 2023-02-09:
> However, we can use gdb-multiarch for remote debugging.

Yes, it is correct gbd, however it is the arm-none-eabi-gdb that you need to use the script
Use
```
ln -s /usr/bin/gdb-multiarch /usr/bin/arm-none-eabi-gdb
```

landoffree0412, 2023-02-13:
Thanks for your help.

With your help I seem to be done as far as embox load.
But I have a question 

How can I test an example like edge detection?
Please give me guide.

I did it as below.

I ran minicom and ran the **'continue'** command in gdb as below. 
But nothing is happened in minicom window. 
**<embox load>**
bccho@bccho:~/embox$ gdb-multiarch build/base/bin/embox
...
(gdb) target extended-remote :3333
Remote debugging using :3333
0xfffffffe in ?? ()
(gdb) monitor reset halt
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
(gdb) load
Loading section .text, size 0x13af88 lma 0x8000000
Loading section .ARM.exidx, size 0x70e8 lma 0x813af88
Loading section .ARM.extab, size 0x1225f lma 0x8142070
Loading section .rodata, size 0x86c00 lma 0x8154300
Loading section .data, size 0x14a8 lma 0x81daf00
Start address 0x08000220, load size 1950583
Transfer rate: 65 KB/sec, 15480 bytes/write.
**(gdb) continue**
Continuing.

**< minicom >**
Welcome to minicom 2.7.1

OPTIONS: I18n                                                                
Compiled on Dec 23 2019, 02:06:26.                                           
Port /dev/ttyACM0, 16:57:12                                                  
                                                                             
Press CTRL-A Z for help on special keys 

### Additional Question :
**1. Do I need to do a QSPI flash?**

bccho@bccho:~/embox$ ./scripts/gdb_run_stm32h7_qspi.sh 
1048576+0 records in
1048576+0 records out
...(snip)
Reading symbols from build/base/bin/embox...
0xfffffffe in ?? ()
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
#0 : stm32h747xih6.bank1.cpu0 (stm32h7x) at 0x08000000, size 0x00100000, buswidth 0, chipwidth 0
#1 : stm32h747xih6.bank2.cpu0 (stm32h7x) at 0x08100000, size 0x00100000, buswidth 0, chipwidth 0
#2 : stm32h747xih6.qspi (stmqspi) at 0x90000000, size 0x00000000, buswidth 0, chipwidth 0

stm32h7x mass erase complete

wrote 1048576 bytes from file /home/bccho/embox/embox_bank0_Qia9.bin to flash bank 0 at offset 0x00000000 in 8.719934s (117.432 KiB/s)

stm32h7x mass erase complete

Padding at 0x081dc3a8 with 24 bytes (bank write end alignment)
wrote 902056 bytes from file /home/bccho/embox/embox_bank1_2Elu.bin to flash bank 1 at offset 0x00000000 in 7.511300s (117.279 KiB/s)

**timeout**
**Flash bank not probed**
**failed erasing sectors 0 to 0**

**timeout**
**Nothing to write to flash bank**
**timeout**
**Nothing to compare with flash bank**
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000220 msp: 0x24000b20

**2. Board configuration needs fix? (**my board : stm32H747i-disco, use conflload : project/opencv/stm32h745i-discovery** ) )**

anton-bondarev, 2023-02-17:
> With your help I seem to be done as far as embox load.

Unfortunately, Embox load is still not correct

**First of all**
Today Embox has not 'ready to use' templates for stm32h747i-disco, only for stm32h745i-discovery
You can add support for this board yourself. Look at how it done for similar stm32h745i-discovery board. Then you can create template for OpenCV examples

anton-bondarev, 2024-07-05:
Hello @landoffree0412, can I close this issue?