embox/embox #2665: Template project/qt4/stm32f746g_discovery is broken

bursian, 2023-02-07:
running start script throw exception, then restart
```
Started shell [tish] on device [diag]
loading start script:
> export PWD=/ 
> export HOME=/ 
> export QT_QPA_FONTDIR=/fonts 
> moveblocks -platform emboxfb 

EXCEPTION (0xe000ed04):
Exception saved context:
  r0=0x200176ec r1=0x200176e4 r2=0x200177dc r3=0x01040807
  r12=0x00000073 lr=0x08019cc3 pc(ret)=0x604475c2 xPSR=0x21000000
MemManage Fault Status register = 0x00
Bus Fault Status register = 0x82
  Bus Fault Address = 0x8f2ff9c9
Usage Fault Status register = 0x0000
Hard Fault Status register = 0x40000000
```

Deryugin, 2023-02-18:
Works fine with `scripts/gdb_run_stm32f7_qspi.sh`

Deryugin, 2023-02-18:
I've added simple script to write binary to STM32 board. If config requires QSPI, this script fails

https://github.com/embox/embox/pull/2668

closed 2023-02-18 by Deryugin

bursian, 2023-02-27:
Confirm:
It works, with some small substitution in my case (i'm using gdb-multiarch):
```
ln -s /usr/bin/gdb-multiarch /usr/bin/arm-none-eabi-gdb
scripts/gdb_run_stm32f7_qspi.sh
```