embox/embox #2779: Missing implementation of pthread_attr_getstack and pthread_getattr_np

ivmai, 2023-05-26:
It would be nice to have a portable way of getting the current stack's address and size (including for the case when just the main thread exists). E.g. the following is working on Linux:
```
#include <pthread.h>
...
  pthread_attr_t attr;
  void *stackaddr;
  size_t stacksize;
  if (pthread_getattr_np(pthread_self(), &attr)) abort();
  if (pthread_attr_getstack(&attr, &stackaddr, &stacksize)) abort();
  (void)pthread_attr_destroy(&attr);
```

ivmai, 2023-06-01:
[pthread_attr_getstack()](https://www.opennet.ru/man.shtml?topic=pthread_attr_getstack&category=3&russian=4) is still missing.

anton-bondarev, 2023-06-05:
@ivmai We've added pthread attr getstack() and demo for getting different symbols 
see [https://github.com/embox/embox/blob/master/src/cmds/api_demos/memory_addresses/demo_addresses_symbols.c](https://github.com/embox/embox/blob/master/src/cmds/api_demos/memory_addresses/demo_addresses_symbols.c)

Its output is
```
root@embox:/#demo_addresses_symbols
system sect info:
data: _data_vma=0x00126bc0:0x001282b0,_data_lma=0x00126bc0, _data_len=0x16f0
bss: _bss_vma=0x0012c000:0x01c4aaa0, _bss_len=0x1b1eaa0
mod sect info:
	data_vma = (0x0x00127460)
	data_vma_end = (0x0x00127460)
	bss_vma = (0x0x00b3243c)
	bss_vma_end = (0x0x00b3243c)
pthread_getattr_np: stackaddr(0x0a43c9b4) stacksize (1f414)
pthread_attr_getstack: stackaddr(0x0a43c9b4) stacksize (1f414)

```

PR #2780 (2023-05-27): pthread attr getstack
Partly resolves #2779