embox/embox #2786: Assertion task_self() != task_kernel_task() in function task_start_exit

ivmai, 2023-06-04:
Source: master (583728d3c63)

Host: Ubuntu 20.04 / x64

How to reproduce: type "exit" in shell.

Output:
```
ASSERTION FAILED on CPU 0
        at /home/user/opensource/embox/src/kernel/task/multi/multi.c:351
        in function task_start_exit

task_self() != task_kernel_task()

 --   00001000 * A R    thread 1  task 1 --------------------------------------

 13 0x0030a466 <__assertion_handle_failure+0x321> ...at/libc/assert/assert.c:45
 12 0x0012692f <task_start_exit+0x2c>        /src/kernel/task/multi/multi.c:349
 11 0x00310e87 <_exit+0x2a>               /src/compat/posix/proc/exit/exit.c:18
 10 0x00310f17 <logging_raw+0x0>                         /src/util/logging.c:22
  9 0x002c9bbb <main_embox__cmd__logout+0x0>              /src/cmds/logout.c:11
  8 0x002ee052 <cmd_exec+0x76>                     /src/framework/cmd/core.c:23
  7 0x0016a413 <process_builtin+0x127>               /src/cmds/shell/tish.c:122
  6 0x0016b2ac <tish_exec+0x73>                      /src/cmds/shell/tish.c:238
  5 0x0016b718 <tish_run+0xfa>                       /src/cmds/shell/tish.c:329
  4 0x0016b812 <main_embox__cmd__sh__tish+0xb>       /src/cmds/shell/tish.c:400
  3 0x002ee052 <cmd_exec+0x76>                     /src/framework/cmd/core.c:23
  2 0x00169ad4 <system_start+0x144>         /src/init/system_start_service.c:34
  1 0x00102c5d <kernel_start+0x15>                        /src/kernel/init.c:24

run 0x00292094 <boot_stub+0x0>               /src/kernel/sched/boot_thread.c:20

 --   00002000       W  thread 4  task 1 --------------------------------------

  5 0x00129280 <thread_context_switch+0xc8> ...kernel/thread/thread_switch.c:36
  4 0x00291bd8 <__schedule+0x2fb>                 /src/kernel/sched/sched.c:417
  3 0x00291c36 <schedule+0x15>                    /src/kernel/sched/sched.c:494
  2 0x00128741 <thread_exit+0x25b>                /src/kernel/thread/core.c:299
  1 0x001279c7 <thread_priority_by_flags+0x0>      /src/kernel/thread/core.c:85

run 0x00167e4b <writer_thread+0x0> ...osix/pthread/pthread_attributes_test.c:13

 --   00002000       W  thread 8  task 2 --------------------------------------

 13 0x00129280 <thread_context_switch+0xc8> ...kernel/thread/thread_switch.c:36
 12 0x00291bd8 <__schedule+0x2fb>                 /src/kernel/sched/sched.c:417
 11 0x00291c36 <schedule+0x15>                    /src/kernel/sched/sched.c:494
 10 0x0029a4b5 <sched_wait_timeout+0xe2> ...ernel/thread/thread_sched_wait.c:67
  9 0x002e4cd0 <af_netlink_sock_recvmsg+0x14e> /src/net/socket/af_netlink.c:152
  8 0x001ae3b1 <krecvmsg+0x2b2>                   /src/net/socket/ksocket.c:284
  7 0x00328f3c <recvmsg+0x27f>               /src/compat/posix/net/socket.c:341
  6 0x0018c3f2 <main_embox__cmd__net__net_service+0x20d> ...er/net_service.c:72
  5 0x002ee052 <cmd_exec+0x76>                     /src/framework/cmd/core.c:23
  4 0x0032a1b7 <exec_call+0x113>               /src/compat/posix/proc/exec.c:40
  3 0x0033e235 <task_exec_callback+0xb> .../vfork_exchanged/exec_exchanged.c:17
  2 0x00125944 <task_trampoline+0x83>         /src/kernel/task/multi/multi.c:59
  1 0x001279b6 <thread_trampoline+0x116>           /src/kernel/thread/core.c:69

run 0x001258c1 <task_trampoline+0x0>          /src/kernel/task/multi/multi.c:59

 --   00001000       W  thread 9  task 3 --------------------------------------

 14 0x00129280 <thread_context_switch+0xc8> ...kernel/thread/thread_switch.c:36
 13 0x00291bd8 <__schedule+0x2fb>                 /src/kernel/sched/sched.c:417
 12 0x00291c36 <schedule+0x15>                    /src/kernel/sched/sched.c:494
 11 0x0029a106 <sched_wait+0xb>       /src/kernel/thread/thread_sched_wait.c:35
 10 0x0029a3f4 <sched_wait_timeout+0x21> ...ernel/thread/thread_sched_wait.c:67
  9 0x0033f260 <poll_table_wait+0x8d>   /src/kernel/task/idesc/poll_table.c:109
  8 0x00330649 <select+0xaa>                 /src/compat/posix/idx/select.c:111
  7 0x00183bc1 <telnet_main_loop+0xdf4>     /src/cmds/net/telnetd/telnetd.c:211
  6 0x00184d2f <main_embox__cmd__net__telnetd+0x127> ...t/telnetd/telnetd.c:347
  5 0x002ee052 <cmd_exec+0x76>                     /src/framework/cmd/core.c:23
  4 0x0032a1b7 <exec_call+0x113>               /src/compat/posix/proc/exec.c:40
  3 0x0033e235 <task_exec_callback+0xb> .../vfork_exchanged/exec_exchanged.c:17
  2 0x00125944 <task_trampoline+0x83>         /src/kernel/task/multi/multi.c:59
  1 0x001279b6 <thread_trampoline+0x116>           /src/kernel/thread/core.c:69

run 0x001258c1 <task_trampoline+0x0>          /src/kernel/task/multi/multi.c:59

 --   00001000       W  thread 10  task 4 -------------------------------------

 16 0x00129280 <thread_context_switch+0xc8> ...kernel/thread/thread_switch.c:36
 15 0x00291bd8 <__schedule+0x2fb>                 /src/kernel/sched/sched.c:417
 14 0x00291c36 <schedule+0x15>                    /src/kernel/sched/sched.c:494
 13 0x0029a106 <sched_wait+0xb>       /src/kernel/thread/thread_sched_wait.c:35
 12 0x0029a3f4 <sched_wait_timeout+0x21> ...ernel/thread/thread_sched_wait.c:67
 11 0x002e93ed <sock_wait+0x7c>                  /src/net/socket/sock_wait.c:17
 10 0x001d1557 <tcp_accept+0x232>                /src/net/socket/tcp_sock.c:336
  9 0x001b45eb <inet_accept+0x27b>                /src/net/socket/af_inet.c:317
  8 0x001ada03 <kaccept+0x352>                    /src/net/socket/ksocket.c:195
  7 0x003280e6 <accept+0x125>                /src/compat/posix/net/socket.c:125
  6 0x00186d03 <main_embox__cmd__net__httpd+0x37e> ...cmds/net/httpd/httpd.c:86
  5 0x002ee052 <cmd_exec+0x76>                     /src/framework/cmd/core.c:23
  4 0x0032a1b7 <exec_call+0x113>               /src/compat/posix/proc/exec.c:40
  3 0x0033e235 <task_exec_callback+0xb> .../vfork_exchanged/exec_exchanged.c:17
  2 0x00125944 <task_trampoline+0x83>         /src/kernel/task/multi/multi.c:59
  1 0x001279b6 <thread_trampoline+0x116>           /src/kernel/thread/core.c:69

run 0x001258c1 <task_trampoline+0x0>          /src/kernel/task/multi/multi.c:59
```

anton-bondarev, 2023-06-05:
@ivmai Thanks
We haven't come up with how the embox should react to "exit" the main shell yet.

ivmai, 2023-06-05:
I see, thank you for the explanation.

ivmai, 2023-06-05:
Maybe just output "system halted" ?

anton-bondarev, 2023-06-08:
> It would be convenient if exit from the shell causes termination of auto_qemu script.

Thanks for the good idea. we thought about this ourself, more exactly using shoutdown in this place. But Embox has shoutdown implementations not for all qemu platforms