embox/embox #2791: UbSan invalid builtin __builtin_clzll

ivmai, 2023-06-06:
Source: master (583728d3c)
Target: Qemu-x86 on Ubuntu/x64
Code: `printf("%f\n", 0.9886);`
Output:
```
UbSan invalid builtin /home/i.maidanski/opensource/embox/src/compat/libc/math/libm_simple/fmod.c:95
UbSan invalid builtin /home/i.maidanski/opensource/embox/src/compat/libc/math/libm_simple/fmod.c:95
0.988600
```

anton-bondarev, 2023-06-08:
Thanks. Do you mean that  there is mess output?
```
UbSan invalid builtin /home/i.maidanski/opensource/embox/src/compat/libc/math/libm_simple/fmod.c:95
UbSan invalid builtin /home/i.maidanski/opensource/embox/src/compat/libc/math/libm_simple/fmod.c:95
```

As I see output '0.988600' is correct

ivmai, 2023-06-08:
Yes, just these messages from UBSan. Sometimes I saw them in the middle of the printed number like this 0.98UbSan....86

anton-bondarev, 2023-06-08:
I create #2791 some tests for representing the issue
Now I think it's better use 	math_openlibm instead of math_simple
In conf/mods.conf
```
	//include embox.compat.libc.math_simple
	include embox.compat.libc.math_openlibm
```
or you can commnet in conf/build.conf
```CFLAGS += -fsanitize=undefined```

anton-bondarev, 2023-06-08:
Thanks. Feel free to open any issue!