embox/embox #2795: Add some tests for sqrt()

anton-bondarev, 2023-06-08:
There is a few tests sqrt() it is better if some other cases are added

AayushGupta69, 2023-06-10:
Hi @anton-bondarev, just wanted to ask if I can work on this? I want to start contributing to open source so would love to start with this one.

anton-bondarev, 2023-06-10:
> Hi @anton-bondarev, just wanted to ask if I can work on this? I want to start contributing to open source so would love to start with this one.

Hi @AayushGupta69 Of course you can! Feel freee ask any questions

AayushGupta69, 2023-06-10:
@anton-bondarev Thanks for replying. I will start working on this. It would be really helpful if you could elaborate on the type of test cases that are to be added.

anton-bondarev, 2023-06-12:
@AayushGupta69 I think it's OK to start with checking different return cases See h[ttps://man7.org/linux/man-pages/man3/sqrt.3.html](ttps://man7.org/linux/man-pages/man3/sqrt.3.html)

Have you already start existing tests?

AayushGupta69, 2023-06-13:
@anton-bondarev, I have reviewed the man pages you mentioned. I just have a small doubt: How do I run these tests? I have successfully configured the project, built the image, and run it on QEMU as mentioned in the README.md file .  I would be grateful if you could please guide me on running the tests?

AayushGupta69, 2023-06-13:
@anton-bondarev Hi, I have written some test cases for the sqrt() function in the file src/compat/libc/math/tests/sqrt_tests.c based on the return cases in the man pages and I was wondering if you could review them for me. Additionally, I have a question: Do I need to add the sqrt_tests module to the conf/mods.conf file in order to run it, or will it run on its own? I have attached a screenshot below for the test cases.

[image]

anton-bondarev, 2023-06-14:
@AayushGupta69 Hi, thanks. But if you want me to review you should create a PR! Now I can only say that you on the right way!:)
Here is the description to create a PR ([https://github.com/embox/embox/wiki/Contributing%3A-Git-workflow](https://github.com/embox/embox/wiki/Contributing%3A-Git-workflow))

Except this before I review your code you must debug it yourself (build, run and check that it's OK or may be you find a bug in main code). So first of all let's run it.

> Do I need to add the sqrt_tests module to the conf/mods.conf file in order to run it.

Yes, sure!

Feel free to ask any questions!

AayushGupta69, 2023-06-14:
@anton-bondarev Hi, I just wanted to give you an update. I have added the sqrt_tests module to the conf/mods.conf file. When I ran it, I encountered 2 errors initially, but I have resolved them as they were basically include errors in the main code. I have attached a screenshot for reference. Currently, I am working on resolving some errors that were encountered in the test cases that I wrote after I ran them.

[image]

anton-bondarev, 2023-06-14:
@AayushGupta69 Ouch, sorry

I think you should remove ` <framework/test/*>` includes

as for isnan try to use OpenLibm math lib version

```
// include embox.compat.libc.math_simple
include embox.compat.libc.math_openlibm
```

AayushGupta69, 2023-06-14:
@anton-bondarev Hi, I have created a pull request for this issue. I would be grateful if you could review it.

[https://github.com/embox/embox/pull/2799](url)

anton-bondarev, 2023-06-14:
@AayushGupta69 Great!
I've approved it and already merged!
Thanks!
I'll close this issue. You can take another one I think it's better if you take a little harder issue

closed 2023-06-15 by anton-bondarev