embox/embox #2822: Clean 'duktape' port

anton-bondarev, 2023-06-30:
Move fib.js to a apropriate templates where it needs
```
	@InitFS
	source "fib.js"
```

anton-bondarev, 2023-06-30:
@AayushGupta69 Hello,
I've created a "Good Second Issue"
Can you try to solve it? :)

AayushGupta69, 2023-06-30:
@anton-bondarev Hi, yeah sure will try my best to solve it.

anton-bondarev, 2023-07-01:
@AayushGupta69 

> Could you please elaborate a little on the issue

yes, sure.

If you look at embox/third-party/duktape/Mybuild file you see that it includes
```
	@InitFS
	source "fib.js"
```
It means that each time when somebody wants to use duktape he includes and fib.js into 'rootfs' . It is rather weird because if it needs fib.js you can put it in your template/rootfs folder.

How you can check that the issue solved . x86/test/lang template includes 'duktape' module. Its start_script inludes following line 
```
"duk fib.js",
```
It seems it's enough to move file fib.fs into apropriate rootfs folder. However you should make sure that it's the only template where 'duktape' is used and the command works OK after moving

AayushGupta69, 2023-07-02:
Hi @anton-bondarev , sorry for the late reply. I will start working on it.

anton-bondarev, 2023-07-03:
> sorry for the late reply

No problem! Do it when it comfortable for you! :)

AayushGupta69, 2023-07-06:
@anton-bondarev Hi,  How am I supposed to run the duktape module? Is this the correct way to run?
`@Runlevel(2) include third_party.cmd.duktape`

anton-bondarev, 2023-07-07:
@AayushGupta69 Hi,
Yes, I've reproduced the problem. It's happend if you add `include third_party.cmd.duktape` into x86/qemu template. To fix this you have to change libmath implamentation in mods.conf comment math_simple and add math_openlibm
```
	//include embox.compat.libc.math_simple
	include embox.compat.libc.math_openlibm
```
except this you need to disable sanitazer. In 'build.conf' comment `-fsanitize=undefined`
```
CFLAGS += -O0 -gdwarf-2
//CFLAGS += -fsanitize=undefined
CFLAGS += -m32
```

However I think you can just use x86/test/lang template. duktape is checking on CI server with fib.js

DmitryPilyuk, 2023-11-11:
Hello, I started working on this issue.

anton-bondarev, 2023-11-11:
@DmitryPilyuk hello, ok, I've assigned it with you

PR #3090 (2023-11-28): Clear 'duktape' port
Fix #2822
closed 2023-11-28 by anton-bondarev