embox/embox #2835: Fix shell scripts according to ShellCheck's recommendations

wawuwo, 2023-07-07:
Hi! Some of the scripts in `scripts` folder do not pass [ShellCheck](https://github.com/koalaman/shellcheck) analysis. For example:
```
$ shellcheck gdb_load_stm32.sh 

In gdb_load_stm32.sh line 3:
grep 'QSPI)' ./conf/lds.conf > /dev/null && echo "This script doesn't work with QSPI sections!" && exit -1
                                                                                                        ^-- SC2242 (error): Can only exit with status 0-255. Other data should be written to stdout/stderr.

In gdb_load_stm32.sh line 6:
GDB_CMD_FILE=$(mktemp --tmpdir=$PWD embox_gdb_stm32f7_qspi.XXXX)
                               ^--^ SC2086 (info): Double quote to prevent globbing and word splitting.

Did you mean: 
GDB_CMD_FILE=$(mktemp --tmpdir="$PWD" embox_gdb_stm32f7_qspi.XXXX)

In gdb_load_stm32.sh line 8:
EMBOX_BIN=$(mktemp --tmpdir=$PWD embox_XXXX.bin)
^-------^ SC2034 (warning): EMBOX_BIN appears unused. Verify use (or export if used externally).
                            ^--^ SC2086 (info): Double quote to prevent globbing and word splitting.

…
```

I would like to make a contribution and I am willing to fix such issues for all shell scripts from said folder. Hence I wanted to ask if this kind of contribution will be of any value for the project?

anton-bondarev, 2023-07-08:
@wawuwo Hi

> I would like to make a contribution and I am willing to fix such issues for all shell scripts from said folder. Hence I wanted to ask if this kind of contribution will be of any value for the project?

Yes, sure.

wawuwo, 2023-07-14:
Sorry for the late reply, I saw your answer and since then I've been fixing the scripts in a forked repository.

I wanted to ask if there is a way to check that I haven't broken something accidentally? If there are typical usage scenarios I could check them myself. (I am striving to be careful but you never know until you check)

anton-bondarev, 2023-07-15:
> Sorry for the late reply

No problem!

> I wanted to ask if there is a way to check that I haven't broken something accidentally? If there are typical usage scenarios I could check them myself. (I am striving to be careful but you never know until you check)

Unfortunataly for full checking you need apropriate boards. I have a couple of such boards. so let me check them when you create PR. I'll try to do it next week.

> I found this `git rebase upstream/master -i --exec make` in contributing guide, is it still relevant?

Yes rebasing is actual (although '--exec make' looks wierd). Howeve you should do rebasing under upstream/master after your PR will be approved/

wawuwo, 2023-07-16:
@anton-bondarev I've finished and ready to open a pull request. But I can't decide if I should squash commits into more larger groups before PR. I intentionally made them as atomic as it reasonably possible to ease bisecting in case of any problems and ended up with 33 commits, mostly one commit per file… Would it be better if I squash them or it's OK to have 33 of them?

anton-bondarev, 2023-07-17:
@wawuwo 

> I've finished and ready to open a pull request.

Great. Do it!:)
You can make PR as is,and after aproving you will squash to one or several commits

anton-bondarev, 2023-08-18:
Closed by
#2847
#2857
#2863
#2876
#2878
#2879
#2880
#2882
#2884
#2885
#2887
#2888

closed 2023-08-18 by anton-bondarev