embox/embox #2842: Change pins for UART6 in board_config for nucleo_f429zi

anton-bondarev, 2023-07-13:
User manual UM1974 (STM32 Nucleo-144 boards (MB1137) ) says that UART6 uses USART_A_TX - PG14 and USART_A_RX PG9
But in board_config/nucleo_f429zi.conf.h 
```
PIN("TX", PC, PIN_6, AF8),
PIN("RX", PC, PIN_7, AF8),
```
are declared

ZZW-IM, 2025-05-17:
Hey！ @anton-bondarev    I would like to solve this issue.
Firstly, I will read the user manual to determine the correct pin configuration and modify the embed/board_comfig/nucleate_f429zi.cnf. h file.
Verify the basic logic of UART driver in QEMU emulator
Let me know if my approach is correct.
Thanks

ZZW-IM, 2025-05-17:
Hey！ @anton-bondarev  I would like to solve this issue.

Firstly, I will read the user manual to determine the correct pin configuration and modify the embed/board_comfig/nucleate_f429zi.cnf. h file.

Secondly, if there is no real hardware, how can I verify it.

Looking forward to your advice.
Thanks
﻿

anton-bondarev, 2025-05-17:
Hi @ZZW-IM Thanks!

> Secondly, if there is no real hardware, how can I verify it.

Hm, It seems it is impossible to check this without real hardware, however I think it is enough if you make correct modifications from manual! Please create a PR

ZZW-IM, 2025-05-19:
Solved #3615  : )   @anton-bondarev