embox/embox #2911: arm/c++_demo fails to build

Deryugin, 2023-08-26:
/home/dde/src/embox/src/include/framework/mod/decls.h:25:47: error: conflicting declaration of 'const test_mod __mod__embox__test__cxx__memory_test' with 'C' linkage
   25 | #define __MOD(mod_nm)            MACRO_CONCAT(__mod__,            mod_nm)
      |                                               ^~~~~~~
/home/dde/src/embox/src/include/util/macro.h:34:32: note: in definition of macro '__MACRO_CONCAT'
   34 | #define __MACRO_CONCAT(m1, m2) m1 ## m2
      |                                ^~
/home/dde/src/embox/src/include/framework/mod/decls.h:25:34: note: in expansion of macro 'MACRO_CONCAT'
   25 | #define __MOD(mod_nm)            MACRO_CONCAT(__mod__,            mod_nm)
      |                                  ^~~~~~~~~~~~
/home/dde/src/embox/src/include/framework/mod/self.h:37:18: note: in expansion of macro '__MOD'
   37 | #define mod_self __MOD(__EMBUILD_MOD__)
      |                  ^~~~~
/home/dde/src/embox/src/include/framework/test/self.h:76:40: note: in expansion of macro 'mod_self'
   76 |         GLOBAL_C const struct test_mod mod_self = {                    \
      |                                        ^~~~~~~~
/home/dde/src/embox/src/tests/c++/test_cxx.h:23:9: note: in expansion of macro '__EMBOX_TEST_SUITE_NM_EXT'
   23 |         __EMBOX_TEST_SUITE_NM_EXT("" _description, &__TEST_FIXTURE_OP(suite_setup), \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/dde/src/embox/src/tests/c++/memory.cpp:26:1: note: in expansion of macro 'EMBOX_TEST_SUITE_EXT'
   26 | EMBOX_TEST_SUITE_EXT("c++ memory test", NULL, NULL, case_setup, NULL);
      | ^~~~~~~~~~~~~~~~~~~~
/home/dde/src/embox/src/include/framework/mod/decls.h:25:47: note: previous declaration with 'C++' linkage
   25 | #define __MOD(mod_nm)            MACRO_CONCAT(__mod__,            mod_nm)
      |                                               ^~~~~~~
/home/dde/src/embox/src/include/util/macro.h:34:32: note: in definition of macro '__MACRO_CONCAT'
   34 | #define __MACRO_CONCAT(m1, m2) m1 ## m2
      |                                ^~
/home/dde/src/embox/src/include/framework/mod/decls.h:25:34: note: in expansion of macro 'MACRO_CONCAT'
   25 | #define __MOD(mod_nm)            MACRO_CONCAT(__mod__,            mod_nm)
      |                                  ^~~~~~~~~~~~
/home/dde/src/embox/src/include/framework/mod/self.h:37:18: note: in expansion of macro '__MOD'
   37 | #define mod_self __MOD(__EMBUILD_MOD__)
      |                  ^~~~~
/home/dde/src/embox/src/include/framework/test/self.h:124:38: note: in expansion of macro 'mod_self'
  124 |         extern const struct test_mod mod_self;
      |                                      ^~~~~~~~
make[4]: *** [mk/image_lib.mk:45: /home/dde/src/embox/build/base/obj/src/tests/c++/memory.o] Error 1
make[3]: *** [mk/build.mk:26: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2

> arm-none-eabi-g++ -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/13.2.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-13.2.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (Arch Repository)

anton-bondarev, 2024-07-05:
@Deryugin Hello, Is it still broken?

Deryugin, 2024-07-28:
Well, this template was deleted two days ago in a0526e6a7f33fe3bb089eb5cc38032102b4bf6aa, so I guess it can't be broken now :)

Deryugin, 2024-07-28:
But `x86/c++_demo` still fails

```
/home/dde/embox/mk/extbld/arch-embox-g++ -fno-rtti  -fvisibility=hidden -I../../stlport -DUSE_SPRINTF_INSTEAD -I/home/dde/embox/third-party/STLport/include -Wno-narrowing -D_STLP_NO_RTTI -I../../stlport  -c -o obj/embox-gcc/so/dll_main.o ../../src/dll_main.cpp
In file included from ../../stlport/stl/_cmath.h:36,
                 from ../../stlport/stl/_cstdlib.h:169,
                 from ../../stlport/stl/_alloc.h:35,
                 from ../../stlport/memory:29,
                 from ../../src/dll_main.cpp:41:
/home/dde/embox/src/compat/cxx/include/cmath:36:17: error: ‘acos’ has not been declared in ‘::’
   36 |         using ::acos;
      |                 ^~~~
/home/dde/embox/src/compat/cxx/include/cmath:39:17: error: ‘asin’ has not been declared in ‘::’
   39 |         using ::asin;
      |                 ^~~~
/home/dde/embox/src/compat/cxx/include/cmath:40:17: error: ‘frexp’ has not been declared in ‘::’
   40 |         using ::frexp;
      |                 ^~~~~
/home/dde/embox/src/compat/cxx/include/cmath:41:17: error: ‘atan’ has not been declared in ‘::’
   41 |         using ::atan;
      |                 ^~~~
/home/dde/embox/src/compat/cxx/include/cmath:42:17: error: ‘ldexp’ has not been declared in ‘::’
   42 |         using ::ldexp;
      |                 ^~~~~
/home/dde/embox/src/compat/cxx/include/cmath:43:17: error: ‘atan2’ has not been declared in ‘::’
   43 |         using ::atan2;
      |                 ^~~~~
/home/dde/embox/src/compat/cxx/include/cmath:45:17: error: ‘cos’ has not been declared in ‘::’
   45 |         using ::cos;
      |                 ^~~
/home/dde/embox/src/compat/cxx/include/cmath:46:17: error: ‘sin’ has not been declared in ‘::’
   46 |         using ::sin;
      |                 ^~~
/home/dde/embox/src/compat/cxx/include/cmath:47:17: error: ‘tan’ has not been declared in ‘::’
   47 |         using ::tan;
      |                 ^~~
/home/dde/embox/src/compat/cxx/include/cmath:48:17: error: ‘cosh’ has not been declared in ‘::’
   48 |         using ::cosh;
      |                 ^~~~
/home/dde/embox/src/compat/cxx/include/cmath:49:17: error: ‘tanh’ has not been declared in ‘::’
   49 |         using ::tanh;
      |                 ^~~~
/home/dde/embox/src/compat/cxx/include/cmath:52:17: error: ‘sinh’ has not been declared in ‘::’
   52 |         using ::sinh;
      |                 ^~~~
In file included from ../../stlport/math.h:30,
                 from /home/dde/embox/src/compat/cxx/include/cmath:12:
/home/dde/embox/build/base/include/math.h:78:17: error: definition of ‘float __builtin_acos(float)’ ambiguates built-in declaration ‘double __builtin_acos(double)’
   78 | #define acos(x) __builtin_acos(x)
      |                 ^~~~~~~~~~~~~~
../../stlport/stl/_cmath.h:189:17: note: in expansion of macro ‘acos’
  189 |   inline __type func (__type x) \
      |                 ^~~~
../../stlport/stl/_cmath.h:289:7: note: in expansion of macro ‘_STLP_MATH_INLINEX’
  289 |       _STLP_MATH_INLINEX(float,func,cf) \
      |       ^~~~~~~~~~~~~~~~~~
../../stlport/stl/_cmath.h:432:1: note: in expansion of macro ‘_STLP_DEF_MATH_INLINE’
  432 | _STLP_DEF_MATH_INLINE(acos, acos)
      | ^~~~~~~~~~~~~~~~~~~~~
/home/dde/embox/build/base/include/math.h:78:17: error: definition of ‘long double __builtin_acos(long double)’ ambiguates built-in declaration ‘double __builtin_acos(double)’
   78 | #define acos(x) __builtin_acos(x)
      |                 ^~~~~~~~~~~~~~
../../stlport/stl/_cmath.h:189:17: note: in expansion of macro ‘acos’
  189 |   inline __type func (__type x) \
      |                 ^~~~
../../stlport/stl/_cmath.h:290:7: note: in expansion of macro ‘_STLP_MATH_INLINEX’
  290 |       _STLP_MATH_INLINEX(long double,func,cf)
      |       ^~~~~~~~~~~~~~~~~~
../../stlport/stl/_cmath.h:432:1: note: in expansion of macro ‘_STLP_DEF_MATH_INLINE’
  432 | _STLP_DEF_MATH_INLINE(acos, acos)
[... 251 lines cut ...]
  452 | _STLP_DEF_MATH_INLINE(tanh, tanh)
      | ^~~~~~~~~~~~~~~~~~~~~
/home/dde/embox/build/base/include/math.h:130:17: error: definition of ‘long double __builtin_tanh(long double)’ ambiguates built-in declaration ‘double __builtin_tanh(double)’
  130 | #define tanh(x) __builtin_tanh(x)
      |                 ^~~~~~~~~~~~~~
../../stlport/stl/_cmath.h:189:17: note: in expansion of macro ‘tanh’
  189 |   inline __type func (__type x) \
      |                 ^~~~
../../stlport/stl/_cmath.h:290:7: note: in expansion of macro ‘_STLP_MATH_INLINEX’
  290 |       _STLP_MATH_INLINEX(long double,func,cf)
      |       ^~~~~~~~~~~~~~~~~~
../../stlport/stl/_cmath.h:452:1: note: in expansion of macro ‘_STLP_DEF_MATH_INLINE’
  452 | _STLP_DEF_MATH_INLINE(tanh, tanh)
      | ^~~~~~~~~~~~~~~~~~~~~
make[7]: *** [../Makefiles/gmake/targets.mak:123: obj/embox-gcc/so/dll_main.o] Error 1
make[7]: Leaving directory '/home/dde/embox/build/extbld/third_party/STLport/libstlportg/STLport-5.2.1/build/lib'
make[6]: *** [Makefile:15: all] Error 2
make[6]: Leaving directory '/home/dde/embox/build/extbld/third_party/STLport/libstlportg/STLport-5.2.1'
make[5]: *** [Makefile:42: /home/dde/embox/build/extbld/third_party/STLport/libstlportg/.built] Error 2
make[4]: *** [/home/dde/embox/build/base/gen/third_party/STLport/libstlportg.extbld_rule.mk:23: /home/dde/embox/build/base/gen/third_party/STLport/libstlportg.__extbld-target] Error 2
make[3]: *** [mk/build.mk:28: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2
```