embox/embox #3034: Input does not work on stm32f103 bluepill board

anton-bondarev, 2023-10-30:
#3029

AlexVoyt, 2023-11-03:
This might be related to https://github.com/embox/embox/pull/3047

rai044der, 2023-11-03:
> This might be related to #3047

Nope, this didn't worked

rai044der, 2023-11-13:
Input does work though with the same elf file if its started not on hardware but with [Qiling](https://github.com/qilingframework/qiling/tree/master) started with python script below

```python
import time
import threading

import sys
sys.path.append("../..")

from qiling.core import Qiling
from qiling.const import QL_ARCH, QL_OS, QL_VERBOSE
from qiling.extensions.mcu.stm32f1 import stm32f103

ql = Qiling(["/home/rai044der/embox_fuzzing/embox/build/base/bin/embox"],
            archtype=QL_ARCH.CORTEX_M, ostype=QL_OS.MCU, env=stm32f103, verbose=QL_VERBOSE.OFF)

ql.hw.create('usart1')
ql.hw.create('rcc')

threading.Thread(target=lambda : ql.run(count=-1)).start()

while True:
    message = input('>> ').encode()

    ql.hw.usart1.send(message + b'\n')

    time.sleep(0.8)
    print(ql.hw.usart1.recv())
```

anton-bondarev, 2023-11-13:
> Input does work though with the same elf file if its started not on hardware but with [Qiling](https://github.com/qilingframework/qiling/tree/master) started with python script below

Thanks! Do you mean that it works on the simulator?

rai044der, 2023-11-13:
> Thanks! Do you mean that it works on the simulator?

Yes, it does work on the simulator. I've tried to load echo example to my board and this worked fine, so hardware is also fine.