embox/embox #3109: server61850/basic_io - closed socket connection

ivm2008, 2023-12-12:
To run the sample IEC 61850 server, you need to build embox with the command: make confload-project/server 61850/x86_qemu & make, run ./scripts/qemu/auto_qemu and start the server by typing basic_io. If you now connect to the server with any 61850 client (for example, to read the data model), then the 61850 server crashes.

[image]

ivm2008, 2023-12-12:
it is possible that the problem with connecting the client to the server is related to the undefined constant SIOCGIFHWADDR (in the file hal/ethernet/linux/ethernet_linux.c). We should find it

anton-bondarev, 2023-12-12:
> If you now connect to the server with any 61850 client (for example, to read the data model), then the 61850 server crashes.

@ivm2008 Could you describe how we can connetc wtih the 61850 client?
Should I build it from libiec61850? And what args I have to use?

ivm2008, 2023-12-12:
After download libiec61850 to the path
``` 
cd <path>
make
make install
```
move to the example folder <path/examples/xxx> and run make. Client should be run under root permissions with argument **ip** of the server iec61850