embox/embox #3171: Command 'rmdir' has been not implemented

anton-bondarev, 2024-02-13:


Shankari02, 2025-05-11:
Hello @anton-bondarev, I would like to solve this issue. 
As far as I have understood, I need to add rmdir() in the struct present in core.h, implement it in the source file and register the function in each filesystem driver.

Let me know if my approach is correct.
Thanks

anton-bondarev, 2025-05-12:
@Shankari02 No, here said about 'command' so it needs to add command 'rmdir' not a function. You can call the commands from the shell.

You are right about rmdir() pointer for the driver we will implement this feature in the future, however I'm not sure that it will be a field in the structure of the file driver operations

We described the simplest file system in this article ([Save settings and logs in a MCU internal flash](https://anton-bondarev.medium.com/save-settings-and-logs-in-a-mcu-internal-flash-1cbde63cba1f) )

Also you need writable file system to check the result Use USBi [https://github.com/embox/embox/wiki/Using-USB-on-qemu](https://github.com/embox/embox/wiki/Using-USB-on-qemu) or NFS [https://github.com/embox/embox/wiki/NFS-server](https://github.com/embox/embox/wiki/NFS-server)

Shankari02, 2025-05-12:
> [@Shankari02](https://github.com/Shankari02) No, here said about 'command' so it needs to add command 'rmdir' not a function. You can call the commands from the shell.

Oh, I see. I was trying to follow the implementation of mkdir command to understand.

> We described the simplest file system in this article ([Save settings and logs in a MCU internal flash](https://anton-bondarev.medium.com/save-settings-and-logs-in-a-mcu-internal-flash-1cbde63cba1f) )

I'll have a look at this!

> Also you need writable file system to check the result Use USBi https://github.com/embox/embox/wiki/Using-USB-on-qemu or NFS https://github.com/embox/embox/wiki/NFS-server

Thanks for this resource. Even I was wondering on how to create a writable fs. I was trying to mount a fs in the embox shell but it wasn't giving any output. I'll try this!

anton-bondarev, 2025-05-22:
> Do I need to add any other important case?

No, I think it's enough for now.