embox/embox #3230: Rename -v flag in mkdir command

cepessh, 2024-04-23:
In the current version of ```mkdir``` passing ```-v``` flag to the command means creating a virtual directory. As ```-v``` flag is widely used as a shortcut for ```--verbose```, I suggest to rename it to ```-s/--symbolic``` just like in ```ln``` command.

Shankari02, 2025-05-11:
Even I feel this could be a good change as it will give better clarity and users can differentiate between the both easily.

anton-bondarev, 2025-05-12:
> Even I feel this could be a good change as it will give better clarity and users can differentiate between the both easily.

Could you try to do this?

Shankari02, 2025-05-12:
Yes, I have implemented it locally already. Will test it once and generate a pull request in a day for sure!

closed 2025-05-19 by anton-bondarev