embox/embox #3251: UART is broken on platform/raspberry_pi/rpi0w

sycamoremoon, 2024-05-07:
> enviroment: make: 4.3, gcc: 12.3, OS:ubuntu 22.04

Following the wiki about [run embox on raspi](https://github.com/embox/embox/wiki/Raspberry-Pi), I'm try to run embox on my raspberry pi 0W. All things are the same except the `make confload-platform/raspberry_pi/rpi0w` and my SD card is `mmcblk` type rather than `sda`  type.

After doing all step, I got nothing from the UART. Seems like this:

[image]

but when I connect the rpi0w to a screen. It flashed the `fb_direct_access` picture correctly about one or two seconds.

---

I later tried other os as raspberry pi os for testing broad. It works great within UART.  

My `/dev/modem` is a soft link to `/dev/ttyUSB0` and bound-rate sets correct.

Shankari02, 2025-05-11:
Hello @sycamoremoon, does this issue still persist?

I can try running it on a pi zero and suggest some solution if possible.

sycamoremoon, 2025-05-11:
> Hello [@sycamoremoon](https://github.com/sycamoremoon), does this issue still persist?
> 
> I can try running it on a pi zero and suggest some solution if possible.

That't great if you can test it again, I am not sure if the issue still exist. Thanks

Shankari02, 2025-05-11:
Sure, I'll test it and get back to the issue if required.

sycamoremoon, 2025-05-12:
I retested it but still same issue. I think the serial port on my board is probably broken.

Shankari02, 2025-05-12:
I should be able to test it within a day or two, once I have the Pi Zero. I’ll keep you updated on how it goes from my end.

ZZW-IM, 2025-05-21:
hello
I don't have real hardware, but after reading the code and user manual, I would like to offer a suggestion that I hope will be helpful for this issue**: )**

In  /templates/rpi0w/mods.conf, the UART driver rpi_mini_uart is configured at Runlevel 2, while the diagnostic module diag is configured at Runlevel 1. 

The driver would be initialized in subsequent stages, but the early access of diag may has caused irreversible errors.

`	@Runlevel(2) include embox.driver.serial.rpi_mini_uart(baud_rate=115200)
	@Runlevel(1) include embox.driver.diag(impl="embox__driver__serial__rpi_mini_uart")`

My conjecture：
`	@Runlevel(1) include embox.driver.serial.rpi_mini_uart(baud_rate=115200)#Load the driver first
	@Runlevel(2) include embox.driver.diag(impl="embox__driver__serial__rpi_mini_uart")# and loads the diagnostic module afterwards`