embox/embox #3335: Add tests for log() from math.h

anton-bondarev, 2024-07-01:
Look at https://github.com/embox/embox/blob/master/src/compat/libc/math/tests/sqrt_tests.c

satyam1024, 2024-07-03:
Hello @anton-bondarev! Nice to meet you! I'd like to give this issue a try, but I'm new to open source. Any help regarding how to set up the application and the requirements for this issue would be greatly appreciated.

anton-bondarev, 2024-07-03:
@satyam1024 Hello. Nice to mmet you!

First of all you need to build and run Embox (on qemu) as described on our wiki ([https://github.com/embox/embox/wiki/Quick-Start](https://github.com/embox/embox/wiki/Quick-Start)

satyam1024, 2024-07-05:
Thank you for your reply. I'm trying to set up the application but keep encountering an error after running make confload-x86/qemu. The error message is:
make[1]: *** No rule to make target 'confload-x86/qemu'. Stop.
make: *** [Makefile:40: confload-x86/qemu] Error 2
Can you help me resolve this issue?

anton-bondarev, 2024-07-05:
> Can you help me resolve this issue?

Can you give me more datails?

In which folder you call this command? 'pwd'
What message will be if call 'make' without args?

what enviroment do you use?
Operating system (`uname -a`)
gcc version (`gcc --version`)
make version (`make --version`)

satyam1024, 2024-07-05:
I'm using Windows 10 (build 19045.4529) and have installed Ubuntu for Linux. I call this command on ubuntu .I ran the make command  in the following directory: /mnt/c/my computer/open source/embox.
 The output message for the make command is: 
">>> Configuration in: conf/ <<<".

system details:

uname -a: Linux [hidden] 5.15.153.1-microsoft-standard-WSL2 #1 SMP Fri Mar 29 23:14:13 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
GCC version: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Make version: GNU Make 4.3

anton-bondarev, 2024-07-05:
> /mnt/c/my computer/open source/embox

the problem may happen due to the space in your path. Try to rename "open source" to "open_source"

satyam1024, 2024-07-05:
Thank you , that seems to be the problem.
Now i am getting following error at the after this command  ./scripts/qemu/auto_qemu
 BUILDGEN /mnt/c/my_computer/open_source/embox/build
make[4]: /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh: No such file or directory
make[4]: *** [mk/script/build/oldconf-gen.mk:78: /mnt/c/my_computer/open_source/embox/build/base/gen/regions.lds.h] Error 127
make[3]: *** [mk/build.mk:39: /mnt/c/my_computer/open_source/embox/build/base/build-gen.timestamp] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2

anton-bondarev, 2024-07-05:
> make[4]: /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh: No such file or directory

Can you show me
`ls -la /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh`
It seems it has not executable flag
to add use
`chmod +x  /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh`

after it do 'make' again

satyam1024, 2024-07-05:
for <code> ls -la /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh </code> 
i am getting output ->    -rwxrwxrwx 1 satyam satyam 314 Jul  5 14:36

this is the gen_ld_region.sh file
[image]

i tried  <code>chmod +x  /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh</code> command it did not fix the error

anton-bondarev, 2024-07-06:
> For the command ` ls -la /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh `
> I receive the following output: -rwxrwxrwx 1 satyam satyam 314 Jul 5 14:36

Hm, it's enough flags to access
can you try to run /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh ?

satyam1024, 2024-07-06:
Thank you for replying.
I am getting this as an output 
<code> -bash: /mnt/c/my_computer/open_source/embox/mk/gen_ld_regions.sh: /bin/sh^M: bad interpreter: No such file or directory</code>

anton-bondarev, 2024-07-06:
> /bin/sh^M: bad interpreter: No such file or directory

It seems the problem like this https://stackoverflow.com/questions/14219092/bash-script-bin-bashm-bad-interpreter-no-such-file-or-directory

satyam1024, 2024-07-06:
Thank you for your help. I'll look into it.

bikakuka, 2024-10-09:
Hello @anton-bondarev , I want to work on this issue. Please assign me this issue.

anton-bondarev, 2024-10-09:
> Hello @anton-bondarev , I want to work on this issue. Please assign me this issue.

Hello @bikakuka Sure. I've assigned it with you

anton-bondarev, 2024-10-12:
Closed #3441

closed 2024-10-12 by anton-bondarev

anton-bondarev, 2024-10-16:
Closed #3443