embox/embox #3433: Error with OpenCV on Stm32

Kektis, 2024-09-26:
Following wiki on how to work with OpenCV on Stm32 https://github.com/embox/opencv_samples. 

I successfully use the commands:

```
$ cd embox/
$ make ext_conf EXT_PROJECT_PATH=<path-to>/opencv_samples
$ make confload-ext_project/opencv_samples/stm32f746g-discovery
```

But when I enter the command 

`make`

Errors occur. 
```

[/bin](tg://bot_command?command=bin)[/sh](tg://bot_command?command=sh): 2: [/home](tg://bot_command?command=home)[/nnizamov](tg://bot_command?command=nnizamov)[/embox](tg://bot_command?command=embox)-master/build/extbld/third_party/gcc/gcc_build/build: Permission denied
make[5]: *** [Makefile:67: [/home](tg://bot_command?command=home)[/nnizamov](tg://bot_command?command=nnizamov)[/embox](tg://bot_command?command=embox)-master/build/extbld/third_party/gcc/gcc_build/.configured] Error 126
make[4]: *** [/home/nnizamov/embox-master/build/base/gen/third_party/gcc/gcc_build.extbld_rule.mk:23: [/home](tg://bot_command?command=home)[/nnizamov](tg://bot_command?command=nnizamov)[/embox](tg://bot_command?command=embox)-master/build/base/gen/third_party/gcc/gcc_build.__extbld-target] Error 2
make[3]: *** [mk/build.mk:25: build] Error 2
make[2]: *** [mk/load.mk:41: build] Error 2
make[1]: *** [mk/main.mk:31: build] Error 2
make: *** [Makefile:40: all] Error 2
```

GSS version 10.0.5

AlekseyZhmulin, 2025-09-19:
Try building with this compiler (https://github.com/embox/opencv_samples/blob/main/opencv_samples/templates/arm-qemu/build.conf#L6)

AlekseyZhmulin, 2025-09-19:
This template builds some libraries from the gcc project, but to build them, you need a compiler with the appropriate version. You can change the gcc_version option (https://github.com/embox/opencv_samples/blob/main/opencv_samples/templates/arm-qemu/mods.conf#L235) to match your gcc compiler version. The default is 13.3.0, but you can also install 14.2.0, 9.3.0, 8.3.0, or 6.3.0.

krivenkodm, 2025-09-19:
Well, I solved it, but a little bit hard way:

Changed 
"third-party/gcc/Makefile":

```
- 	$(MKDIR) $(GCC_BUILD_DIR) && cd $(GCC_BUILD_DIR) && ( \
- 		$(PKG_SOURCE_DIR)/configure \
+ 	$(MKDIR) $(GCC_BUILD_DIR) && cd $(GCC_BUILD_DIR) && ( \
+ 		CC=/usr/bin/gcc CXX=/usr/bin/g++ \
+ 		$(PKG_SOURCE_DIR)/configure \
  			--with-cross-host=embox \
  			--target=$(GCC_TARGET) \
  			--prefix=$(PKG_INSTALL_DIR) \
```

"/embox/templates/ext_project/opencv_samples/arm-qemu/mods.conf":

include third_party.gcc.gcc_build(gcc_version="13.3.0") -> include third_party.gcc.gcc_build(gcc_version="9.3.0")

After that:

```
make confclean
make confload-ext_project/opencv_samples/arm-qemu

rm -rf ~/embox/build/extbld/third_party/gcc/gcc_build
export EMBOX_GCC_ENV="$HOME/embox/build/base/gen/embox_gcc_env.sh"
unset CC CXX
make -j"$(nproc)" \
  CC="$HOME/embox/mk/extbld/arch-embox-gcc" \
  CXX="$HOME/embox/mk/extbld/arch-embox-g++"
```
And:

```
qemu-system-arm -M integratorcp   -kernel build/base/bin/embox   -m 256 -smp 1   -serial mon:stdio -nographic   -net none
```

But I dont think this is a good solution

AlekseyZhmulin, 2025-09-22:
1) Set the gcc_version option according to your compiler version. If your arm-none-eabi-gcc version is 14.2.1, then set the gcc_version option as follows:
```
include third_party.gcc.gcc_build(gcc_version="14.2.0")
```

2) Make sure the gcc build is a cross build, not a "Canadian build." Check that build system is equal to host system.
For example:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... arm-none-eabi
```
If this is the problem, try removing the --with-cross-host option (https://github.com/embox/embox/blob/master/third-party/gcc/Makefile#L66). You can also try explicitly specifying the build system and host system using the --build and --host options.
Please let me know if any of this helped.