embox/embox #3483: mips: Do not set general exception vector correctly

123serge123, 2024-12-02:
In *exc_init.c*  mips_first_exception_handler is copied to fixed address KSEG0+0x180 but never check/set c0 EBase register.
At least need to call **mips_write_c0_ebase(KSEG0)**

123serge123, 2024-12-04:
Currently I use this simple patch:
```diff --git a/src/arch/mips/kernel/exc_init.c b/src/arch/mips/kernel/exc_init.c
index 1f3518c..1a19f6a 100644
--- a/src/arch/mips/kernel/exc_init.c
+++ b/src/arch/mips/kernel/exc_init.c
@@ -24,6 +24,9 @@ void mips_exception_init(void) {
 
 	flush_cache((unsigned long)(KSEG0 + 0x180), 0x80);
 
+	/* explicitly set ebase */
+	mips_write_c0_ebase(KSEG0);
+
 	execution_hazard_barrier();
 
 	/* Configure status register */
```

anton-bondarev, 2024-12-04:
> Currently I use this simple patch:
> 
> ```diff
> index 1f3518c..1a19f6a 100644
> --- a/src/arch/mips/kernel/exc_init.c
> +++ b/src/arch/mips/kernel/exc_init.c
> @@ -24,6 +24,9 @@ void mips_exception_init(void) {
>  
>  	flush_cache((unsigned long)(KSEG0 + 0x180), 0x80);
>  
> +	/* explicitly set ebase */
> +	mips_write_c0_ebase(KSEG0);
> +
>  	execution_hazard_barrier();
>  
>  	/* Configure status register */
> ```

Great!
Could you please create a PR with these changes?

123serge123, 2024-12-04:
Never did it before but I'll try :)

123serge123, 2024-12-05:
@AlekseyZhmulin 
Formally correct but for prev mips isa revisions EBase register may be optional.

123serge123, 2024-12-10:
Fixed in commit (22685b9fcf3cb457719277e2b4963df7d7af1460)

closed 2024-12-10 by 123serge123