embox/embox #3484: mips: flush_cache incorrectly work at ealier boot stage

123serge123, 2024-12-02:
First call of **flush_cache** is made from _exc_init.c_ before **kernel_start** and module _mips_cache_ init. In this case data and instruction cache line size are zero (not autodetected yet) so *flush_cache** do nothing. To turn off autodetection need refactoring of module _mips_cache_

anton-bondarev, 2024-12-03:
@123serge123 Thanks.
Maybe you already have some suggestions to solve this issue?

123serge123, 2024-12-04:
At this time I've made quick ~~and dirty hack~~ workaround to remove dependencies of Bailkal-T config, turn off *use_l2_cache* (set to false) and preset data/instruction cache line size in module _mips_cache_. Something like this:
```
diff --git a/src/arch/mips/mm/mips_cache.c b/src/arch/mips/mm/mips_cache.c
index 38a5fb3..ec2917a 100644
--- a/src/arch/mips/mm/mips_cache.c
+++ b/src/arch/mips/mm/mips_cache.c
@@ -12,7 +12,6 @@
 #include <asm/io.h>
 #include <asm/mipsregs.h>
 #include <asm/system.h>
-#include <drivers/mips/global_control_block.h>
 #include <embox/unit.h>
 #include <framework/mod/options.h>
 #include <kernel/panic.h>
@@ -23,17 +22,30 @@
 
 EMBOX_UNIT_INIT(mips_cache_init);
 
+#ifndef CONFIG_SYS_ICACHE_LINE_SIZE
+#define CONFIG_SYS_ICACHE_LINE_SIZE	32
+#endif
+
+#ifndef CONFIG_SYS_DCACHE_LINE_SIZE
+#define CONFIG_SYS_DCACHE_LINE_SIZE	32
+#endif
+
+#ifndef CONFIG_SYS_SCACHE_LINE_SIZE
+#define CONFIG_SYS_SCACHE_LINE_SIZE	0
+#endif
+
 #if USE_L2_CACHE
-static uint32_t l2_line_size;
+#include <drivers/mips/global_control_block.h>
+static uint32_t l2_line_size = CONFIG_SYS_SCACHE_LINE_SIZE;
 #endif
 
 #if USE_CACHE_SIZE_AUTO
-static uint32_t l1i_line_size;
-static uint32_t l1d_line_size;
+static uint32_t l1i_line_size = CONFIG_SYS_ICACHE_LINE_SIZE;
+static uint32_t l1d_line_size = CONFIG_SYS_DCACHE_LINE_SIZE;
 #endif
 
 static inline unsigned long icache_line_size(void) {
-#ifdef USE_CACHE_SIZE_AUTO
+#if USE_CACHE_SIZE_AUTO
 	return l1i_line_size;
 #else
 	return CONFIG_SYS_ICACHE_LINE_SIZE;
@@ -41,7 +53,7 @@ static inline unsigned long icache_line_size(void) {
 }
 
 static inline unsigned long dcache_line_size(void) {
-#ifdef USE_CACHE_SIZE_AUTO
+#if USE_CACHE_SIZE_AUTO
 	return l1d_line_size;
 #else
 	return CONFIG_SYS_DCACHE_LINE_SIZE;
@@ -49,7 +61,7 @@ static inline unsigned long dcache_line_size(void) {
 }
 
 static inline unsigned long scache_line_size(void) {
-#ifdef USE_L2_CACHE
+#if USE_L2_CACHE
 	return l2_line_size;
 #else
 	return CONFIG_SYS_SCACHE_LINE_SIZE;
```
This patch (+set ebase one) allows to run my device (Ralink SoC rt5350 - mips32r2 EL).
Other ways of problem solving:
1. Move *exc_init* into later boot stage (after module *mips_cache*).
2. Directrly call **mips_cache_init()** in *start* before setup of generic exception handler

anton-bondarev, 2024-12-04:
@123serge123 Thanks a lot!

It's cool that you've run embox on rt5350!

You could create a PR for this.
But I would like to suggest create a new platform platform/ralink/rt5350 with such changes
I can assist you of course if you agree

123serge123, 2024-12-04:
> You could create a PR for this.
Oh... Step by step 
> But I would like to suggest create a new platform platform/ralink/rt5350 with such changes
Already done in local repo but not commited yet
> I can assist you of course if you agree

anton-bondarev, 2024-12-04:
> Oh... Step by step

Yes, sure!

123serge123, 2024-12-10:
Fixed in commit (b7918d3c63fe0dfefb3d7dbf6e92d0a9a6731c3f)

closed 2024-12-10 by 123serge123