embox/embox #3491: USB support outdated or broken

123serge123, 2024-12-12:
While porting Ralink EHCI platform driver found this problem.
Compilation of *ehci_hcd* module failed.
After quick fix of compilation error registration of new ehci_hcd silently hang in *usb_core*.
So usb support outdated/broken or *mips* arch miss something critical

anton-bondarev, 2024-12-12:
@123serge123 Thanks! We'll check ehci_hcd on x86/qemu soon

anton-bondarev, 2024-12-13:
@123serge123 I fixed ehci_hcd and checked ehci_pci driver on x86/qemu it works OK.

I think we need ehci_ralink driver which extends ehci_hcd (see src/drivers/usb/chipidia for example) 

```
static int imx_usb_init(void) {
	clk_enable("usboh3");

	ehci_mxc_hc.base = IMX_USB_CORE_BASE;
	ehci_mxc_hc.irq_num = IMX6_USB0_IRQ;
	ehci_mxc_hc.ehci = NULL;
	ehci_mxc_hc.usbphy = imx_usb_phy_create(1);

	usbmisc_imx6q_init(1);
	imx_usb_phy_enable(1);
	imx_usb_powerup(1);

	ehci_mxc_init(&ehci_mxc_hc);

	ehci_hcd_register((void *) (ehci_mxc_hc.base + 0x100), ehci_mxc_hc.irq_num);

	/* Configure GPIO pins */

	return 0;
}
```

123serge123, 2024-12-13:
@anton-bondarev 
Ok. I'll try my *ehci_platform* module again.

anton-bondarev, 2024-12-14:
> Sorry. My fault. Yet another compilation error was for **NOMMU** arch (as mips).

Thanks. We'll try to fix this soon. I think it must work without MMU (NOMMU)