embox/embox #3494: x86/qemu with UAS-storage is broken

anton-bondarev, 2024-12-13:
During ./scripts/qemu/auto_qemu_with_usb_storage with x86/qemu following error is happened
```
[info] (pci) Apple OHCI usb host driver inserted

  ______
 |  ____|                                            __          __
 | |___  _ __ ___            ____  ____  ____  _____/ /   _____ / /
 |  ___|| '_ ` _ \          / __ \/ __ \/ __ \/ ___/ /   |_____| |
 | |____| | | | | |_ _ _   / /_/ / /_/ / /_/ (__  )_/    |_____| |
 |______|_| |_| |_(_|_|_)  \____/\____/ .___/____(_)           | |
                                     /_/                        \_\
 ASSERTION FAILED on CPU 0
	at /home/anton/embox/src/lib/libds/indexator/index_priv.c:57
	in function ind_set_bit

ind->mask != NULL

 --   00001000 * A R    thread 1  task 1 --------------------------------------

 22 0x002961d6 <__assertion_handle_failure+0x321> ...at/libc/assert/assert.c:44
 21 0x002a185a <ind_set_bit+0x201>  ././src/lib/libds/indexator/index_priv.c:47
 20 0x002a03d8 <index_lock+0x15>    ././src/lib/libds/indexator/index_lock.c:31
 19 0x0029fc7d <index_alloc+0x2e>  ././src/lib/libds/indexator/index_alloc.c:14
 18 0x0022c137 <usb_device_init+0x43a>        ././src/drivers/usb/usb_hub.c:246
 17 0x0022c3ae <usb_hub_port_init+0x127>      ././src/drivers/usb/usb_hub.c:288
 16 0x0022af93 <usb_new_device+0x228>          ././src/drivers/usb/usb_hub.c:56
 15 0x0022c5b6 <usb_hub_port_event+0x1d1>     ././src/drivers/usb/usb_hub.c:310
 14 0x0022c63f <usb_hub_event+0x73>           ././src/drivers/usb/usb_hub.c:327
 13 0x0022ce33 <usb_hub_probe+0x603>          ././src/drivers/usb/usb_hub.c:389
 12 0x0021c3fb <usb_driver_find+0x1a0>      ././src/drivers/usb/usb_driver.c:98
 11 0x0021c52a <usb_driver_probe+0x14>     ././src/drivers/usb/usb_driver.c:123
 10 0x0022b505 <usb_new_device+0x79a>          ././src/drivers/usb/usb_hub.c:56
  9 0x0011a549 <ohci_start+0x1275>   ././src/drivers/usb/hc/ohci/ohci_pci.c:183
  8 0x00218e1a <usb_hcd_register+0x38f>      ././src/drivers/usb/usb_core.c:312
  7 0x0011d023 <ohci_pci_init+0x12a> ././src/drivers/usb/hc/ohci/ohci_pci.c:636
  6 0x00215151 <pci_drv_probe+0x25e>        ././src/drivers/pci/pci_driver.c:18
  5 0x00215308 <pci_mod_enable+0x127>       ././src/drivers/pci/pci_driver.c:32
  4 0x0023323e <mod_enable+0xf8>                ././src/framework/mod/core.c:54
  3 0x00233984 <runlevel_set+0x1ee>         ././src/framework/mod/runlevel.c:55
  2 0x00103a26 <init+0x32>                             ././src/kernel/init.c:69
  1 0x0010396c <kernel_start+0x20>                     ././src/kernel/init.c:30

run 0x002269e7 <boot_stub+0x0>            ././src/kernel/sched/boot_thread.c:18
```

sycamoremoon, 2024-12-27:
calculate ➜ embox (master) ✗ cat ~/0001-lib-libds-indexator-Fix-index_init.patch 
```
From 341df38e15a779258f3ef9839a92b5524e936481 Mon Sep 17 00:00:00 2001
From: AlekseyZhmulin <ZhmulinAleksey@yandex.ru>
Date: Tue, 3 Sep 2024 18:18:13 +0300
Subject: [PATCH] lib: libds: indexator: Fix index_init()

---
 src/lib/libds/indexator/index_init.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/lib/libds/indexator/index_init.c b/src/lib/libds/indexator/index_init.c
index dd990fa0b..4bdefc28a 100644
--- a/src/lib/libds/indexator/index_init.c
+++ b/src/lib/libds/indexator/index_init.c
@@ -28,5 +28,5 @@ void index_init(struct indexator *ind, size_t start, size_t capacity,
 	ind->max = ind->prev = ind->end = ind->clamp_max = start + capacity - 1;
 
 	ind->mask = (unsigned long *)data;
-	memset(data, 0, binalign_bound(capacity, CHAR_BIT) / CHAR_BIT);
+	memset(data, 0, INDEX_DATA_LEN(capacity) * INDEX_DATA_BIT);
 }
-- 
2.47.1
```
roughly, I just found the commit which cause the question. But I don't know if this solution will cause other question :/
Use
```
patch -R -p1 < 0001-lib-libds-indexator-Fix-index_init.patch
```
Could make the Oops disappear shortly :)

anton-bondarev, 2024-12-30:
@sycamoremoon Thank you very much.
It'was a bug. And we fixed it in #3496

sycamoremoon, 2024-12-30:
Oh, you're welcome :)

anton-bondarev, 2025-01-08:
Fixed #3496