embox/embox #3500: Add tests for log2() from math.h

anton-bondarev, 2025-01-08:
Look at https://github.com/embox/embox/blob/master/src/compat/libc/math/tests/sqrt_tests.c

CryoColossus, 2025-01-08:
Can I do this? Its my first issue :)

anton-bondarev, 2025-01-09:
@CryoColossus 
First of all you need to know how to build and run your code. We described it in the [https://github.com/embox/embox/wiki/Quick-Start](https://github.com/embox/embox/wiki/Quick-Start)

Yes. there is a config file (**Mybuild** ) where you describe your new source file.
Besides this you have to add a new module into **conf/mods.conf**. It is a config file for your target system

walterhackin, 2025-03-14:
I can take this, please, assign it with me

anton-bondarev, 2025-03-15:
> I can take this, please, assign it with me

@walterhackin Sure. I've assigned it with you

anton-bondarev, 2025-06-06:
Closed by #3538