embox/embox #3506: USB-storage is not worked on platform/imx/imx6/sabrelite

anton-bondarev, 2025-01-14:
the following problem happens
```
[error] (ehci_hcd) qtd=0x19966000, buf=0x1996600c, token=0x00080a80, next=0x19966040
[error] (ehci_hcd) qtd=0x19966040, buf=0x1996604c, token=0x80120980, next=0x19966080
[error] (ehci_hcd) qtd=0x19966080, buf=0x1996608c, token=0x80008880, next=0x00000001
[error] (hub) GET_DESC failed

[error] (hub) usb_device_init failed ret -360
[error] (hub) usb_hub_port_init failed
[error] (hub) failed

```
when './scripts/qemu/auto_qemu_with_usb_storage' is used with
`make confload-platform/imx/imx6/sabrelite`

eatvector, 2025-05-25:
he USB driver on the imx6/sabrelite platform fails when submitting a USB_REQ_GET_DESCRIPTOR request. The problem appears when the driver submits the USB request to the hardware queue and enables scheduling—while an interrupt should be triggered upon completion to wake the waiting thread, no such interrupt occurs, ultimately causing the request to time out.

eatvector, 2025-05-25:
The USB driver on the imx6/sabrelite platform fails when submitting a USB_REQ_GET_DESCRIPTOR request. The problem appears when the driver submits the USB request to the hardware queue and enables scheduling—while an interrupt should be triggered upon completion to wake the waiting thread, no such interrupt occurs, ultimately causing the request to time out.